esphome/esphome #10025: Modbus Controller: Register ranges are incorrect when there are overlapping ranges

pascallj, 2025-08-02:
### The problem

In the current implementation of the `ModbusController::create_register_ranges_()` the ranges are created incorrectly when there are overlapping registers in the range.

For example:
```yaml
text_sensor:
  - id: reg7_and_8
    platform: modbus_controller
    register_type: holding
    modbus_controller_id: home_modbus_controller
    address: 7
    response_size: 4

  - id: reg8
    platform: modbus_controller
    register_type: holding
    modbus_controller_id: home_modbus_controller
    address: 8
    response_size: 2
```
```
[modbus_controller:338]: Register: 0x7 2 0 4 offset=0 skip=0 addr=0x5ca17611e908
[modbus_controller:351]: Started new range
[modbus_controller:338]: Register: 0x8 1 0 2 offset=0 skip=0 addr=0x5ca17611ea38
[modbus_controller:410]: Add range 0x7 2 skip:0
[modbus_controller:338]: Register: 0x8 1 0 2 offset=0 skip=0 addr=0x5ca17611ea38
[modbus_controller:351]: Started new range
[modbus_controller:422]: Add last range 0x8 1 skip:0
[...]
[modbus_controller:437]: sensormap
[modbus_controller:439]:  Sensor type=3 start=0x7 offset=0x0 count=2 size=4
[modbus_controller:439]:  Sensor type=3 start=0x8 offset=0x0 count=1 size=2
[modbus_controller:443]: ranges
[modbus_controller:445]:   Range type=3 start=0x7 count=2 skip_updates=0
[modbus_controller:445]:   Range type=3 start=0x8 count=1 skip_updates=0
```
The second sensor should have been using the same data as the first range and therefore no second range should have been created.

Also in this example:

```yaml
text_sensor:
  - id: reg7_and_8
    platform: modbus_controller
    register_type: holding
    modbus_controller_id: home_modbus_controller
    address: 7
    response_size: 4

  - id: reg7
    platform: modbus_controller
    register_type: holding
    modbus_controller_id: home_modbus_controller
    address: 7
    response_size: 2

  - id: reg8
    platform: modbus_controller
    register_type: holding
    modbus_controller_id: home_modbus_controller
    address: 8
    response_size: 2
```
```
[modbus_controller:338]: Register: 0x7 2 0 4 offset=0 skip=0 addr=0x5d3fa88b9918
[modbus_controller:351]: Started new range
[modbus_controller:338]: Register: 0x7 1 0 2 offset=0 skip=0 addr=0x5d3fa88b9a48
[modbus_controller:338]: Register: 0x8 1 0 2 offset=0 skip=0 addr=0x5d3fa88b9b78
[modbus_controller:371]: Re-use previous register - change to register: 0x7 1 offset=0
[modbus_controller:422]: Add last range 0x7 2 skip:0
[...]
[modbus_controller:437]: sensormap
[modbus_controller:439]:  Sensor type=3 start=0x7 offset=0x0 count=2 size=4
[modbus_controller:439]:  Sensor type=3 start=0x7 offset=0x0 count=1 size=2
[modbus_controller:439]:  Sensor type=3 start=0x7 offset=0x0 count=1 size=2
[modbus_controller:443]: ranges
[modbus_controller:445]:   Range type=3 start=0x7 count=2 skip_updates=0
```
the range is correct, but second sensor is changed incorrectly, because the offset has been stripped off.

I don't mind working on a fix/another implementation, but before I do, I try to understand the current implementation but I cannot wrap my head around the intended way of doing things currently:

What is the situation we are trying to solve in [line 357](https://github.com/esphome/esphome/blob/4f58e1c8b92f233af368f7d1698bcead16f9c8e2/esphome/components/modbus_controller/modbus_controller.cpp#L357)? Why are we looking at the previous sensor and not just at the full span of the range? There should be enough information in there to determine if the current sensor can fit inside the range, or if the range should be extended. I have a hunch looking at the previous sensor is causing issues here, because the current sensor can fall through the `if` mentioned above and still be assigned `prev`.

As the code maintainer is not responding to other PR's, I am tagging @stegm here, who implemented this code in https://github.com/esphome/esphome/pull/2981.

### Which version of ESPHome has the issue?

2025.8.0-dev

### What type of installation are you using?

pip

### What platform are you using?

Host

### Component causing the issue

modbus_controller

### YAML Config

```yaml
esphome:
  name: modbus-debug-2
  friendly_name: Modbus-Debug-2
  min_version: 2025.7.0
  name_add_mac_suffix: false

host:
  mac_address: "f6:5f:4c:7a:fa:6a"

uart:
  - id: home_modbus_uart
    baud_rate: 9600
    port: /dev/fakeUSB0

modbus:
  - uart_id: home_modbus_uart
    id: home_modbus

logger:
  level: VERY_VERBOSE

modbus_controller:
  - address: 1
    modbus_id: home_modbus
    id: home_modbus_controller
    update_interval: 5s
    setup_priority: -50
```

### Anything in the logs that might be useful for us?

```txt

```

stegm, 2025-08-06:
Line 357 attempts to reuse a previous register. However, I must admit that this if condition also seems a bit strange to me. It seems to me that the condition only handles the case when the current range matches the previous range, which makes the condition more complex than necessary. The 'else if' clause handles the case when the new range extends the previous one.

So I don't think your case is handled at all. It might even be that this case is wrong (because of the handling of the offset). I wonder if both case could not be handle together in one if-case.

```
if current start is within r start to end
   remove ix
   adapt offset of current
   extend r if necessary
   insert current
   decrement iterator
```

pascallj, 2025-08-06:
Thank you for your response!

That was what I had in mind too. However before implementing anything I wanted to make sure I didn't overlook some edge case I didn't understand. Will try to implement this in the future.