esphome/esphome #10026: syslog component stopped at 01.08.2025 00:00:00

markus-seidl, 2025-08-02:
### The problem

I have the following partial config (see next field)

Which sends syslog to my Grafana Alloy Server, which is configured like this:

```
loki.source.syslog "esphome" {

  listener {
    address  = "0.0.0.0:51801"
    protocol = "udp"
    label_structured_data = true
    syslog_format = "rfc3164"
    labels   = { component = "loki.source.syslog", protocol = "udp", cluster="esphome" }
  }

  relabel_rules = loki.relabel.esphome_syslog.rules
  forward_to = [loki.write.fenrir.receiver]
}
```

and it reports 

```
grafana-alloy-syslog  | ts=2025-08-02T11:59:06.200390851Z level=warn msg="error parsing syslog stream" component_path=/ component_id=loki.source.syslog.esphome err="expecting a Stamp timestamp [col 9]"
grafana-alloy-syslog  | ts=2025-08-02T11:59:08.196815585Z level=warn msg="error parsing syslog stream" component_path=/ component_id=loki.source.syslog.esphome err="expecting a Stamp timestamp [col 9]"
```

After some internet searching I saw that others had the issue, that the timestamp was not correctly formatted. I found no easy way to see the actual timestamp being sent (suggestions? would that help?), but I assume that this is the issue as it stopped exactly at that timestamp. From Grafana logs:

[image]

### Which version of ESPHome has the issue?

2025.7.4

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Component causing the issue

syslog

### YAML Config

```yaml
esphome:
  name: "floor-switch"
  name_add_mac_suffix: false
  min_version: 2024.6.0
  project:
    name: "template.floor-switch-shelly-plus-1pm"
    version: "1.0.0"
  platformio_options:
    board_build.f_cpu: 160000000L

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y
      CONFIG_ESP32_DEFAULT_CPU_FREQ_160: y
      CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ: "160"

logger:

udp:
  addresses: 192.168.1.125

syslog:
  port: 51801
```

### Anything in the logs that might be useful for us?

```txt
No relevant logs on the device
```

clydebarrow, 2025-08-02:
Thanks. You could monitor the actual data using a tool like Wireshark. I'll look into this but won't be for a few days.

clydebarrow, 2025-08-02:
try this:

```
external_components:
  - source: github://pr#10034
    components: [syslog]
    refresh: 1h
```

closed 2025-08-02 by clydebarrow

markus-seidl, 2025-08-03:
Thanks, it works! (for now?)

I've tracked down the actual code that parses this on alloys side, this is the unit test for the parser: https://github.com/leodido/go-syslog/blob/develop/rfc3164/example_test.go .
Also the https://datatracker.ietf.org/doc/html/rfc3164#section-5.1 specifies a ISO 8601 format. I thought this would look like this   2025-08-02T22:30:26Z , but maybe the old rfc3164 has some common format different than the rfc specifies.
Don't know what's actually "correct", just my investigation.

clydebarrow, 2025-08-03:
Section 5.1 mentions common extensions, the canonical form is in https://www.rfc-editor.org/rfc/rfc3164.html#section-4.1.2