esphome/esphome #10031: ERROR: Cannot establish a connection to the component registry

DChe47, 2025-08-02:
### The problem

I use ESPHome from the command line on several computers (Windows 11, Linux Mint). 
After some ESPHome update, I have errors:
`CMake Error at C:/Users/Dima/.platformio/packages/framework-espidf/tools/cmake/build.cmake:552 (message):
  ERROR: Cannot establish a connection to the component registry.  Are you
  connected to the internet?

  URL: https://components-file.espressif.com/components/espressif/mdns.json`
This is an example, the file name may be different. But if I try to access this file through the Browser, then I download it successfully. So I'm fine with the Internet.
Sometimes some projects are built without errors on one computer, but have this error on another.
I found similar errors in the Espressif project, there was something about a TLS certificate. Unfortunately, I can't fix this problem on my own. Please, help me!

### Which version of ESPHome has the issue?

2025.7.4

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Component causing the issue

CMake

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt
C:\WORK\ESP32\ESPHome>esphome compile esp32_enter.yaml
INFO ESPHome 2025.7.4
INFO Reading configuration esp32_enter.yaml...
INFO Detected timezone 'Europe/Moscow'
INFO Generating C++ source...
INFO Integrations changed, cleaning cmake cache...
INFO Deleting C:\WORK\ESP32\ESPHome\.esphome\build\esp32-enter\.pioenvs\esp32-enter\CMakeCache.txt
INFO Compiling app...
Processing esp32-enter (board: esp32dev; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32/releases/download/53.03.13/platform-espressif32.zip)
------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (53.3.13) > Espressif ESP32 Dev Module
HARDWARE: ESP32 160MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf @ 3.50302.0 (5.3.2)
 - tool-cmake @ 3.30.2
 - tool-esptoolpy @ 4.8.6
 - tool-mklittlefs @ 3.2.0
 - tool-ninja @ 1.9.0
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403
 - toolchain-esp32ulp @ 2.35.0-20220830
 - toolchain-xtensa-esp-elf @ 13.2.0+20240530
Reading CMake configuration...
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.50.1.windows.1")
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/Dima/.platformio/packages/toolchain-xtensa-esp-elf/bin/xtensa-esp32-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Users/Dima/.platformio/packages/toolchain-xtensa-esp-elf/bin/xtensa-esp32-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Users/Dima/.platformio/packages/toolchain-xtensa-esp-elf/bin/xtensa-esp32-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
Dependencies lock doesn't exist, solving dependencies.

-- Configuring incomplete, errors occurred!

fatal: not a git repository (or any of the parent directories): .git
CMake Error at C:/Users/Dima/.platformio/packages/framework-espidf/tools/cmake/build.cmake:552 (message):
  ERROR: Cannot establish a connection to the component registry.  Are you
  connected to the internet?

  URL: https://components-file.espressif.com/components/espressif/mdns.json

Call Stack (most recent call first):
  C:/Users/Dima/.platformio/packages/framework-espidf/tools/cmake/project.cmake:710 (idf_build_process)
  CMakeLists.txt:3 (project)

============================================= [FAILED] Took 101.51 seconds =============================================
```

swoboda1337, 2025-08-02:
Did you try again sounds like a temporary outage

DChe47, 2025-08-03:
> Did you try again sounds like a temporary outage

I've tried many times and even tried using a VPN. The problem remains.

swoboda1337, 2025-08-03:
> > Did you try again sounds like a temporary outage
> 
> I've tried many times and even tried using a VPN. The problem remains.

Can you access this file: https://components-file.espressif.com/components/espressif/mdns.json
Or this site: https://components.espressif.com/components/espressif/mdns/versions/1.8.2