esphome/esphome #10045: seeed_mr24hpc1 fails to compile on ESP8266 (and ESP32-C6)

andrewjsmith76, 2025-08-03:
### The problem

When trying to use the seeed_mr24hpc1 component, I get compilation errors - see below

I also tried using an ESP32-C6 (with appropriately adjusted yaml) and got a waaaay longer list of compile errors...

### Which version of ESPHome has the issue?

2025.7.4

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Component causing the issue

seeed_mr24hpc1

### YAML Config

```yaml
esphome:
  name: bnl2
  friendly_name: BNL2

esp8266:
  board: d1_mini
      
logger:
  baud_rate: 0

api:

ota:
  - platform: esphome
    password: "397a868b564ea45b9cde97637dbe60bd"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.1.247
    gateway: 192.168.1.1
    subnet: 255.255.255.0

  ap:
    ssid: "Bnl2 Fallback Hotspot"
    password: "OalMgpBTap6f"

captive_portal:

web_server:

uart:
  id: uart_mmwave
  tx_pin: GPIO1
  rx_pin: GPIO3
  baud_rate: 115200

seeed_mr24hpc1:
  id: mmwave_hub
  uart_id: uart_mmwave

binary_sensor:
  - platform: seeed_mr24hpc1
    has_target:
      name: "Presence Information"
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2025.7.4
INFO Reading configuration /config/esphome/bnl2.yaml...
INFO Detected timezone 'Europe/Berlin'
INFO Generating C++ source...
INFO Compiling app...
Processing bnl2 (board: d1_mini; framework: arduino; platform: platformio/espressif8266@4.2.1)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- ESPAsyncTCP @ 2.0.0
|-- ESP8266WiFi @ 1.0
|-- ESPAsyncWebServer @ 3.7.10
|-- DNSServer @ 1.1.1
|-- ESP8266mDNS @ 1.2
|-- ArduinoJson @ 7.4.2
Compiling .pioenvs/bnl2/src/esphome/components/seeed_mr24hpc1/seeed_mr24hpc1.cpp.o
Compiling .pioenvs/bnl2/src/esphome/components/template/text_sensor/template_text_sensor.cpp.o
Compiling .pioenvs/bnl2/src/esphome/components/text_sensor/filter.cpp.o
Compiling .pioenvs/bnl2/src/esphome/components/text_sensor/text_sensor.cpp.o
src/esphome/components/seeed_mr24hpc1/seeed_mr24hpc1.cpp: In member function 'void esphome::seeed_mr24hpc1::MR24HPC1Component::r24_frame_parse_open_underlying_information_(uint8_t*)':
src/esphome/components/seeed_mr24hpc1/seeed_mr24hpc1.cpp:437:21: error: 'class esphome::seeed_mr24hpc1::MR24HPC1Component' has no member named 'existence_boundary_select_'
437 |   } else if ((this->existence_boundary_select_ != nullptr) &&
|                     ^~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/seeed_mr24hpc1/seeed_mr24hpc1.cpp:439:15: error: 'class esphome::seeed_mr24hpc1::MR24HPC1Component' has no member named 'existence_boundary_select_'
439 |     if (this->existence_boundary_select_->has_index(data[FRAME_DATA_INDEX] - 1)) {
|               ^~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/seeed_mr24hpc1/seeed_mr24hpc1.cpp:440:13: error: 'class esphome::seeed_mr24hpc1::MR24HPC1Component' has no member named 'existence_boundary_select_'
440 |       this->existence_boundary_select_->publish_state(S_BOUNDARY_STR[data[FRAME_DATA_INDEX] - 1]);
|             ^~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/seeed_mr24hpc1/seeed_mr24hpc1.cpp:442:21: error: 'class esphome::seeed_mr24hpc1::MR24HPC1Component' has no member named 'motion_boundary_select_'
442 |   } else if ((this->motion_boundary_select_ != nullptr) &&
|                     ^~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/seeed_mr24hpc1/seeed_mr24hpc1.cpp:444:15: error: 'class esphome::seeed_mr24hpc1::MR24HPC1Component' has no member named 'motion_boundary_select_'
444 |     if (this->motion_boundary_select_->has_index(data[FRAME_DATA_INDEX] - 1)) {
|               ^~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/seeed_mr24hpc1/seeed_mr24hpc1.cpp:445:13: error: 'class esphome::seeed_mr24hpc1::MR24HPC1Component' has no member named 'motion_boundary_select_'
445 |       this->motion_boundary_select_->publish_state(S_BOUNDARY_STR[data[FRAME_DATA_INDEX] - 1]);
|             ^~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/seeed_mr24hpc1/seeed_mr24hpc1.cpp: In member function 'void esphome::seeed_mr24hpc1::MR24HPC1Component::r24_frame_parse_work_status_(uint8_t*)':
src/esphome/components/seeed_mr24hpc1/seeed_mr24hpc1.cpp:519:16: error: 'class esphome::seeed_mr24hpc1::MR24HPC1Component' has no member named 'scene_mode_select_'
519 |     if ((this->scene_mode_select_ != nullptr) && (this->scene_mode_select_->has_index(data[FRAME_DATA_INDEX]))) {
|                ^~~~~~~~~~~~~~~~~~
src/esphome/components/seeed_mr24hpc1/seeed_mr24hpc1.cpp:519:57: error: 'class esphome::seeed_mr24hpc1::MR24HPC1Component' has no member named 'scene_mode_select_'
519 |     if ((this->scene_mode_select_ != nullptr) && (this->scene_mode_select_->has_index(data[FRAME_DATA_INDEX]))) {
|                                                         ^~~~~~~~~~~~~~~~~~
src/esphome/components/seeed_mr24hpc1/seeed_mr24hpc1.cpp:520:13: error: 'class esphome::seeed_mr24hpc1::MR24HPC1Component' has no member named 'scene_mode_select_'
520 |       this->scene_mode_select_->publish_state(S_SCENE_STR[data[FRAME_DATA_INDEX]]);
|             ^~~~~~~~~~~~~~~~~~
src/esphome/components/seeed_mr24hpc1/seeed_mr24hpc1.cpp:542:16: error: 'class esphome::seeed_mr24hpc1::MR24HPC1Component' has no member named 'scene_mode_select_'
542 |     if ((this->scene_mode_select_ != nullptr) && (this->scene_mode_select_->has_index(data[FRAME_DATA_INDEX]))) {
|                ^~~~~~~~~~~~~~~~~~
src/esphome/components/seeed_mr24hpc1/seeed_mr24hpc1.cpp:542:57: error: 'class esphome::seeed_mr24hpc1::MR24HPC1Component' has no member named 'scene_mode_select_'
542 |     if ((this->scene_mode_select_ != nullptr) && (this->scene_mode_select_->has_index(data[FRAME_DATA_INDEX]))) {
|                                                         ^~~~~~~~~~~~~~~~~~
src/esphome/components/seeed_mr24hpc1/seeed_mr24hpc1.cpp:543:13: error: 'class esphome::seeed_mr24hpc1::MR24HPC1Component' has no member named 'scene_mode_select_'
543 |       this->scene_mode_select_->publish_state(S_SCENE_STR[data[FRAME_DATA_INDEX]]);
|             ^~~~~~~~~~~~~~~~~~
src/esphome/components/seeed_mr24hpc1/seeed_mr24hpc1.cpp: In member function 'void esphome::seeed_mr24hpc1::MR24HPC1Component::r24_frame_parse_human_information_(uint8_t*)':
src/esphome/components/seeed_mr24hpc1/seeed_mr24hpc1.cpp:582:21: error: 'class esphome::seeed_mr24hpc1::MR24HPC1Component' has no member named 'unman_time_select_'
582 |   } else if ((this->unman_time_select_ != nullptr) &&
|                     ^~~~~~~~~~~~~~~~~~
src/esphome/components/seeed_mr24hpc1/seeed_mr24hpc1.cpp:586:13: error: 'class esphome::seeed_mr24hpc1::MR24HPC1Component' has no member named 'unman_time_select_'
586 |       this->unman_time_select_->publish_state(S_UNMANNED_TIME_STR[data[FRAME_DATA_INDEX]]);
|             ^~~~~~~~~~~~~~~~~~
Compiling .pioenvs/bnl2/src/esphome/components/time/automation.cpp.o
*** [.pioenvs/bnl2/src/esphome/components/seeed_mr24hpc1/seeed_mr24hpc1.cpp.o] Error 1
========================== [FAILED] Took 8.28 seconds ==========================
```

ssieb, 2025-08-03:
Add at least one of https://esphome.io/components/seeed_mr24hpc1#select

limengdu, 2025-08-04:
You can refer to this example to check if anything is missing or if there are any errors in the entity names:
https://github.com/esphome/esphome/blob/dev/tests/components/seeed_mr24hpc1/common.yaml

jesserockz, 2025-08-04:
As per all other components in ESPHome, any component/hub code either needs to wrap any and every piece of code that touches select/sensor/number etc with`#ifdef` as the headers will not be available when they are not present in YAML.

This component has some of these checks and is missing some in various places.

limengdu, 2025-08-11:
@andrewjsmith76 What power supply are you using for the radar? In your working environment, is the radar reporting serial data normally? Please check the radar's operation using TTL.

limengdu, 2025-08-11:
@andrewjsmith76 If it's product-related, you can contact Seeed's technical support for assistance (techsupport@seeed.io).

andrewjsmith76, 2025-08-11:
Hello limengdu,

Thank you for your pointer. I'm powering the sensor off of the D1 mini's 5V pin. 

How do I check the radar's operation using TTL?

limengdu, 2025-08-12:
[image]
Like this, connect to the computer's serial monitor software to check if the radar is reporting data normally. Use your ESP8266 for power supply inspection.

andrewjsmith76, 2025-08-15:
Thanks. I have an FT232RL and connected it as shown. Then I used this command on my mac:
screen /dev/tty.usbserial-A50285BI ... and was shown a blank screen ... I guess this means the sensor is dead?

andrewjsmith76, 2025-08-20:
So, I purchased a Xiao-ESP32-C3 board... the one that the sensor is designed to be attached to (like a 'shield') and used the example code linked to above: [https://github.com/esphome/esphome/blob/dev/tests/components/seeed_mr24hpc1/common.yaml]

It compiles and installs just fine. Here's the log output:
`[06:23:48]E (246) esp_core_dump_flash: No core dump partition found!
[06:23:48][I][logger:169]: Log initialized
[06:23:48][C][safe_mode:082]: There have been 0 suspected unsuccessful boot attempts
[06:23:48][D][esp32.preferences:142]: Writing 1 items: 0 cached, 1 written, 0 failed
[06:23:48][I][app:049]: Running through setup()
[06:23:48][C][uart.arduino_esp32:077]: Running setup
[06:23:48][C][wifi:048]: Running setup
[06:23:48][C][wifi:061]: Starting
[06:23:48][C][wifi:061]:  Local MAC: 58:8C:81:A1:A2:84
[06:23:48][D][wifi:495]: Starting scan
[06:23:48][W][component:279]: wifi set Warning flag: scanning for networks
[06:23:48][D][wifi:516]: Found networks:
<wifi info redacted>
[06:23:50][I][wifi:630]: Connected
[06:23:50][C][wifi:434]:   Local MAC: 58:8C:81:A1:A2:84
[06:23:50][C][wifi:439]:   SSID: [redacted]
[06:23:50][C][wifi:442]:   IP Address: [redacted]
[06:23:50][C][wifi:446]:   BSSID: [redacted]
[06:23:50][C][wifi:446]:  Hostname: 'xiao-mmwave'
[06:23:50][C][wifi:446]:  Signal strength: -45 dB ▂▄▆█
[06:23:50][C][wifi:455]:   Channel: 11
[06:23:50][C][wifi:455]:  Subnet: 255.255.255.0
[06:23:50][C][wifi:455]:  Gateway: [redacted]
[06:23:50][C][wifi:455]:  DNS1: [redacted]
[06:23:50][C][wifi:455]:  DNS2: [redacted]
[06:23:50][D][wifi:639]: Disabling AP
[06:23:50][C][web_server:282]: Running setup
[06:23:50][C][api:034]: Running setup
[06:23:50][W][component:279]: api set Warning flag: unspecified
[06:23:50][C][seeed_mr24hpc1:065]: Running setup
[06:23:50][D][number:012]: 'Custom Mode': Sending state 0.000000
[06:23:50][D][sensor:104]: 'Current Custom Mode': Sending state 0.00000  with 0 decimals of accuracy
[06:23:50][D][text_sensor:069]: 'Custom Mode Status': Sending state 'Not in custom mode'
[06:23:50][D][number:012]: 'Custom Mode': Sending state 0.000000
[06:23:50][C][seeed_mr24hpc1:094]: Set up MR24HPC1 complete
[06:23:50][I][app:091]: setup() finished successfully!
[06:23:50][W][component:307]: wifi cleared Warning flag
[06:23:50][I][app:149]: ESPHome version 2025.7.5 compiled on Aug 20 2025, 06:22:44
[06:23:50][C][wifi:613]: WiFi:
[06:23:50][C][wifi:434]:   Local MAC: 58:8C:81:A1:A2:84
[06:23:50][C][wifi:439]:   SSID: [redacted]
[06:23:50][C][wifi:442]:   IP Address: [redacted]
[06:23:50][C][wifi:446]:   BSSID: [redacted]
[06:23:50][C][wifi:446]:  Hostname: 'xiao-mmwave'
[06:23:50][C][wifi:446]:  Signal strength: -45 dB ▂▄▆█
[06:23:50][C][wifi:455]:   Channel: 11
[06:23:50][C][wifi:455]:  Subnet: 255.255.255.0
[06:23:50][C][wifi:455]:  Gateway: [redacted]
[06:23:50][C][wifi:455]:  DNS1: [redacted]
[06:23:50][C][wifi:455]:  DNS2: [redacted]
[06:23:50][C][logger:246]: Logger:
[06:23:50][C][logger:246]:  Max Level: DEBUG
[06:23:50][C][logger:246]:  Initial Level: DEBUG
[06:23:50][C][logger:252]:   Log Baud Rate: 115200
[06:23:50][C][logger:252]:  Hardware UART: USB_CDC
[06:23:50][C][logger:259]:   Task Log Buffer Size: 768
[06:23:50][C][uart.arduino_esp32:151]: UART Bus 0:
[06:23:50][C][uart.arduino_esp32:152]:   TX Pin: GPIO3
[06:23:50][C][uart.arduino_esp32:153]:   RX Pin: GPIO2
[06:23:50][C][uart.arduino_esp32:155]:   RX Buffer Size: 256
[06:23:50][C][uart.arduino_esp32:157]:   Baud Rate: 115200 baud
[06:23:50][C][uart.arduino_esp32:157]:  Data Bits: 8
[06:23:50][C][uart.arduino_esp32:157]:  Parity: NONE
[06:23:50][C][uart.arduino_esp32:157]:  Stop bits: 1
[06:23:50][C][captive_portal:099]: Captive Portal:
[06:23:50][C][web_server:310]: Web Server:
[06:23:50][C][web_server:310]:  Address: xiao-mmwave.local:80
[... 129 lines cut ...]
[06:26:17][D][select:015]: 'Scene': Sending state Living Room (index 1)
[06:26:17][D][number:012]: 'Custom Mode': Sending state 0.000000
[06:26:17][D][sensor:104]: 'Current Custom Mode': Sending state 0.00000  with 0 decimals of accuracy
[06:26:27][D][number:054]: 'Sensitivity' - Setting number value
[06:26:27][D][number:113]:   New number value: 1.000000
[06:26:39][D][number:054]: 'Motion Trigger Time' - Setting number value
[06:26:39][D][number:113]:   New number value: 62.000000
[06:26:40][D][number:054]: 'Motion To Rest Time' - Setting number value
[06:26:40][D][number:113]:   New number value: 1349.000000
[06:27:00][D][button:010]: 'End Of Custom Mode Settings' Pressed.
[06:27:00][D][number:012]: 'Custom Mode': Sending state 0.000000
[06:27:02][D][number:054]: 'Custom Mode' - Setting number value
[06:27:02][D][number:113]:   New number value: 2.000000
[06:27:02][D][number:012]: 'Custom Mode': Sending state 2.000000
[06:27:04][D][button:010]: 'End Of Custom Mode Settings' Pressed.
[06:27:04][D][number:012]: 'Custom Mode': Sending state 0.000000
[06:27:16][D][button:010]: 'Module Restart' Pressed.
...`

I can play around with the settings and some of them seem to ‘stick’ while others don’t … I don’t see any log entries that indicate presence/motion detection.

Is there something else I'm doing wrong?

Best regards,
Andrew

closed 2025-11-27 by github-actions[bot]

unhuzpt, 2026-01-11:
Hello

Please note this thread: https://community.home-assistant.io/t/uart-issue-esphome/971765/12

After ESPHome v2025/11 all devices using mr24hpc1 just stopped working.