esphome/esphome #10047: ESPHome 2025.7.4 Touch Input Compile Error

tesanders, 2025-08-03:
### The problem

The attached code has been executing successfully for several months and is currently executing with ESPHome 2025.5.1   The compile error occurs when I attempt to upgrade to 2025.7.4   The same error occurs for the arduino and esp-idf frameworks.

### Which version of ESPHome has the issue?

2025.7.4

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

esp32_touch

### YAML Config

```yaml
################################################################################
#  FILE:  sump-level.yaml
#  DATE:  12/23/2024
################################################################################

################################################################################
# Substitution Variables
################################################################################
substitutions:
  device_ip_address: xxx.xxx.xxx.xxx
  device_gateway: xxx.xxx.xxx.xxx
  device_subnet: xxx.xxx.xxx.xxx
  device_sampling_time: 30s

################################################################################
# Board Configuration
################################################################################
esphome:
  name: sump-level
  friendly_name: Sump Level

esp32:
  board: um_feathers3
  framework:
    type: arduino

################################################################################
# Enable logging
################################################################################
logger:

################################################################################
# Enable Home Assistant API
################################################################################
api:
  encryption:
    key: ""

################################################################################
# OTA
################################################################################
ota:
  - platform: esphome
    password: ""

################################################################################
# WiFi
################################################################################
wifi:
  networks:
    - ssid: !secret wifi_ssid
      password: !secret wifi_password
  manual_ip:
    static_ip: ${device_ip_address}
    gateway: ${device_gateway}
    subnet: ${device_subnet}
  ap:
    ssid: "Sump-Level Fallback Hotspot"
    password: ""

################################################################################
# Captive Portal
################################################################################  
captive_portal:

################################################################################
# Capacitance Input for Level Measurement
################################################################################  
esp32_touch:
  setup_mode: False
  id: esp32_touch_1
  measurement_duration: 0.25ms

binary_sensor:
  - platform: esp32_touch
    id: esp32_touch_pad
    pin: GPIO14
    threshold: 0

sensor:
  - platform: template
    name: Sump Level Inches
    lambda: |-
        float raw_cap = (float) round((id(esp32_touch_pad).get_value()) / 10000);
        return (float) ((raw_cap * 0.0189) - 0.6575);
    update_interval: 10s
    filters:
      - sliding_window_moving_average:
          window_size: 4
          send_every: 4
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2025.7.4
INFO Reading configuration /config/esphome/sump-level.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing sump-level (board: um_feathers3; framework: arduino; platform: https://github.com/pioarduino/platform-espressif32/releases/download/53.03.13/platform-espressif32.zip)
--------------------------------------------------------------------------------
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
 - framework-arduinoespressif32 @ 3.1.3 
 - framework-arduinoespressif32-libs @ 5.3.0+sha.489d7a2b3a 
 - tool-esptoolpy @ 4.8.6 
 - tool-mklittlefs @ 3.2.0 
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403 
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403 
 - toolchain-riscv32-esp @ 13.2.0+20240530 
 - toolchain-xtensa-esp-elf @ 13.2.0+20240530
Dependency Graph
|-- Networking @ 3.1.3
|-- AsyncTCP @ 3.4.5
|-- WiFi @ 3.1.3
|-- FS @ 3.1.3
|-- Update @ 3.1.3
|-- ESPAsyncWebServer @ 3.7.10
|-- ESP32 Async UDP @ 3.1.3
|-- DNSServer @ 3.1.3
|-- ESPmDNS @ 3.1.3
|-- noise-c @ 0.1.10
Compiling .pioenvs/sump-level/src/main.cpp.o
Compiling .pioenvs/sump-level/lib64d/WiFi/WiFiMulti.cpp.o
Compiling .pioenvs/sump-level/lib64d/WiFi/WiFiSTA.cpp.o
Compiling .pioenvs/sump-level/lib64d/WiFi/WiFiScan.cpp.o
/config/esphome/sump-level.yaml: In lambda function:
/config/esphome/sump-level.yaml:84:55: error: 'class esphome::esp32_touch::ESP32TouchBinarySensor' has no member named 'get_value'; did you mean 'get_name'?
   84 |         float raw_cap = (float) round((id(esp32_touch_pad).get_value()) / 10000);
      |                                                       ^~~~~~~~~
      |                                                       get_name
Compiling .pioenvs/sump-level/lib01c/FS/FS.cpp.o
Archiving .pioenvs/sump-level/lib64d/libWiFi.a
Compiling .pioenvs/sump-level/lib01c/FS/vfs_api.cpp.o
Compiling .pioenvs/sump-level/libbc6/Update/HttpsOTAUpdate.cpp.o
*** [.pioenvs/sump-level/src/main.cpp.o] Error 1
========================== [FAILED] Took 5.80 seconds ==========================
```

0n3man, 2025-08-03:
I use touch 4 different lamps.  The code for three of the lamps utilize the get_value() method.  One lamp uses the following esp32 section:

esp32:
  board: esp32dev

Two of the lamps have 

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino

The esp32dev device complies and works without issue.  The two with board esp32-s3-devkitc-1 produce the same error mentioned in this thread.  On the failing lamps I've tried both framework types (arudino and ESP-IDF), neither compiled successfully.

I have a forth lamp that uses the same esp32-s3-devkitc-1 board.  With this one I just use a binary_sensor with platform esp32_touch.  While this complies successfully, the binary sensor never reports a state transition.  It always remains as off.  The threshold is set to 520000.  This is code from that lamp:

esp32_touch:
# set setup_mode True and log level to Debug above when determining touch threashold value
#  setup_mode: false
  setup_mode: true  
  measurement_duration: 0.25ms
  
binary_sensor:
  - platform: esp32_touch
    name: "Touch Pad Pin 13"
    pin: GPIO13
    threshold: 520000
    #threshold: 500000
    filters:
      # Small filter, to debounce the spurious events.
      - delayed_on: 10ms
      - delayed_off: 10ms
    #on_press:
    on_click:
    - min_length: 10ms
      max_length: 500ms
      # Short touch to turn light on and off
      then:
        - logger.log: 
            level: info
            format: "Touch turn switch on"
 ...
    - min_length: 500ms
      max_length: 2000ms
      then:
        - logger.log: "${devicename} long click"
 ...
    - min_length: 2001ms
      max_length: 120000ms
      then:
        - logger.log: "${devicename} Unexpected long click"

Here are some log entries showing the threshold being crossed:
15:56:44	[D]	[esp32_touch:103]	Touch Pad 'Touch Pad Pin 13' (T13): 456107
15:56:44	[D]	[esp32_touch:103]	Touch Pad 'Touch Pad Pin 13' (T13): 447356
15:56:44	[D]	[sensor:104]	 'Uptime Sensor': Sending state 889.19598 s with 0 decimals of accuracy
15:56:44	[D]	[esp32_touch:103]	Touch Pad 'Touch Pad Pin 13' (T13): 500852
15:56:45	[D]	[esp32_touch:103]	Touch Pad 'Touch Pad Pin 13' (T13): 615266
15:56:45	[D]	[esp32_touch:103]	Touch Pad 'Touch Pad Pin 13' (T13): 618189
15:56:45	[D]	[esp32_touch:103]	Touch Pad 'Touch Pad Pin 13' (T13): 617681
15:56:45	[D]	[esp32_touch:103]	Touch Pad 'Touch Pad Pin 13' (T13): 462919
15:56:46	[D]	[esp32_touch:103]	Touch Pad 'Touch Pad Pin 13' (T13): 446461
15:56:46	[D]	[esp32_touch:103]	Touch Pad 'Touch Pad Pin 13' (T13): 455572

The binary sensor never changes state from off to on, and the debug from the code that should print when the event happens is never generated.

AntorFr, 2025-08-06:
@ssieb  thanks for your feed back.
This as introduced an un-documented breaking change with version 2025.7.0 
the doc ([raw-values](https://esphome.io/components/binary_sensor/esp32_touch.html#raw-values) is documenting 
return id(esp32_touch_pad).get_value(); for calibration that is not working anymore.

Is it possible to calibrate the S2/S3 from an other way ? 

This feature is used for [onju voice satellite](https://github.com/tetele/onju-voice-satellite) ans can't compile anymore

closed 2025-08-07 by kbx81