esphome/esphome #10120: problems with neopixel platform

Operrr, 2025-08-07:
### The problem

After updating ESPHOME to version 2025.7.5, all devices using the Neopixel platform stopped working. ESP32-S2 and ESP32-S3 worked like night lights, but after the update they are in a constant reboot. If you flash the same code in ESP8266, everything works fine. This already happened a couple of months ago and I solved the problem by installing an older version of ESPHOME. If you remove everything related to neopixel from the code, the device will boot and work normally, including the web interface.

### Which version of ESPHome has the issue?

2025.7.5

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

platform: neopixelbus

### YAML Config

```yaml
esphome:
  name: night-light                                    
  friendly_name: Ночник
esp32:
  board: esp32-c3-devkitm-1                             
  framework:
    type: arduino
#-------------------------------------------------------------------------------------------------
logger:                                                 
  level: DEBUG                                          
api:                                                    
  encryption:
    key: "secret key :-)"
ota:                                                    
  - platform: esphome
    password: "Mypassword"
wifi:                                                   
  output_power: 16db                                    
  ssid: !secret wifi_ssid
  password: !secret wifi_password
#-------------------------------------------------------------------------------------------------
web_server:                                            
  port: 80
  version: 3
  local: true                                           
  ota: false                              
  auth:
    username: !secret web_server_username
    password: !secret web_server_password
#-------------------------------------------------------------------------------------------------
binary_sensor:                      
  - platform: status
    name: "WiFi"   
#-------------------------------------------------------------------------------------------------
sensor:
  - platform: wifi_signal                               
    name: "WiFi Signal dB"
    id: wifi_signal_db
    update_interval: 15s
    entity_category: "diagnostic"
  - platform: copy                                     
    source_id: wifi_signal_db
    name: "WiFi Signal Percent"
    filters:
      - lambda: return min(max(2 * (x + 100.0), 0.0), 100.0);
    unit_of_measurement: "%"
    entity_category: "diagnostic"
[... 146 lines cut ...]
[18:48:53]Build:Feb  7 2021
[18:48:53]rst:0xc (RTC_SW_CPU_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
[18:48:53]Saved PC:0x40381a18
[18:48:53]SPIWP:0xee
[18:48:53]mode:DIO, clock div:1
[18:48:53]load:0x3fcd5820,len:0x1144
[18:48:53]load:0x403cc710,len:0xad8
[18:48:53]load:0x403ce710,len:0x2f80
[18:48:53]entry 0x403cc710
[18:48:53]E (233) rmt(legacy): CONFLICT! driver_ng is not allowed to be used with the legacy driver
[18:48:53]
[18:48:53]abort() was called at PC 0x42047077 on core 0
[18:48:53]Core  0 register dump:
[18:48:53]MEPC    : 0x40381a76  RA      : 0x40387d8a  SP      : 0x3fcde230  GP      : 0x3fc90a00  
[18:48:53]TP      : 0x00000000  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
[18:48:53]S0/FP   : 0x3fcde25c  S1      : 0x3fcde25c  A0      : 0x3fcde268  A1      : 0x3fcde24a  
[18:48:53]A2      : 0x00000000  A3      : 0x3fcde295  A4      : 0x00000001  A5      : 0x3fc99000  
[18:48:53]A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x0000003e  S3      : 0x3fcde42c  
[18:48:53]S4      : 0x00042d44  S5      : 0x4038149c  S6      : 0x3c100000  S7      : 0xffff0000  
[18:48:53]S8      : 0x00060020  S9      : 0x00010020  S10     : 0x00000005  S11     : 0x3fcde3c0  
[18:48:53]T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
[18:48:53]MSTATUS : 0x00001801  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
[18:48:53]MHARTID : 0x00000000  
[18:48:53]
```

ssieb, 2025-08-08:
Not crashing is good, but you should be using the rmt led component anyway.

andrewjswan, 2025-08-10:
> but you should be using the rmt led component anyway

It is less stable, especially on the Arduino framework. 
Tested on the matrix, and led lights, artifacts from time to time, although on neopixel they have not been for years.