esphome/esphome #10136: bl0942 component doesn't work on the esp-idf framework

MikeTTTTT, 2025-08-08:
### The problem

The bl0942 uart power meter does not work on the esp-idf framework. The device log contains messages
[00:05:18][W][bl0942:082]: BL0942 invalid checksum! 0x0C != 0x03
[00:05:18][W][bl0942:082]: BL0942 invalid checksum! 0x3D != 0x02
[00:05:18][W][bl0942:082]: BL0942 invalid checksum! 0xDB != 0x00
[00:05:18][W][bl0942:082]: BL0942 invalid checksum! 0x61 != 0x4E
[00:05:18][W][bl0942:082]: BL0942 invalid checksum! 0x08 != 0x20

### Which version of ESPHome has the issue?

2025.7.5

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

bl0942

### YAML Config

```yaml
esphome:
  name: sh19-fuseboxright
  friendly_name: SH19_FuseBoxRight

esp32:
  board: esp32dev
  framework:
    type: esp-idf

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "="

ota:
  - platform: esphome
    password: "="

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Sh19-Fuseboxright"
    password: "="

captive_portal:

# Настройка UART для связи с BL0942
uart:
  - id: uart_1
    rx_pin: GPIO18
    tx_pin: GPIO19
    data_bits: 8
    baud_rate: 9600
    stop_bits: 1
    parity: NONE
    rx_buffer_size: 256
    debug:
      direction: BOTH
      dummy_receiver: true
      after:
        delimiter: "\n"
      sequence:
        - lambda: UARTDebug::log_hex(direction, bytes, 32);

sensor:
  - platform: bl0942
    uart_id: uart_1
    address: 3
    voltage:
      name: "Voltage1"
      unit_of_measurement: "V"
    voltage_reference: 5
    current:
      name: "Сurrent1"
      unit_of_measurement: "A"
    current_reference: 1
    power:
      name: "Power1"
      unit_of_measurement: "W"
    energy:
      name: "Meter1"
      unit_of_measurement: "Wh"
    update_interval: 10s  # Обновление каждые 10 секунд
```

### Anything in the logs that might be useful for us?

```txt
ESP-IDF uart log:
[00:04:48][D][uart_debug:114]: >>> 58 AA
[00:04:58][D][uart_debug:114]: >>> 58 AA
[00:05:08][D][uart_debug:114]: >>> 58 AA
[00:05:18][D][uart_debug:114]: >>> 58 AA
[00:05:18][W][bl0942:082]: BL0942 invalid checksum! 0x0C != 0x03
[00:05:18][W][bl0942:082]: BL0942 invalid checksum! 0x3D != 0x02
[00:05:18][W][bl0942:082]: BL0942 invalid checksum! 0xDB != 0x00
[00:05:18][W][bl0942:082]: BL0942 invalid checksum! 0x61 != 0x4E
[00:05:18][W][bl0942:082]: BL0942 invalid checksum! 0x08 != 0x20
[00:05:19][D][uart_debug:114]: <<< 00 00 55 00 00 00 68 02 00 1C 50 00 00 00 00 00 00 00 20 4E 00 02 03 00 09 00 55 00 00 00 5D 02 00 F3 4C 00 00 00 00 00 00 00 20 4E 00 02 03 00 41 00 55 00 00 00 68 02 00 06 55 00 00 00 00 00 00 00 20 4E 00 02 03 00 1A 00 55 00 00 00 5A 02 00 02 54 00 00 00 00 00 00 00 20 4E 00 02 03 00 2D 00 55 00 00 00 68 02 00 5C 52 00 00 00 00 00 00 00 20
[00:05:19][W][bl0942:058]: Junk on wire. Throwing away partial message (5 bytes)
[00:05:19][D][uart_debug:114]: <<< 4E 00 02 03 00

Arduino uart log:
[23:59:35][D][uart_debug:114]: >>> 58 AA
[23:59:35][D][sensor:104]: 'Voltage1': Sending state 121.20000 V with 1 decimals of accuracy
[23:59:35][D][sensor:104]: 'Сurrent1': Sending state 0.00000 A with 2 decimals of accuracy
[23:59:35][D][sensor:104]: 'Power1': Sending state 0.00000 W with 1 decimals of accuracy
[23:59:35][D][sensor:104]: 'Meter1': Sending state 0.00000 Wh with 3 decimals of accuracy
[23:59:35][D][uart_debug:114]: <<< 55 00 00 00 5E 02 00 3C 51 00 00 00 00 00 00 00 20 4E 00 02 03 00 F2
[23:59:36][W][bl0942:058]: Junk on wire. Throwing away partial message (1 bytes)
[23:59:36][D][uart_debug:114]: <<< 00
[23:59:45][D][uart_debug:114]: >>> 58 AA
[23:59:45][D][sensor:104]: 'Voltage1': Sending state 123.40000 V with 1 decimals of accuracy
[23:59:45][D][sensor:104]: 'Сurrent1': Sending state 0.00000 A with 2 decimals of accuracy
[23:59:45][D][sensor:104]: 'Power1': Sending state 0.00000 W with 1 decimals of accuracy
[23:59:45][D][sensor:104]: 'Meter1': Sending state 0.00000 Wh with 3 decimals of accuracy
[23:59:45][D][uart_debug:114]: <<< 55 00 00 00 69 02 00 1A 51 00 00 00 00 00 00 00 20 4E 00 02 03 00 09
[23:59:45][W][bl0942:058]: Junk on wire. Throwing away partial message (1 bytes)
[23:59:46][D][uart_debug:114]: <<< 00
```

### Additional information

I checked with an oscilloscope - the uart requests and responses are coming in correctly, every 10 seconds. The problem is with the bl0942 component parsing the uart buffer

ei-ke, 2025-09-01:
I can confirm this issue which is still present in version 2025.8.2.

dwmw2, 2025-09-03:
Ack, away this week but will look when I'm home.

BaukF, 2025-09-05:
@ei-ke this [yaml-file](<script src="https://gist.github.com/BaukF/bbbf3717e86af278a66641ea00d4d32e.js"></script>) works for me (tm)

ei-ke, 2025-09-05:
In addition to the variant below I also tried dropping the baud rate to 4800 which doesn't cause an error but there is just no bl0942 any longer in the logs.

esphome.yaml
```
# Shelly Shelly 1PM Mini Gen3 - Model: S3SW-001P8EU - Ver1
substitutions:
  label: "Energie 37"
  # Wenn ausgegebener Wert zu niedrig ist muss dieser Wert verringert werden
  cur_ref: "251213.46469622"
  # Wenn ausgegebener Wert zu niedrig ist muss dieser Wert verringert werden
  vol_ref: "15873.35944299"

external_components:
  - source:
      type: git
      url: https://github.com/dwmw2/esphome
      ref: bl0942
    components: [ bl0942 ]
    
esphome:
  name: energie-37
  friendly_name: Energie 37

esp32:
  board: esp32-c3-devkitm-1
  flash_size: 8MB
  framework:
    type: esp-idf
    sdkconfig_options:
      COMPILER_OPTIMIZATION_SIZE: y

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: ***

ota:
  - platform: esphome
    password: ***

time:
  - platform: homeassistant
    id: homeassistant_time

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Energie-37"
    password: ***

captive_portal:

uart:
  id: uart_0
  tx_pin: GPIO6
  rx_pin: GPIO7
  baud_rate: 9600
  stop_bits: 1

light:
  - platform: status_led
    name: LED
    id: led
    internal: true
[... 385 lines cut ...]
[18:46:54][C][wifi_signal.sensor:010]:   State Class: 'measurement'
[18:46:54][C][wifi_signal.sensor:010]:   Unit of Measurement: 'dBm'
[18:46:54][C][wifi_signal.sensor:010]:   Accuracy Decimals: 0
[18:46:54][C][wifi_signal.sensor:010]:   Device Class: 'signal_strength'
[18:46:54][E][uart:015]: Reading from UART timed out at byte 2!
[18:46:54][W][bl0942:064]: Failed to read 23 available bytes from uart
[18:46:54][C][wifi_info:010]: IP Address 'IP Addresse'
[18:46:54][E][uart:015]: Reading from UART timed out at byte 2!
[18:46:54][W][bl0942:064]: Failed to read 23 available bytes from uart
[18:46:54][C][mdns:124]: mDNS:
[18:46:54][C][mdns:124]:   Hostname: energie-37
[18:46:54][E][uart:015]: Reading from UART timed out at byte 2!
[18:46:54][W][bl0942:064]: Failed to read 23 available bytes from uart
[18:46:54][E][uart:015]: Reading from UART timed out at byte 2!
[18:46:54][W][bl0942:064]: Failed to read 23 available bytes from uart
INFO Processing unexpected disconnect from ESPHome API for energie-37 @ 192.168.4.21
WARNING Disconnected from API
INFO Successfully resolved energie-37 @ 192.168.4.21 in 0.000s
WARNING Can't connect to ESPHome API for energie-37 @ 192.168.4.21: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.4.21', port=6053))]: [Errno 111] Connect call failed ('192.168.4.21', 6053) (SocketAPIError)
INFO Trying to connect to energie-37 @ 192.168.4.21 in the background
INFO Successfully resolved energie-37 @ 192.168.4.21 in 0.000s
INFO Successfully resolved energie-37 @ 192.168.4.21 in 0.000s
INFO Successfully resolved energie-37 @ 192.168.4.21 in 0.000s

```

dwmw2, 2025-09-05:
Oops, need to limit `rcv` to be no larger than `avail`

BaukF, 2025-09-05:
@ei-ke : I tried dropping to 4800 too, but then I got no input other than 58 AA. Going back to 9600 gave me back the intermittent working reads as I said earlier. And in my case, it is persistently working (removing debug lines disabled any output). My logs currently show:

<head></head>
21:35:09 | [D] | [uart_debug:114] | >>> 58 AA
-- | -- | -- | --
21:35:09 | [W] | [bl0942:082] | BL0942 invalid checksum! 0x0E != 0x00
21:35:09 | [D] | [sensor:103] | 'Voltage': Sending state 214.07201 V with 1 decimals of accuracy
21:35:09 | [D] | [sensor:103] | 'Current': Sending state 0.14878 A with 2 decimals of accuracy
21:35:09 | [D] | [sensor:103] | 'Power': Sending state 2.77013 W with 1 decimals of accuracy
21:35:09 | [D] | [sensor:103] | 'Energy': Sending state 2.06378 kWh with 3 decimals of accuracy
21:35:09 | [D] | [sensor:103] | 'Frequency': Sending state 49.92511 Hz with 2 decimals of accuracy
21:35:09 | [D] | [uart_debug:114] | <<< 00 55 D1 91 00 07 EE 33 62 81 00 7E 06 00 A4 1A 00 46 4E 00 00 01 00 0E 00 55 1D 92 00 2B E6 33 13 81 00 77 06 00 A4 1A 00 3E 4E 00 00 01 00 03 00 55 32 92 00 21 E9 33 72 80 00 7E 06 00 A4 1A 00 36 4E 00 00 01 00 98 00 55 FF 91 00 9A D9 33 00 7F 00 73 06 00 A4 1A 00 3E 4E 00 00 01 00 D9 00 55 88 91 00 01 D6 33 70 7E 00 6C 06 00 A4 1A 00 46 4E 00 00 01 00 7C
21:35:14 | [D] | [uart_debug:114] | >>> 58 AA
21:35:19 | [D] | [uart_debug:114] | >>> 58 AA
21:35:24 | [D] | [sensor:103] | 'temp_analog_reading': Sending state 0.90900 V with 2 decimals of accuracy
21:35:24 | [D] | [resistance:042] | 'temp_resistance_reading' - Resistance 3801.8Ω
21:35:24 | [D] | [sensor:103] | 'temp_resistance_reading': Sending state 3801.75659 Ω with 1 decimals of accuracy
21:35:24 | [D] | [ntc:026] | 'Temperature' - Temperature: 53.1°C
21:35:24 | [D] | [sensor:103] | 'Temperature': Sending state 53.07986 °C with 1 decimals of accuracy
21:35:24 | [D] | [uart_debug:114] | >>> 58 AA
21:35:29 | [D] | [uart_debug:114] | >>> 58 AA
21:35:34 | [D] | [uart_debug:114] | >>> 58 AA
21:35:34 | [W] | [bl0942:082] | BL0942 invalid checksum! 0x36 != 0x00
21:35:34 | [D] | [sensor:103] | 'Voltage': Sending state 214.54910 V with 1 decimals of accuracy
21:35:34 | [D] | [sensor:103] | 'Current': Sending state 0.14871 A with 2 decimals of accuracy
21:35:34 | [D] | [sensor:103] | 'Power': Sending state 2.78859 W with 1 decimals of accuracy
21:35:34 | [D] | [sensor:103] | 'Energy': Sending state 2.06378 kWh with 3 decimals of accuracy
21:35:34 | [D] | [sensor:103] | 'Frequency': Sending state 49.94506 Hz with 2 decimals of accuracy
21:35:34 | [D] | [uart_debug:114] | <<< 00 55 1A 91 00 FA B8 33 4B 81 00 6F 06 00 A4 1A 00 3E 4E 00 00 01 00 36 00 55 3D 91 00 A1 AD 33 6C 7F 00 8B 06 00 A4 1A 00 3E 4E 00 00 01 00 3C 00 55 9C 91 00 60 DB 33 D4 7C 00 7F 06 00 A4 1A 00 3E 4E 00 00 01 00 97 00 55 EE 91 00 2F F7 33 37 7D 00 7E 06 00 A4 1A 00 36 4E 00 00 01 00 FF 00 55 FB 91 00 7D E9 33 06 7C 00 76 06 00 A4 1A 00 3E 4E 00 00 01 00 E4
21:35:39 | [D] | [uart_debug:114] | >>> 58 AA

dwmw2, 2025-09-05:
Updated https://github.com/dwmw2/esphome/commits/bl0942/

dwmw2, 2025-09-05:
I probably messed up the resync part where it has to move the 0x55 header byte to the start. Can you add more logging to that? I'll be home next week and should be able to do some testing.

ei-ke, 2025-09-05:
Will do. Just with this config or any other additions you'd like to see?
```
    debug:
      direction: BOTH
      dummy_receiver: true
      after:
        delimiter: "\n"
      sequence:
        - lambda: UARTDebug::log_hex(direction, bytes, 32);
```

ei-ke, 2025-09-05:
I've nearly no programming knowledge (except for changing 1-2 lines until they work - trial and error) but I'll have a look at "what" is done in the .cpp and maybe find some spots where logging might help.

ei-ke, 2025-09-07:
I've attached approx 1h of logs, but the error seems to repeat itself. I can have a look if I'm able to attach a pullup resistor to the PCB.

[logs_energie-37_run.txt](https://github.com/user-attachments/files/22198614/logs_energie-37_run.txt)

ei-ke, 2025-09-08:
Thank you for your work and the hint with the dummy_receiver.
The latest commit does have a new issue:

```
In file included from src/esphome/components/uart/uart.h:6,
                 from src/esphome/components/bl0942/bl0942.h:5,
                 from src/esphome/components/bl0942/bl0942.cpp:1:
src/esphome/components/bl0942/bl0942.cpp: In member function 'virtual void esphome::bl0942::BL0942::setup()':
src/esphome/core/log.h:198:21: error: cannot convert 'const esphome::LogString*' to 'const char*'
  198 | #define LOG_STR(s) (reinterpret_cast<const LogString *>(s))
      |                    ~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                     |
      |                     const esphome::LogString*
src/esphome/components/bl0942/bl0942.cpp:187:30: note: in expansion of macro 'LOG_STR'
  187 |     this->status_set_warning(LOG_STR("BL0942 setup failed!"));
      |                              ^~~~~~~
In file included from src/esphome/components/bl0942/bl0942.h:3:
src/esphome/core/component.h:205:39: note:   initializing argument 1 of 'void esphome::Component::status_set_warning(const char*)'
  205 |   void status_set_warning(const char *message = nullptr);
      |                           ~~~~~~~~~~~~^~~~~~~~~~~~~~~~~
Compiling .pioenvs/energie-37/src/esphome/components/gpio/binary_sensor/gpio_binary_sensor.cpp.o
Compiling .pioenvs/energie-37/src/esphome/components/gpio/switch/gpio_switch.cpp.o
*** [.pioenvs/energie-37/src/esphome/components/bl0942/bl0942.cpp.o] Error 1
```

dwmw2, 2025-09-09:
Discussing this in [discord](https://discord.com/channels/429907082951524364/1414555402946871316/1414724774051053708) you might need https://github.com/esphome/esphome/pull/9248 to fix the pull up on the UART RX pin, which might make the noise go away.

BaukF, 2025-09-09:
> [@BaukF](https://github.com/BaukF) I'm fairly sure your data aren't strictly "repeated", just batched

Yes that was what I meant with separate datasets. I'll drop the term repeat ;-) I changed the update interval and it divides up the messages:

<code>[08:51:21][D][sensor:103]: 'Voltage': Sending state 211.13773 V with 1 decimals of accuracy
[08:51:21][D][sensor:103]: 'Current': Sending state 0.18980 A with 2 decimals of accuracy
[08:51:21][D][sensor:103]: 'Power': Sending state -8.46980 W with 1 decimals of accuracy
[08:51:21][D][sensor:103]: 'Energy': Sending state 0.05205 kWh with 3 decimals of accuracy
[08:51:21][D][sensor:103]: 'Frequency': Sending state 49.92511 Hz with 2 decimals of accuracy

[08:51:22][D][sensor:103]: 'Voltage': Sending state 211.10893 V with 1 decimals of accuracy
[08:51:22][D][sensor:103]: 'Current': Sending state 0.19497 A with 2 decimals of accuracy
[08:51:22][D][sensor:103]: 'Power': Sending state -9.89430 W with 1 decimals of accuracy
[08:51:22][D][sensor:103]: 'Energy': Sending state 0.05205 kWh with 3 decimals of accuracy
[08:51:22][D][sensor:103]: 'Frequency': Sending state 49.94506 Hz with 2 decimals of accuracy

[08:51:23][D][sensor:103]: 'Voltage': Sending state 211.27312 V with 1 decimals of accuracy
[08:51:23][D][sensor:103]: 'Current': Sending state 0.19440 A with 2 decimals of accuracy
[08:51:23][D][sensor:103]: 'Power': Sending state -9.81711 W with 1 decimals of accuracy
[08:51:23][D][sensor:103]: 'Energy': Sending state 0.05205 kWh with 3 decimals of accuracy
[08:51:23][D][sensor:103]: 'Frequency': Sending state 49.92511 Hz with 2 decimals of accuracy

[08:51:24][D][sensor:103]: 'Voltage': Sending state 211.39999 V with 1 decimals of accuracy
[08:51:24][D][sensor:103]: 'Current': Sending state 0.19381 A with 2 decimals of accuracy
[08:51:24][D][sensor:103]: 'Power': Sending state -9.19463 W with 1 decimals of accuracy
[08:51:24][D][sensor:103]: 'Energy': Sending state 0.05205 kWh with 3 decimals of accuracy
[08:51:24][D][sensor:103]: 'Frequency': Sending state 49.92511 Hz with 2 decimals of accuracy
[08:51:24][I][safe_mode:042]: Boot seems successful; resetting boot loop counter</code>

(I added linefeeds for better visibility)

Not sure whether we now have a state that is acceptable?

dwmw2, 2025-09-09:
What's your update interval? Still 5 seconds? You're getting four readings there in about 4 seconds. Does it then go quiet for 16 seconds, and then you get another batch of readings?

In my `bl0942-for-bauke` branch I had also made it receive only one packet at a time, deferring any more to the next loop() invocation. I've changed that back, in a [bl0942-for-bauke-2](https://github.com/dwmw2/esphome/commits/bl0942-for-bauke-2) branch which is now exactly the same as PR #10646 except for the `LOG_STR()` compile fix for older ESPHome. Do you get the whole batch of messages with the *same* timestamp now?

ei-ke, 2025-09-09:
[bl0942-for-bauke-2](https://github.com/dwmw2/esphome/commits/bl0942-for-bauke-2) Works fine for me with and without the debug options.

#9248 solves the junk on wire <3

daxy01, 2025-09-10:
I got my Shelly PM mini Gen3 yesterday and got it on esphome within minutes (of course) 😀 
I didn't get any bl0942 readings and then stumbled on this issue. With the "bl0942-for-bauke-2" I got readings so great great work!

However, I still get these right before every reading (every 5 secs):
```
[18:13:34][W][bl0942:157]: Requesting update
[18:13:34][W][bl0942:102]: Junk on wire. Throwing away 3 bytes
```
Today I applied https://github.com/esphome/esphome/pull/9248 but it did seem to work. Attached is the logs and (part) of my config at the bottom of this file:
[bl0492-9248-and-bauke-2.txt](https://github.com/user-attachments/files/22258663/bl0492-9248-and-bauke-2.txt)

Is there anything I missed or doing wrong?

daxy01, 2025-09-10:
Ah, that explains a lot 😀 Thanks!

This indeed resolves the junk messages, but I’m still getting the “Requesting update” one every 5 secs. Is that expected?

ei-ke, 2025-09-10:
I'm also getting the request message. I think it's just a FYI, but I could be wrong.

Matze2018, 2025-10-05:
Hi All,
None of the tipps solve my bl0942 junk on wire problems.

[08:00:17.972][W][bl0942:082]: BL0942 invalid checksum! 0xB1 != 0x00
[08:00:17.981][W][bl0942:082]: BL0942 invalid checksum! 0xAF != 0x00
[08:00:17.992][W][bl0942:082]: BL0942 invalid checksum! 0xAF != 0x00
[08:00:18.001][W][bl0942:082]: BL0942 invalid checksum! 0xAD != 0x00
[08:00:18.216][W][bl0942:058]: Junk on wire. Throwing away partial message (5 bytes)
[08:00:18.605][W][bl0942:082]: BL0942 invalid checksum! 0xAF != 0x00
[08:00:18.614][W][bl0942:082]: BL0942 invalid checksum! 0xB1 != 0x00
[08:00:18.624][W][bl0942:082]: BL0942 invalid checksum! 0x57 != 0x4E
[08:00:18.840][W][bl0942:058]: Junk on wire. Throwing away partial message (5 bytes)

that´s actual my uart config:
uart:
  id: uart_0
  tx_pin: GPIO6
  rx_pin: 
    number: GPIO7
    mode:
      pullup: true
      input: true
  baud_rate: 9600
  stop_bits: 1

maybe somone has a solution?

dwmw2, 2025-10-05:
> Hi All, None of the tipps solve my bl0942 junk on wire problems.

Please can you enable UART debug logs so we can see what it's sending?

Matze2018, 2025-10-05:
Hi, here the debug output:

[09:56:28.376][W][bl0942:082]: BL0942 invalid checksum! 0xB5 != 0x20
[09:56:28.485][D][uart_debug:114]: <<< 00:E0:00:55:E3:C9:00:8F:3B:33:37:7C:02:C5:15:00:01:00:00:36:4E:00:20
[09:56:28.583][W][bl0942:058]: Junk on wire. Throwing away partial message (3 bytes)
[09:56:28.689][D][uart_debug:114]: <<< 01:00:74
[09:56:33.368][D][uart_debug:114]: >>> 58:AA
[09:56:33.381][W][bl0942:082]: BL0942 invalid checksum! 0x7A != 0x20
[09:56:33.490][D][uart_debug:114]: <<< 00:E0:00:55:51:CA:00:00:4D:33:02:7D:02:52:15:00:01:00:00:26:4E:00:20
[09:56:33.588][W][bl0942:058]: Junk on wire. Throwing away partial message (3 bytes)
[09:56:33.693][D][uart_debug:114]: <<< 01:00:39
[09:56:38.356][D][uart_debug:114]: >>> 58:AA
[09:56:38.369][W][bl0942:082]: BL0942 invalid checksum! 0xD8 != 0x20
[09:56:38.476][D][uart_debug:114]: <<< 00:E2:00:55:3A:CA:00:AC:56:33:73:7B:02:DD:15:00:01:00:00:2E:4E:00:20
[09:56:38.575][W][bl0942:058]: Junk on wire. Throwing away partial message (3 bytes)

ThomDietrich, 2025-10-21:
#9248 fixes the otherwise non-working config at https://devices.esphome.io/devices/Shelly-PM-Mini-Gen3
Thanks!

geotibi, 2025-10-29:
I have the below config

```
bk72xx:
  board: cbu

uart:
  id: uart_bus
  tx_pin: GPIO6
  rx_pin: 
    number: GPIO7
    mode:
      pullup: true
      input: true
  baud_rate: 9600
  stop_bits: 1
  debug:
    direction: BOTH
    dummy_receiver: true
    after:
      delimiter: "\n"
    sequence:
      - lambda: UARTDebug::log_hex(direction, bytes, 32);
```

and in the debug i get

```
22:48:40
[D]
[uart_debug:114]
>>> 58 AA
22:49:39
[D]
[sensor:131]
'Uptime': Sending state 2041.43994 s with 0 decimals of accuracy
22:49:39
[D]
[sensor:131]
'EARU Internal Temperature': Sending state 39.34000 °C with 1 decimals of accuracy
22:49:40
[D]
[uart_debug:114]
>>> 58 AA
22:50:39
[D]
[sensor:131]
'Uptime': Sending state 2101.44189 s with 0 decimals of accuracy
22:50:39
[D]
[sensor:131]
'EARU Internal Temperature': Sending state 39.34000 °C with 1 decimals of accuracy
22:50:40
[D]
[uart_debug:114]
>>> 58 AA
22:50:59
[D]
[sensor:131]
'WiFi Signal': Sending state -37.00000 dBm with 0 decimals of accuracy
22:51:39
[D]
[sensor:131]
'Uptime': Sending state 2161.44409 s with 0 decimals of accuracy
22:51:39
[D]
[sensor:131]
'EARU Internal Temperature': Sending state 39.34000 °C with 1 decimals of accuracy
22:51:40
[D]
[uart_debug:114]
>>> 58 AA
22:52:39
[D]
[sensor:131]
'Uptime': Sending state 2221.44604 s with 0 decimals of accuracy
22:52:39
[D]
[sensor:131]
'EARU Internal Temperature': Sending state 39.34000 °C with 1 decimals of accuracy
22:52:40
[D]
[uart_debug:114]
>>> 58 AA
22:53:39
[D]
[sensor:131]
'Uptime': Sending state 2281.44800 s with 0 decimals of accuracy
22:53:39
[D]
[sensor:131]
'EARU Internal Temperature': Sending state 39.34000 °C with 1 decimals of accuracy
22:53:40
[D]
[uart_debug:114]
>>> 58 AA
22:54:39
[D]
[sensor:131]
'Uptime': Sending state 2341.44995 s with 0 decimals of accuracy
22:54:39
[D]
[sensor:131]
'EARU Internal Temperature': Sending state 39.34000 °C with 1 decimals of accuracy
22:54:40
[D]
[uart_debug:114]
>>> 58 AA
22:55:39
[D]
[sensor:131]
'Uptime': Sending state 2401.45190 s with 0 decimals of accuracy
22:55:39
[D]
[sensor:131]
'EARU Internal Temperature': Sending state 39.34000 °C with 1 decimals of accuracy
22:55:40
[D]
[uart_debug:114]
>>> 58 AA
22:55:59
[D]
[sensor:131]
'WiFi Signal': Sending state -36.00000 dBm with 0 decimals of accuracy
22:56:39
[D]
[sensor:131]
'Uptime': Sending state 2461.45410 s with 0 decimals of accuracy
22:56:39
[D]
[sensor:131]
'EARU Internal Temperature': Sending state 39.34000 °C with 1 decimals of accuracy
22:56:40
[D]
[uart_debug:114]
>>> 58 AA
22:57:39
[D]
[sensor:131]
'Uptime': Sending state 2521.45605 s with 0 decimals of accuracy
22:57:39
[D]
[sensor:131]
'EARU Internal Temperature': Sending state 39.34000 °C with 1 decimals of accuracy
22:57:40
[D]
[uart_debug:114]
>>> 58 AA
22:58:39
[D]
[sensor:131]
'Uptime': Sending state 2581.45801 s with 0 decimals of accuracy
22:58:39
[D]
[sensor:131]
'EARU Internal Temperature': Sending state 39.34000 °C with 1 decimals of accuracy
22:58:40
[D]
[uart_debug:114]
>>> 58 AA
22:59:39
[D]
[sensor:131]
'Uptime': Sending state 2641.45801 s with 0 decimals of accuracy
22:59:39
[D]
[sensor:131]
'EARU Internal Temperature': Sending state 39.34000 °C with 1 decimals of accuracy
22:59:40
[D]
[uart_debug:114]
>>> 58 AA
23:00:39
[D]
[sensor:131]
'Uptime': Sending state 2701.46191 s with 0 decimals of accuracy
23:00:39
[D]
[sensor:131]
'EARU Internal Temperature': Sending state 39.34000 °C with 1 decimals of accuracy
23:00:40
[D]
[uart_debug:114]
>>> 58 AA
23:01:00
[D]
[sensor:131]
'WiFi Signal': Sending state -37.00000 dBm with 0 decimals of accuracy
```

What I'm doing wrong because Current, Frequency, Voltage ... are NA

532910, 2025-11-13:
only pullup is enough for my PMMini, rx_full_threshold is not required

geotibi, 2025-11-26:
This is only for ESP32 and not for bl0942

    This feature is only available on [<Platform.ESP32: 'esp32'>].
    rx_full_threshold: 23