esphome/esphome #10151: Incorrect config validation when using ADE7880 power sensor

JS3910, 2025-08-09:
### The problem

The config validation in this case reports a duplicate sensor and refuses to compile. This is not correct since the name of the phase is prefixed to all sensors in each phase. This worked before (earlier this year) but after updating to 2025.7.5 I can't update my power sensor.

### Which version of ESPHome has the issue?

2025.7.5

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Component causing the issue

ade7880

### YAML Config

```yaml
substitutions:
  device_name: "stove-p"
  friendly_name: "Stove"
  current_gain_a: "3151169"
  current_gain_b: "3200051"
  voltage_gain_a: "-773412"
  voltage_gain_b: "-767198"
  power_gain_a: "-1345512"
  power_gain_b: "-1341812"
  phase_angle_a: "172"
  phase_angle_b: "168"

esphome:
  name: ${device_name}
  comment: "Shelly 3EM"
  friendly_name: ${friendly_name}
  name_add_mac_suffix: False
  project:
    name: "Shelly.3EM"
    version: "3EM"

esp8266:
  board: esp12e

i2c:
  sda: GPIO12
  scl: GPIO14
  frequency: 200kHz

sensor:
  - id: ade7880_device
    platform: ade7880
    update_interval: 5s
    irq0_pin:
      number: GPIO13
    irq1_pin:
      number: GPIO5
    reset_pin:
      number: GPIO16
    frequency: 50Hz
    phase_a:
      name: "L1"
      voltage:
        name: "Voltage"
        icon: mdi:alpha-v
        unit_of_measurement: V
        accuracy_decimals: 1
      calibration:
        current_gain: ${current_gain_a}
        voltage_gain: ${voltage_gain_a}
        power_gain: ${power_gain_a}
        phase_angle: ${phase_angle_a}
    phase_b:
      name: "L2"
      voltage:
        name: "Voltage"
        icon: mdi:alpha-v
        unit_of_measurement: V
        accuracy_decimals: 1
      calibration:
        current_gain: ${current_gain_b}
        voltage_gain: ${voltage_gain_b}
        power_gain: ${power_gain_b}
        phase_angle: ${phase_angle_b}
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2025.7.5
INFO Reading configuration /root/config/stove-power.yaml...
Failed config

sensor.ade7880: [source /root/config/stove-power.yaml:31]
  id: ade7880_device
  platform: ade7880
  update_interval: 5s
  irq0_pin: 
    number: GPIO13
  irq1_pin: 
    number: GPIO5
  reset_pin: 
    number: GPIO16
  frequency: 50Hz
  phase_a: 
    name: L1
    voltage: 
      name: Voltage
      icon: mdi:alpha-v
      unit_of_measurement: V
      accuracy_decimals: 1
    calibration: 
      current_gain: 3151169
      voltage_gain: -773412
      power_gain: -1345512
      phase_angle: 172
  phase_b: 
    name: L2
    
    Duplicate sensor entity with name 'Voltage' found. Each entity must have a unique name within its platform across all devices.
    voltage: 
      name: Voltage
      icon: mdi:alpha-v
      unit_of_measurement: V
      accuracy_decimals: 1
    calibration: 
      current_gain: 3200051
      voltage_gain: -767198
      power_gain: -1341812
      phase_angle: 168
```

edwardtfn, 2025-08-10:
This duplicated name check came with 2025.7.0.
Please take a look at this thread on Discord: https://discord.com/channels/429907082951524364/1392599593312718948

bdraco, 2025-08-10:
The `final_validate` function in that component modifies the configuration in-place after the initial validation has already occurred. The sensor names need to be properly prefixed during the initial configuration schema validation, not in a final validation step.

bdraco, 2025-08-10:
https://github.com/esphome/esphome/pull/10155#issuecomment-3172384912

JS3910, 2025-08-10:
Working, thank you and good job! <3

PR #10155 (2025-08-10): [ade7880] Fix duplicate sensor name validation error
# What does this implement/fix?

This PR fixes a configuration validation issue with the ADE7880 component where sensor names were being modified after the duplicate entity validation had already run, causing false positive "Duplicate sensor entity" errors.

The component's `final_validate` function was prefixing sensor names with channel names (e.g., "L1 Voltage", "L2 Voltage") too late in the validation process. This fix moves the name prefixing to occur during schema validation, before the duplicate check runs.

## Types of changes

- [x] [... cut]
closed 2025-08-10 by JS3910