esphome/esphome #10159: Duplicate sensor entity with name 'Battery' found.

andrewjswan, 2025-08-10:
### The problem

Duplicate sensor entity with name 'Battery' found. Each entity must have a unique name within its platform across all devices.
But at the same time, `each sensor` belongs to its `own device` and does not intersect.

### Which version of ESPHome has the issue?

2025.7.4

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

esphome/devices

### YAML Config

```yaml
esphome:
  devices:
    - id: black_find
      name: "NUT Black"
    - id: golden_find
      name: "NUT Golden"

sensor:
  - platform: template
    id: black_nut_battery_level
    name: Battery
    device_class: battery
    unit_of_measurement: '%'
    state_class: measurement
    accuracy_decimals: 0
    update_interval: ${update_interval}
    device_id: black_find

  - platform: template
    id: golden_nut_battery_level
    name: Battery
    device_class: battery
    unit_of_measurement: '%'
    state_class: measurement
    accuracy_decimals: 0
    update_interval: ${update_interval}
    device_id: golden_find
```

### Anything in the logs that might be useful for us?

```txt
Duplicate sensor entity with name 'Battery' found. Each entity must have a unique name within its platform across all devices.
  platform: template
  id: golden_nut_battery_level
  name: Battery
  device_class: battery
  unit_of_measurement: %
  state_class: measurement
  accuracy_decimals: 0
  update_interval: 60s
  device_id: golden_find
```

bdraco, 2025-08-20:
> I can understand why the ID has to be unique, but why does ESPHome enforce uniqueness of names? Surely that is my problem if I use the same names and confuse myself.

Unique names are required because thats how all the object id hashes are generated. core functionality depends on object_id hashes as the `get_object_id_hash()` function is used throughout ESPHome for:

- Persistent storage keys (saving/restoring entity states)
- API message keys for entity identification
- Entity lookups and references

DAVe3283, 2025-08-20:
Not being able to use duplicate object names for different sub-devices materially reduces the usefulness of the feature, at least for my use case. I have an ESP32 that monitors various stats on different houseplants.

Previously, I avoided using the device name feature, so the names were all set explicitly (`Neon Pothos Soil Moisture`, `Amaryllis Soil Moisture`, etc.). With the update, I tried to create multiple sub-devices (`Neon Pothos`, `Amaryllis`, etc.) but that meant all the sensor names became `Soil Moisture`, and I too hit this issue.

Could the internal hash could be calculated from device name (if present) + object name? Because having all the plants broken out to their own device + area really would be slick.

bdraco, 2025-08-20:
Names need to be unique for each device_id.  The validation currently allows one unique name per device_id.

Although there are still some places that don't know about device_id that will not work correctly https://github.com/esphome/esphome/issues/10331

DAVe3283, 2025-08-20:
Hmm, I might have done something wrong under 2025.7.x, because I just tried again with 2025.8.0 and it seems to have worked as expected! At least for the b_parasite component.

RFDarter, 2025-08-21:
> Unique names are required

When did this get changed?
So names are forced to me unique?
So https://github.com/esphome/esphome-webserver/issues/122 should be fixed.