esphome/esphome #10170: Olimex ESP32-POE-ISO-EA-IND fails to properly initialize Ethernet

ggiesen, 2025-08-11:
### The problem

I just acquired a couple of Olimex [ESP32-POE-ISO-EA-IND](https://www.olimex.com/Products/IoT/ESP32/ESP32-POE-ISO/open-source-hardware) Revision M boards for use as bluetooth proxies. I flashed using the Bluetooth ready-made project at https://esphome.io/projects/.

Everything flashes and installs fine, but when the board boots, I see link on the Ethernet port (green LED solid and yellow LED blinking), but my switch shows no MAC addresses learned on the port. Board never receives a DHCP IP address, and repeats `[W][ethernet:265]: Connecting via ethernet failed! Re-connecting...` over and over in the logs. The port on my switch is a known good port which I've plugged another ESPHome device (ESP32-POE-ISO-IND) and works perfectly.

Logs from the device:

```
[19:40:50]ets Jul 29 2019 12:21:46
[19:40:50]
[19:40:50]rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
[19:40:50]configsip: 0, SPIWP:0xee
[19:40:50]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[19:40:50]mode:DIO, clock div:2
[19:40:50]load:0x3fff0030,len:7112
[19:40:50]load:0x40078000,len:15688
[19:40:50]load:0x40080400,len:4
[19:40:50]load:0x40080404,len:3876
[19:40:50]entry 0x4008064c
[19:40:50]I (29) boot: ESP-IDF 5.1.6 2nd stage bootloader
[19:40:50]I (29) boot: compile time May 21 2025 23:49:04
[19:40:50]I (29) boot: Multicore bootloader
[19:40:50]I (33) boot: chip revision: v3.1
[19:40:50]I (37) boot.esp32: SPI Speed      : 40MHz
[19:40:50]I (42) boot.esp32: SPI Mode       : DIO
[19:40:50]I (46) boot.esp32: SPI Flash Size : 4MB
[19:40:50]I (51) boot: Enabling RNG early entropy source...
[19:40:50]I (56) boot: Partition Table:
[19:40:50]I (60) boot: ## Label            Usage          Type ST Offset   Length
[19:40:50]I (67) boot:  0 otadata          OTA data         01 00 00009000 00002000
[19:40:50]I (75) boot:  1 phy_init         RF data          01 01 0000b000 00001000
[19:40:50]I (82) boot:  2 app0             OTA app          00 10 00010000 001c0000
[19:40:50]I (90) boot:  3 app1             OTA app          00 11 001d0000 001c0000
[19:40:50]I (97) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
[19:40:50]I (105) boot: End of partition table
[19:40:50]I (109) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=4699ch (289180) map
[19:40:50]I (222) esp_image: segment 1: paddr=000569c4 vaddr=3ffbdb60 size=04870h ( 18544) load
[19:40:50]I (229) esp_image: segment 2: paddr=0005b23c vaddr=40080000 size=04ddch ( 19932) load
[19:40:50]I (238) esp_image: segment 3: paddr=00060020 vaddr=400d0020 size=cef38h (847672) map
[19:40:51]I (544) esp_image: segment 4: paddr=0012ef60 vaddr=40084ddc size=117d0h ( 71632) load
[19:40:51]I (586) boot: Loaded app from partition at offset 0x10000
[19:40:51]I (586) boot: Disabling RNG early entropy source...
[19:40:51]I (597) cpu_start: Multicore app
[19:40:51]I (598) cpu_start: Pro cpu up.
[19:40:51]I (598) cpu_start: Starting app cpu, entry point is 0x40082380
[19:40:51]I (0) cpu_start: App cpu up.
[19:40:51]I (616) cpu_start: Pro cpu start user code
[19:40:51]I (616) cpu_start: cpu freq: 160000000 Hz
[19:40:51]I (616) cpu_start: Application information:
[19:40:51]I (620) cpu_start: Project name:     olimex-esp32-poe-iso
[19:40:51]I (626) cpu_start: App version:      2025.5.0
[19:40:51]I (631) cpu_start: Compile time:     May 21 2025 23:48:28
[19:40:51]I (638) cpu_start: ELF file SHA256:  8d92b2a440c76cec...
[19:40:51]I (644) cpu_start: ESP-IDF:          5.1.6
[19:40:51]I (648) cpu_start: Min chip rev:     v0.0
[19:40:51]I (653) cpu_start: Max chip rev:     v3.99 
[19:40:51]I (658) cpu_start: Chip rev:         v3.1
[19:40:51]I (663) heap_init: Initializing. RAM available for dynamic allocation:
[19:40:51]I (670) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
[19:40:51]I (676) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
[19:40:51]I (682) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
[19:40:51]I (688) heap_init: At 3FFC87F8 len 00017808 (94 KiB): DRAM
[19:40:51]I (694) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
[19:40:51]I (701) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
[19:40:51]I (707) heap_init: At 400965AC len 00009A54 (38 KiB): IRAM
[19:40:51]I (715) spi_flash: detected chip: generic
[19:40:51]I (718) spi_flash: flash io: dio
[19:40:51]I (723) coexist: coex firmware version: 994b23e
[19:40:51]I (728) app_start: Starting scheduler on CPU0
[19:40:51]0;32mI (732) app_start: Starting scheduler on CPU1
[19:40:51]mI (732) main_task: Started on CPU0
[19:40:51]I (741) main_task: Calling app_main()
[19:40:51]I (900) mai[I][logger:171]: Log initialized
[19:40:51][C][safe_mode:079]: There have been 2 suspected unsuccessful boot attempts
[19:40:51][D][esp32.preferences:114]: Saving 1 preferences to flash...
[19:40:51][D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[19:40:51][I][app:029]: Running through setup()...
[19:40:51][C][esp32_ble:031]: Setting up BLE...
[19:40:51][C][ethernet:041]: Setting up Ethernet...
[19:40:51][D][esp-idf:000]: I (1251) esp_eth.netif.netif_glue: 08:d1:f9:dd:ee:f7
[19:40:51][D][esp-idf:000]: I (1252) esp_eth.netif.netif_glue: ethernet attached to netif
[19:40:53][D][esp32_ble:288]: Enabling BLE...
[19:40:53][D][esp-idf:000]: I (2755) BTDM_INIT: BT controller compile version [f7360be]
[19:40:53][D][esp-idf:000]: I (2759) phy_init: phy_version 4840,02e0d70,Sep  2 2024,19:39:07
[19:40:53][D][esp-idf:000]: I (2829) phy_init: Saving new calibration data due to checksum failure or outdated calibration data, mode(0)
[19:40:53][D][esp32_ble_tracker:295]: Starting scan, set scanner state to STARTING.
[19:40:53][I][ethernet:248]: Starting ethernet connection
[19:40:53][W][component:167]: Component ethernet set Warning flag: waiting for IP configuration
[19:40:53][D][esp-idf:000][btController]: I (2756) BTDM_INIT: Bluetooth MAC: 08:d1:f9:dd:ee:f6
[19:41:08][W][ethernet:265]: Connecting via ethernet failed! Re-connecting...
[19:41:23][W][ethernet:265]: Connecting via ethernet failed! Re-connecting...
[19:41:38][W][ethernet:265]: Connecting via ethernet failed! Re-connecting...
[19:41:53][W][ethernet:265]: Connecting via ethernet failed! Re-connecting...
```

### Which version of ESPHome has the issue?

2025.5.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

Ethernet

### YAML Config

```yaml
esphome:
  name: olimex-esp32-poe-iso
  friendly_name: Bluetooth Proxy
  min_version: 2025.5.0
  name_add_mac_suffix: true

esp32:
  board: esp32-poe-iso
  framework:
    type: esp-idf

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO17_OUT
  phy_addr: 0
  power_pin:
    number: GPIO12
    ignore_strapping_warning: true

api:
logger:
ota:
  - platform: esphome
    id: ota_esphome

esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

bluetooth_proxy:
  active: true

button:
  - platform: safe_mode
    id: button_safe_mode
    name: Safe Mode Boot

  - platform: factory_reset
    id: factory_reset_btn
    name: Factory reset
```

### Anything in the logs that might be useful for us?

```txt
[19:40:51][C][ethernet:041]: Setting up Ethernet...
[19:40:51][D][esp-idf:000]: I (1251) esp_eth.netif.netif_glue: 08:d1:f9:dd:ee:f7
[19:40:51][D][esp-idf:000]: I (1252) esp_eth.netif.netif_glue: ethernet attached to netif
[19:40:53][I][ethernet:248]: Starting ethernet connection
[19:40:53][W][component:167]: Component ethernet set Warning flag: waiting for IP configuration
[19:41:08][W][ethernet:265]: Connecting via ethernet failed! Re-connecting...
[19:41:23][W][ethernet:265]: Connecting via ethernet failed! Re-connecting...
[19:41:38][W][ethernet:265]: Connecting via ethernet failed! Re-connecting...
[19:41:53][W][ethernet:265]: Connecting via ethernet failed! Re-connecting...
```

ggiesen, 2025-08-11:
I if I add:

```
esphome:
  platformio_options:
    board_build.flash_mode: dio
```

to the config in https://github.com/esphome/esphome/issues/10170#issuecomment-3173376785, then the boot loop stops but I'm back to broken Ethernet.

bdraco, 2025-08-11:
Is it the WROOM chip or the WROVER chip.  The ready made projects only work with WROOM

ggiesen, 2025-08-11:
> Is it the WROOM chip or the WROVER chip?  The ready made projects only work with WROOM. The pins are different for ethernet with WROVER, and listed in the docs on the Olimex site

WROOM

bdraco, 2025-08-13:
Can you post the sdkconfig.* files from `.esphome/build/HOSTNAME`?

ggiesen, 2025-08-14:
> Here is an older config I have for a working one.
> 
> substitutions:
>   devicename: "upstairsdesk89proxy"
>   friendly_name: "upstairsdesk89proxy"
>   upper_devicename: upstairsdesk89proxy
> 
> esphome:
>   name: $devicename
>   friendly_name: $friendly_name
> 
> esp32:
>   board: esp32-poe-iso
>   framework:
>     type: esp-idf
> 
> esp32_ble_tracker:
>   scan_parameters:
>     active: false
> bluetooth_proxy:
>   active: true
> 
>     
> ethernet:
>   type: LAN8720
>   mdc_pin: GPIO23
>   mdio_pin: GPIO18
>   clk_mode: GPIO17_OUT
>   phy_addr: 0
>   power_pin: GPIO12
> 
> 
> logger:
> 
> 
> api:
> 
> ota:
>   platform: esphome
>   
> 
> 
> button:
>   - platform: restart
>     name: Restart
> 
> 
> time:
>   - platform: homeassistant
>     id: homeassistant_time
>   - platform: sntp
>     id: sntp_time

So it looks like changing:

```
esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms

bluetooth_proxy:
  active: true
  connection_slots: 3
```

to your config:

```
esp32_ble_tracker:
  scan_parameters:
    active: false

bluetooth_proxy:
  active: true
```

is enough to make Ethernet working/responsive again.

nnae06, 2025-08-14:
I tried with no success. Still the same problem. Ethernet crashes. Must be reconfigured with Usb.

stromnet, 2025-08-14:
I found [another similar issue](https://community.home-assistant.io/t/m5-atom-echo-voice-assistant-with-bluetooth-proxy/617596/3) which hinted that a clean build might help. Indeed, this build had some other components (wifi) from my initial esp32 experiments running "wizard". With a `esphome clean` and then a `run` it seems to work with bluetooth components enabled, and IP is received :tada:

Perhaps this can help original poster as well?

Small note for the bluetooth proxy project config, it seems to use deprecated config:
```
WARNING [ethernet] The 'clk_mode' option is deprecated and will be removed in ESPHome 2026.1. Please update your configuration to use 'clk' instead.
```