esphome/esphome #10194: Humidity is always 1.00%

ksaye, 2025-08-11:
### The problem

Using the board nodemcuv2, Humidity is always shown as 1.00%.  Using the simple config below, the temperature is correct BUT the humidity is always 1.00%

I have 3 boards with the same DHT22 / AM2302 and all show the same issue.

### Which version of ESPHome has the issue?

ESPHome version 2025.7.4 compiled on Aug  4 2025, 06:25:52

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Component causing the issue

DHT22

### YAML Config

```yaml
esphome:
  name: esphome-web-e062f6
  friendly_name: "Shop"

esp8266:
  board: nodemcuv2
  framework:
    version: recommended

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "XXXXXXXXXXXXX"

ota:
  - platform: esphome
    password: "XXXXXXXXXXXXXXX"

binary_sensor:   
  - platform: status
    name: "Sensor Status"

switch:
  - platform: restart
    name: "Restart"

wifi:
  ssid: "iotdevices"
  password: "XXXXXXXXXXXXXXXX"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome-Web-E062F6"
    password: "XXXXXXXXXXXXXX"

captive_portal:

sensor:
  - platform: dht
    pin: D2
    model: DHT22
    temperature:
      name: "Temperature"
    humidity: 
      name: "Humidity"
      accuracy_decimals: 2
    update_interval: 60s

  - platform: wifi_signal
    name: "WiFi Signal Strength"
    update_interval: 60s
  
  - platform: uptime
    name: "Uptime"
```

### Anything in the logs that might be useful for us?

```txt
Logging:

INFO ESPHome 2025.7.5
INFO Reading configuration /config/esphome/esphome-web-e062f6.yaml...
INFO Starting log output from 192.168.15.159 using esphome API
INFO Successfully resolved esphome-web-e062f6 @ 192.168.15.159 in 0.000s
INFO Successfully connected to esphome-web-e062f6 @ 192.168.15.159 in 0.003s
INFO Successful handshake with esphome-web-e062f6 @ 192.168.15.159 in 2.369s
[15:49:18][I][app:149]: ESPHome version 2025.7.4 compiled on Aug  4 2025, 06:25:52
[15:49:18][C][wifi:613]: WiFi:
[15:49:18][C][wifi:434]:   Local MAC: 18:FE:34:E0:62:F6
[15:49:18][C][wifi:439]:   SSID: [redacted]
[15:49:18][C][wifi:442]:   IP Address: 192.168.15.159
[15:49:18][C][wifi:446]:   BSSID: [redacted]
[15:49:18][C][wifi:446]:   Hostname: 'esphome-web-e062f6'
[15:49:18][C][wifi:446]:   Signal strength: -55 dB ▂▄▆█
[15:49:18][C][wifi:455]:   Channel: 11
[15:49:18][C][wifi:455]:   Subnet: 255.255.255.0
[15:49:18][C][wifi:455]:   Gateway: 192.168.15.1
[15:49:18][C][wifi:455]:   DNS1: 192.168.15.1
[15:49:18][C][wifi:455]:   DNS2: 208.67.222.222
[15:49:18][C][logger:246]: Logger:
[15:49:18][C][logger:246]:   Max Level: DEBUG
[15:49:18][C][logger:246]:   Initial Level: DEBUG
[15:49:18][C][logger:252]:   Log Baud Rate: 115200
[15:49:18][C][logger:252]:   Hardware UART: UART0
[15:49:18][C][uptime.sensor:033]: Uptime Sensor 'Uptime'
[15:49:18][C][uptime.sensor:033]:   State Class: 'total_increasing'
[15:49:18][C][uptime.sensor:033]:   Unit of Measurement: 's'
[15:49:18][C][uptime.sensor:033]:   Accuracy Decimals: 0
[15:49:18][C][uptime.sensor:033]:   Device Class: 'duration'
[15:49:18][C][uptime.sensor:033]:   Icon: 'mdi:timer-outline'
[15:49:18][C][uptime.sensor:034]:   Type: Seconds
[15:49:18][C][status:034]: Status Binary Sensor 'Sensor Status'
[15:49:18][C][status:034]:   Device Class: 'connectivity'
[15:49:18][C][restart:079]: Restart Switch 'Restart'
[15:49:18][C][restart:079]:   Restore Mode: always OFF
[15:49:18][C][restart:087]:   Icon: 'mdi:restart'
[15:49:18][C][dht:018]: DHT:
[15:49:18][C][dht:019]:   Pin: GPIO4
[15:49:18][C][dht:020]:   Model: DHT22 or equivalent
[15:49:18][C][dht:022]:   Internal pull-up: ON
[15:49:18][C][dht:023]:   Update Interval: 60.0s
[15:49:18][C][dht:024]:   Temperature 'Temperature'
[15:49:18][C][dht:024]:     State Class: 'measurement'
[15:49:18][C][dht:024]:     Unit of Measurement: '°C'
[15:49:18][C][dht:024]:     Accuracy Decimals: 1
[15:49:18][C][dht:024]:     Device Class: 'temperature'
[15:49:18][C][dht:025]:   Humidity 'Humidity'
[15:49:18][C][dht:025]:     State Class: 'measurement'
[15:49:18][C][dht:025]:     Unit of Measurement: '%'
[15:49:18][C][dht:025]:     Accuracy Decimals: 2
[15:49:18][C][dht:025]:     Device Class: 'humidity'
[15:49:18][C][captive_portal:099]: Captive Portal:
[15:49:18][C][esphome.ota:073]: Over-The-Air updates:
[15:49:18][C][esphome.ota:073]:   Address: esphome-web-e062f6.local:8266
[15:49:18][C][esphome.ota:073]:   Version: 2
[15:49:18][C][esphome.ota:080]:   Password configured
[15:49:18][C][safe_mode:018]: Safe Mode:
[15:49:18][C][safe_mode:019]:   Boot considered successful after 60 seconds
[15:49:18][C][safe_mode:019]:   Invoke after 10 boot attempts
[15:49:18][C][safe_mode:019]:   Remain for 300 seconds
[15:49:18][C][web_server.ota:224]: Web Server OTA
[15:49:18][C][api:207]: API Server:
[15:49:18][C][api:207]:   Address: esphome-web-e062f6.local:6053
[15:49:18][C][api:212]:   Using noise encryption: YES
[15:49:18][C][wifi_signal.sensor:010]: WiFi Signal 'WiFi Signal Strength'
[15:49:18][C][wifi_signal.sensor:010]:   State Class: 'measurement'
[15:49:18][C][wifi_signal.sensor:010]:   Unit of Measurement: 'dBm'
[15:49:18][C][wifi_signal.sensor:010]:   Accuracy Decimals: 0
[15:49:18][C][wifi_signal.sensor:010]:   Device Class: 'signal_strength'
[15:49:18][C][mdns:122]: mDNS:
[15:49:18][C][mdns:122]:   Hostname: esphome-web-e062f6
[15:49:57][D][dht:043]: Temperature 38.5°C Humidity 1.0%
[15:49:57][D][sensor:104]: 'Temperature': Sending state 38.50000 °C with 1 decimals of accuracy
[15:49:57][D][sensor:104]: 'Humidity': Sending state 1.00000 % with 2 decimals of accuracy
[15:50:07][D][sensor:104]: 'Uptime': Sending state 638629.43750 s with 0 decimals of accuracy
[15:50:10][D][sensor:104]: 'WiFi Signal Strength': Sending state -58.00000 dBm with 0 decimals of accuracy
[15:50:57][D][dht:043]: Temperature 38.6°C Humidity 1.0%
[15:50:57][D][sensor:104]: 'Temperature': Sending state 38.60000 °C with 1 decimals of accuracy
[15:50:57][D][sensor:104]: 'Humidity': Sending state 1.00000 % with 2 decimals of accuracy
```

randybb, 2025-08-12:
Have you tested it with something else? Like with some arduino code or with tasmota?
Had some DHT22 ~10 years ago connected to stm32, but all died - temperature was quite ok, but humidity reporting had similar problems. But this is a known issue so I guess you have some proof that your sensor is not broken.

ksaye, 2025-08-13:
Great suggestion.

Let me test some simple Arduino code.  ETA is by end of week.

You are receiving this email because you authored the thread.

clydebarrow, 2025-08-13:
Try removing the `model:` setting and let it auto-detect.