esphome/esphome #10202: ESP32-C3, LD2420 suspect UART problem

rabackes, 2025-08-12:
### The problem

Build some ESP32-C3 based presence sensors. Hardware is a ESP-C3 SuperMini board and a LD2420. Thel LD2420 is connected to GPIO20 and 21. 
The first firmware was created with ESPhome (on HomeAssistant) 2025.6.1 and 2025.603 and is still working on 2 devices.
After upgrading to ESPhome 2025.7.x that setup no longer works - it seems the C3 has problems reading die output of the LD2420. First noticed that with a newly soldered device. To make sure, there is no hardware issue, I 'upgraded' a working device to 2025.7.3 - and it stopped working with the newer version.
Using the nearly the same YAML with a ESP D1Mini32 (ESP32 WROOM) - only adjusting the platform and the GPIOs works flawless with the SAME LD2420.

Here the YAML:

### Which version of ESPHome has the issue?

2025.7.x

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

ld2024 or uart

### YAML Config

```yaml
`substitutions:
  devicename: ba-ba-pres
  location: Backes Ersatzbad
  friendly_name: "BA Toi Presence"
  ip_address: 172.16.20.130
  ip_gateway: 172.16.20.1
  ip_subnet: 255.255.255.0
  ip_dns1: 172.16.20.1
  
  
  project_name: "rb.presence"
  project_version: "1.0"

<<: !include include/ESP32-C3-SuperMini.yaml

# ld2420 Radar Presense
uart:
  id: ld2420_uart
  tx_pin: GPIO20  
  rx_pin: GPIO21
  baud_rate: 115200
  parity: NONE
  stop_bits: 1
  
# Example configuration entry
ld2420:

select:
  - platform: ld2420
    operating_mode:
      name: Operating Mode

button:
  - platform: ld2420
    apply_config:
      name: "${friendly_name} Apply Config"
    factory_reset:
      name: "${friendly_name} Factory Reset"
    restart_module:
      name: "${friendly_name} Restart Module"
    revert_config:
      name: "${friendly_name} Undo Edits"
[... 272 lines cut ...]
[C][api:217]:   Using noise encryption: NO
[C][wifi_signal.sensor:010]: WiFi Signal 'WiFi Signal dB'
                                                           State Class: 'measurement'
                                                                                       Unit of Measurement: 'dBm'
                                                                                                                   Accuracy Decimals: 0
[C][wifi_signal.sensor:010]:   Device Class: 'signal_strength'
[C][wifi_info:012]: SSID 'ESP Connected SSID'
[C][wifi_info:013]: BSSID 'ESP Connected BSSID'
[C][wifi_info:010]: IP Address 'ESP IP Address'
[C][mdns:122]: mDNS:
                      Hostname: ke-la-pres
[D][text_sensor:069]: 'ESP IP Address': Sending state '172.16.20.131'
[D][api:146]: Accept 172.16.129.200
[W][component:307]: api cleared Warning flag
[D][main:285]: API is connected!
[D][api.connection:1466]: Home Assistant 2025.8.1 (172.16.129.200) connected
[D][text_sensor:069]: 'ESP Connected BSSID': Sending state '6E:D7:9A:CA:D0:31'
[D][sensor:104]: 'WiFi Signal dB': Sending state -57.00000 dBm with 0 decimals of accuracy
[D][sensor:104]: 'WiFi Signal Percent': Sending state 86.00000 Signal % with 0 decimals of accuracy
```

### Additional information

I know that RX and TX are exchanged compared to the docs of the C3 Board, but only this configuration worked on the breadboard and later with the soldered version.
I already tried changing them back in any thinkable way - no sucess.

rabackes, 2025-08-12:
With esp-idf the LD2420 works - but that breaks the on-device Webserver, it doesen't show any Sensors nor the log. At the beginning of my project I had some more trouble, especially the WIFI connection with esp-idf. That was the reason for choosing arduino.
But for now, I can live with that.
Nevertheless, I hope that someone could give that uart component some love...

Thanks,
Rainer

rabackes, 2025-08-12:
Could be, the LD2420 is quite picky, so that the issue only shows there? I also tried the same with an D1-32Mini (ESP32 WROOM), that works without problems. 
I'll try that again with your YAML-additions and give you feedback.

edwardtfn, 2025-08-12:
> I also tried the same with an D1-32Mini (ESP32 WROOM), that works without problems.

With Arduino??

rabackes, 2025-08-12:
Only a sidenote: With esp-idf, the device webserver doesn't show any information, also on this board.

closed 2025-11-19 by github-actions[bot]