esphome/esphome #10233: ERROR Unexpected exception while reading configuration

fraggermcg, 2025-08-14:
### The problem

Cross-posting from: https://community.home-assistant.io/t/unexpected-exception-while-reading-configuration-on-simple-media-player-yaml-config/920561

### Which version of ESPHome has the issue?

2025.7.5

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Component causing the issue

media_player

### YAML Config

```yaml
esphome:
  name: esphome-smart-speaker
  friendly_name: ESPHome smart speaker

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf #note changed from arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "[redacted]"

i2s_audio:
  - id: i2s
    i2s_lrclk_pin: GPIO10
    i2s_bclk_pin: GPIO20

#media_player:  #note this was the old working code before I tried to switch 
# over to esp-idf and speaker
#  - platform: i2s_audio
#    name: ESPHome I2S Media Player
#    dac_type: external
#    i2s_dout_pin: GPIO21
#    mode: mono

speaker:
  - platform: i2s_audio
    id: minispeaker
    i2s_audio_id: i2s
    dac_type: external
    i2s_dout_pin:   
      number: GPIO21 #DIN Pin of the MAX98357A Audio Amplifier
    sample_rate: 48000
    buffer_duration: 100ms

#note this is the section that generates the error, without it everything is ok
media_player:
  - platform: speaker
    id: ESPHomeplayer
    announcement_pipeline:
        speaker: minispeaker
        num_channels: 1

ota:
  - platform: esphome
    password: "[redacted]"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  output_power: 8.5dB
  use_address: 192.168.1.121
  manual_ip:
    static_ip: 192.168.1.121
    gateway: 192.168.1.1
    subnet: 255.255.255.0
    dns1: 192.168.1.1 

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome-Smart-Speaker"
    password: "[redacted]"

captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2025.7.5
INFO Reading configuration /config/esphome-smart-speaker.yaml…
ERROR Unexpected exception while reading configuration:
Traceback (most recent call last):
File “/usr/local/bin/esphome”, line 10, in
sys.exit(main())
^^^^^^
File “/esphome/esphome/main.py”, line 1087, in main
return run_esphome(sys.argv)
^^^^^^^^^^^^^^^^^^^^^
File “/esphome/esphome/main.py”, line 1065, in run_esphome
config = read_config(dict(args.substitution) if args.substitution else {})
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File “/esphome/esphome/config.py”, line 1162, in read_config
res = load_config(command_line_substitutions)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File “/esphome/esphome/config.py”, line 1016, in load_config
return _load_config(command_line_substitutions)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File “/esphome/esphome/config.py”, line 1006, in _load_config
return validate_config(config, command_line_substitutions)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File “/esphome/esphome/config.py”, line 930, in validate_config
result.run_validation_steps()
File “/esphome/esphome/config.py”, line 178, in run_validation_steps
task.step.run(self)
File “/esphome/esphome/config.py”, line 806, in run
self.comp.final_validate_schema(conf)
File “/esphome/esphome/components/psram/init.py”, line 65, in validate_psram_mode
if config[CONF_SPEED] == “120MHZ”:
~~~~~~^^^^^^^^^^^^
TypeError: ‘Invalid’ object is not subscriptable
```

### Additional information

Been pulling my hair out for several hours now trying to figure out why I keep getting the error above when attempting to validate a simple yaml config for a speaker running on an esp32-c3 device. I had it working on arduino framework using media_player i2s audio platform, but wanted to switch over to esp-idf and the newer media_player speaker platform as it has more features, particularly buffer size which I’m struggling with.

goberhammer, 2025-08-22:
I'm experiencing the exact same problem.

I suspect that, despite what what the documentation says, maybe PSRAM is mandatory for mediaplayer, and esp32-c3 does not have PSRAM

fraggermcg, 2025-08-24:
Ah yes OK. I've ordered a few different ones so will try out and report back

closed 2025-10-01 by swoboda1337