esphome/esphome #10234: Deep sleep not compiling on ESP32-C6 with 2025.8.0b1 and dev.

markusg1234, 2025-08-14:
### The problem

Previously deep sleep was working in dev for an ESP32-C6. It now comes up with the errors below.

Suspect it maybe related to PR#9395 

### Which version of ESPHome has the issue?

2025.8.0-b1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

deep_sleep

### YAML Config

```yaml
esphome:
  name: moisture-sensor-1
  friendly_name: Moisture sensor 1
  on_boot: 
    then:
      - switch.turn_on: moisture_sensor_power

esp32:
  board: esp32-c6-devkitm-1
  framework:
    type: esp-idf

deep_sleep:
  id: moisture_sensor_1
  run_duration: 25sec
  sleep_duration: 4hours
```

### Anything in the logs that might be useful for us?

```txt
src/esphome/components/deep_sleep/deep_sleep_esp32.cpp: In member function 'void esphome::deep_sleep::DeepSleepComponent::deep_sleep_()':
src/esphome/components/deep_sleep/deep_sleep_esp32.cpp:123:5: error: 'gpio_deep_sleep_hold_en' was not declared in this scope; did you mean 'gpio_sleep_sel_en'?
  123 |     gpio_deep_sleep_hold_en();
      |     ^~~~~~~~~~~~~~~~~~~~~~~
      |     gpio_sleep_sel_en
*** [.pioenvs/moisture-sensor-1/src/esphome/components/deep_sleep/deep_sleep_esp32.cpp.o] Error 1
========================== [FAILED] Took 6.88 seconds ==========================
```

bjornh2, 2025-08-20:
I found a workaround :)
Copy your yaml file to your esphome directory, In my case C:\Users\bjorn\esp\esphome
run command: esphome run yourfile.yaml
In my case: esphome run hallenv.yaml
It compiled and uploaded it to my esp32-C6 and my changes was included :)

bdraco, 2025-08-20:
Can you try https://github.com/esphome/esphome/pull/10345 I don't have access to an ESP32-C6 device for testing. This fix is based on ESP-IDF source code analysis and the compilation error reported in the issue. Testing on actual ESP32-C6 hardware would be appreciated.

bdraco, 2025-08-20:
https://github.com/esphome/esphome/pull/10345#issuecomment-3208282728

markusg1234, 2025-08-21:
I can confirm that #10345 fixes the issue on a EXP32 C6.
Thanks.

closed 2025-08-24 by jesserockz