esphome/esphome #10259: Issues with I2S_audio microphone DC offset correction

jpcornil-git, 2025-08-16:
### The problem

I discovered two issues with the current (2025.7.4) implementation:
- Overflow when DC offset is around/larger than full scale/1000 (`DC_OFFSET_MOVING_AVERAGE_COEFFICIENT_DENOMINATOR`)

   In the code below (L402), the multiplication is performed before the division while dc_offset_ has the same scale than q31 samples (1 = full scale)

```
  this->dc_offset_ = new_offset / DC_OFFSET_MOVING_AVERAGE_COEFFICIENT_DENOMINATOR +
                     (DC_OFFSET_MOVING_AVERAGE_COEFFICIENT_DENOMINATOR - 1) * this->dc_offset_ /
                         DC_OFFSET_MOVING_AVERAGE_COEFFICIENT_DENOMINATOR;
```

- Very slow settling time:

   In the above code, the scale factor for the update is 1/1000 (1/`DC_OFFSET_MOVING_AVERAGE_COEFFICIENT_DENOMINATOR`) but this is applied after every block of `total_samples`
=> Same as compounding an averaging filter over `total_samples` followed by a decimation by that number and a lowpass filter with a 1/1000 scale factor
==> Effective scale factor is 1/1000/`total_samples` (in my use case, with 16KHz and 128, this gives a cutoff pulsation of 16KHz/128/1000 rd/s or a time constant of 8 seconds which is huge for speech processing)

I also had a look a the `dev` branch where a code change has been implemented for the `fix_dc_offset_` method but I see a few issues with the proposed implementation:

- Gain error

   The DC gain of the filter is not 1.0 bur `R` (easy to see by setting x(n) and x(n-1) to the same value). This is not critical given that `R` is close to 1 but not very clean

- Cutoff frequency/settling time is varying with sampling rate (varying between 2.5/15Hz for 16KHz/96KHz respectively)

   This one is more annoying (15Hz is about right but 2.5Hz starts to be slow again for some applications)

 
I would propose to reuse the simple first order lowpass filter of the 2025.7.4 but apply it for every sample instead of on the average of a block of samples and take into account the sample rate to compute its cutoff frequency as illustrated below:

```
...
// DC filter cutoff frequency sets to 20 Hz (Q16)
static const uint32_t DC_OFFSET_LP_CUTOFF_RAD_S_Q16 = (uint32_t)(2 * M_PI * 20 * (1 << 16));

...
void I2SAudioMicrophone::fix_dc_offset_(std::vector<uint8_t> &data) {
  /**
   * Compute DC offset using a first order lowpass discrete filter 
   *
   * y(n) = y(n-1) + μ * (x(n-1) - y(n-1))
   *
   * with ωc ~= μ * Fs (when ωc << Fs)
   *
  */
  const size_t bytes_per_sample = this->audio_stream_info_.samples_to_bytes(1);
  const uint32_t total_samples = this->audio_stream_info_.bytes_to_samples(data.size());
  const uint32_t mu_q16 = DC_OFFSET_LP_CUTOFF_RAD_S_Q16 / this->audio_stream_info_.get_sample_rate();

  if (total_samples == 0) {
    return;
  }

  for (uint32_t sample_index = 0; sample_index < total_samples; ++sample_index) {
    const uint32_t byte_index = sample_index * bytes_per_sample;
    int32_t sample = audio::unpack_audio_sample_to_q31(&data[byte_index], bytes_per_sample);
    this->dc_offset_ += mu_q16 * ((sample - this->dc_offset_) >> 16);
    sample -= this->dc_offset_;
    audio::pack_q31_as_audio_sample(sample, &data[byte_index], bytes_per_sample);
  }

  ESP_LOGD(TAG, "DC offset: %" PRIx32 "(%" PRIx32 ")", this->dc_offset_, mu_q16);
}

```

### Which version of ESPHome has the issue?

2025.7.4

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Component causing the issue

i2s_audio/microphone

### YAML Config

```yaml
esp32:
  board: esp32dev
  framework:
    type: esp-idf
...
i2s_audio:
  - id: i2s_speaker_bus
    i2s_lrclk_pin:
      number: GPIO0
      ignore_strapping_warning: true
    i2s_bclk_pin: GPIO26

# Microphone (built-in)
microphone:
  - id: stickc_microphone
    platform: i2s_audio
    i2s_audio_id: i2s_speaker_bus
    i2s_din_pin: GPIO34
    channel: left # SPM1423 Select connected to GND
    sample_rate: 16000 # 16000 Hz * 64 = 1.024 MHz PDM rate
    bits_per_sample: 16bit # Forced by the driver for PDM mode
    adc_type: external
    pdm: true
    use_apll: true
    correct_dc_offset: true
...
```

### Anything in the logs that might be useful for us?

```txt

```

jpcornil-git, 2026-02-14:
Current implementation is still:
- using a **fixed** relative cutoff frequency (20/96000 = 1/4800) that is OK with a 96KHz sampling frequency (20Hz) but very low when e.g. 16KHz is used instead (3.33Hz)
- introducing a small gain error 2/(1+R) in the audio passband

Settling time of the filter being ~ln(error band)/ln(R) samples , this leads to more than 200 msec with a 1% error band and a 16KHz sampling frequency, i.e. 6x slower than with 96KHz and standard audio highpass filter and issue with voice activity detector/VAD)

Proposed/cookbook implementation above is fixing both issues.

egormanga, 2026-10-22:
Still relevant

jpcornil-git, 2026-10-22:
Agree, I'm running with the proposed fix without any issue

kVinsom, 2026-11-01:
I will work on this. I plan to make the DC correction sample-rate aware, check fixed-point overflow and channel independence, and add regression tests for settling time and gain across sample rates, building on the analysis above.

closed 2026-11-03 by bdraco