esphome/esphome #10326: Problem with Hide SSID WiFi

11058, 2025-08-20:
### The problem

When updating to 2025.8.0, devices using WiFi stopped connecting. The problem is solved by disabling the Hide SSID parameter in the WiFi settings.
Once you make the SSID open, everything starts working as before.

### Which version of ESPHome has the issue?

2025.8.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

wifi

### YAML Config

```yaml
on all devices without exception that use Wi-Fi. With Etnernet everything works correctly.
```

### Anything in the logs that might be useful for us?

```txt
[17:59:26][I][wifi:327]: Connecting to 'SRVPRMiot'
[17:59:26][W][wifi_esp32:478]: STA interface not initialized
[17:59:26][E][wifi:383]: wifi_sta_connect_ failed
```

MichaelGoedken, 2025-08-20:
> Can confirm this on my end. Updated 3 of my 12 KAUF plugs and stopped there because the 3 did not connect back to the AP. I didn’t release I had the SSID hidden. None of my YAML mentions hidden SSID.
> 
> Turned off hide SSID on the AP (Unifi APs) and rebooted the KAUF plugs and all three were back on the network. Even flashed again and they came back normally. Able to access all three via dns .local to see their logs in a web browser. Switched hidden SSID back on the AP and all three KAUF plugs work fine still.

After hiding the SSID again, they seem to work fine until you reboot them. Then I have to unhide the SSID again to get it to connect.

rugroovy2, 2025-08-20:
> After hiding the SSID again, they seem to work fine until you reboot them. Then I have to unhide the SSID again to get it to connect.

None I didn’t reboot them again. I don’t typically reboot them often.  Only during esphome updates.  So it’s possible they will not come back after a reboot.  But my test was Unhide SSID - plug grabs IP - rehide SSID and it’s fine.   I would imagine this is point to something in the initial start up not grabbing DHCP or seeing an AP to grab DHCP for DNS.  Also my YAML hard codes SSID and IP (but not DNS) for the plugs.

clydebarrow, 2025-08-20:
Do you have `fast_connect` set as required for connecting to hidden SSIDs? When raising an issue, it's always useful to show your yaml.

See https://arc.net/l/quote/sfumfjvr

clydebarrow, 2025-08-21:
I tested this and marking the network as `hidden` was sufficient to make it work, with or without fast-connect. No issue connecting initially and across power-cycles.

```
wifi:
  networks:
    - ssid: YouCantSeeMe
      password: redacted
      hidden: true
```

```
[10:02:26][I][wifi:327]: Connecting to 'YouCantSeeMe'
[10:02:31][I][wifi:676]: Connected
[10:02:31][C][wifi:442]:   Local MAC: 34:85:18:29:08:40
[10:02:31][C][wifi:447]:   SSID: 'YouCantSeeMe'
[10:02:31][C][wifi:450]:   IP Address: 192.168.2.184
[10:02:31][C][wifi:454]:   BSSID: 94:83:C4:5F:70:F6
[10:02:31][C][wifi:454]:   Hostname: 'c3-test'
[10:02:31][C][wifi:454]:   Signal strength: -31 dB ▂▄▆█
[10:02:31][C][wifi:465]:   Channel: 9
[10:02:31][C][wifi:465]:   Subnet: 255.255.252.0
[10:02:31][C][wifi:465]:   Gateway: 192.168.1.254
[10:02:31][C][wifi:465]:   DNS1: 192.168.1.254
```

11058, 2025-08-21:
I tried the solution with **hidden: true** and it worked!

I always had the network defined simply as:
wifi:
ssid: !secret wifi_ssid
password: !secret wifi_password

I hope this will help many, because due to the devices falling off I had to drive 30 km to a country house, run with a laptop to reflash the controllers and look for the cause.

closed 2025-08-21 by 11058