esphome/esphome #10338: Modbus reading broken in 2025.8.0

MrDabolina99, 2025-08-20:
### The problem

No (valid?) data is received from a modbus temperature sensor (CWT-TH03S-H) after update. Going back to 2025.7.5 makes it work again.

Link to sensor doc. https://store.comwintop.com/products/rs485-modbus-water-proof-temperature-humidity-sensor-probe?variant=42131365888227

### Which version of ESPHome has the issue?

2025.8.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

modbus

### YAML Config

```yaml
esphome:
  name: saunacontrol
  friendly_name: "Sauna control"
  platformio_options:
    build_flags: "-DBOARD_HAS_PSRAM"
    board_build.arduino.memory_type: qio_opi
    board_build.flash_mode: dio
    board_upload.maximum_ram_size: 524288
  on_boot:
    then:
      - climate.control: 
          id: sauna_controller
          mode: "OFF"
          target_temperature: 75
      - switch.turn_off:
          id: mains_relay
      - if:
          condition:
            binary_sensor.is_off: door
          then:
            - lvgl.widget.show: door_open

preferences:
  flash_write_interval: 
    days: 99

esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_ESP32S3_DEFAULT_CPU_FREQ_240: y
      CONFIG_ESP32S3_DATA_CACHE_64KB: y
      CONFIG_SPIRAM_FETCH_INSTRUCTIONS: y
      CONFIG_SPIRAM_RODATA: y

psram:
  mode: octal
  speed: 80MHz

# Enable logging
logger:
  baud_rate: 0
  logs:
[... 845 lines cut ...]
              then:
                - climate.control:
                    id: sauna_controller
                    target_temperature: !lambda return id(sauna_controller).target_temperature + 1;
            on_long_press_repeat:
              then:
                - climate.control:
                    id: sauna_controller
                    target_temperature: !lambda return id(sauna_controller).target_temperature + 1;
```

### Anything in the logs that might be useful for us?

```txt
[21:40:49][W][modbus:135]: Modbus CRC Check failed! F020!=00
[21:40:49][W][modbus:135]: Modbus CRC Check failed! F020!=00
[21:40:49][W][modbus:135]: Modbus CRC Check failed! F020!=00
[21:40:49][W][modbus:135]: Modbus CRC Check failed! F020!=00
[21:40:50][W][modbus:135]: Modbus CRC Check failed! F020!=00
[21:40:50][D][modbus_controller:039]: Modbus command to device=1 register=0x00 no response received - removed from send queue
```

### Additional information

Reverting back to 2025.7.5 makes modbus work again.

ssieb, 2025-08-20:
That's really weird.  It's getting loopback.

MrDabolina99, 2025-08-20:
I reverted back to 2025.7.5 once again and enabled uart debug:
[23:18:46][D][uart_debug:114]: >>> 01:03:00:00:00:02:C4:0B
[23:18:46][D][uart_debug:114]: <<< 01:03:04:02:1F:00:C9:0A:1B
[23:18:56][D][uart_debug:114]: >>> 01:03:00:00:00:02:C4:0B
[23:18:56][D][uart_debug:114]: <<< 01:03:04:02:1F:00:C9:0A:1B
[23:19:06][D][uart_debug:114]: >>> 01:03:00:00:00:02:C4:0B
[23:19:06][D][uart_debug:114]: <<< 01:03:04:02:1F:00:C9:0A:1B

ssieb, 2025-08-20:
Someone on discord has a similar issue.  No idea what's going on here.

MrDabolina99, 2025-08-21:
Good find, thanks!

Setting esp-idf version to 5.4.1 solves the issue.

jimmy-zhening-luo, 2025-08-28:
> Good find, thanks!
> 
> Setting esp-idf version to 5.4.1 solves the issue.

Does platform need to be reverted to 2025.7.x as well? Because I set esp-idf to 5.4.1 on a fresh device and it's still failing to receive despite other non-affected units succeeding.

jimmy-zhening-luo, 2025-08-28:
It doesn't feel as straightforward as people are saying. Some are reporting that reverting to 2025.7 fixed (and in this case they did not know about the 5.4.1 esp-idf fix): echavet/MitsubishiCN105ESPHome#378

Others are reporting that reverting to 2025.6 fixed things, because reverting to 2025.7 did not fix things. Those two **also** did not know about the 5.4.1 esp-idf fix: echavet/MitsubishiCN105ESPHome#355

Please stop reporting the regression from 5.4.1 to 5.4.2 as the root cause and reverting to 5.4.1 as the fix. This is clearly much more complex.

Anyway, you should also be aware that reverting platform versions does not properly clean files, and that upon each platform version revert using the built-in ESPHome add-on backups (thank god I overrode the default to save up to 15 versions, since I know this sort of fuckery can transcend many minor versions), I am getting various compilation errors instructing me to run "idf.py full clean" in the mysterious /data/blah blah blah/ build-path that is inaccessible without Dockering in (i.e. not doable unless you are a gigantic brainiac who figures out to set a manual build-path like I did).

disruptivepatternmaterial, 2025-09-05:
> It doesn't feel as straightforward as people are saying. Some are reporting that reverting to 2025.7 fixed (and in this case they did not know about the 5.4.1 esp-idf fix): [...]

I have mostly s3s here, but also some older hardware and to make them all work again I hade to go back to 2025.7.5. I am just stunned at how you break one of the oldest technologies there is...the uart. What's next they are going to make electrons + ? goodness...

Ralim, 2025-09-15:
I've reverted esphome back to `2025.7.5` to compile for devices that use modbus, and that worked perfectly using the 5.3 default idf release for that version.

MrDabolina99, 2025-09-21:
Just realized it's been over a month since I reported this issue. And sadly, none seems to be looking into fixing it. None is even assigned to the issue.

exciton, 2025-09-26:
Is everyone having this issue using RS485/modbus?

MarcoPolet, 2025-09-27:
> Is everyone having this issue using RS485/modbus?

Can't speak for everyone :-) but for me yes I use RS485/modbus (to connect to my Deye inverter), and have this issue still present indeed. Quite a hassle, I revert ESPHome Builder to 2025.7.3 from a backup, then update my ESPHome-Deye device) and then revert back to latest ESPHome builder.

bigretromike, 2025-09-28:
yes, rs485/modbus pin 1,3

pixelgrb, 2025-09-30:
Im not sure this will be helpful.  Sorry if this is just noise.

**TLDR: cold boot the ESP32** to make sure whatever version you test actually works.

I ran into a UART initialization issue on a [different project](https://github.com/echavet/MitsubishiCN105ESPHome) that had been going on for a few months.  I managed to narrow it down to a single change in [ESP-IDF v5.4.1](https://github.com/espressif/esp-idf/issues/14787#issuecomment-3349562750).  That was introduced in ESPHome 2025.8.0.  So downgrading ESPHome to 2025.7.5 ... OR ... locking ESP-IDF to v5.4.0 resolves the issue for us.

Now, you may very well be running into a different issue.  But there is a common pattern between this thread and others on that project (e.g. https://github.com/echavet/MitsubishiCN105ESPHome/issues/378) - and that's everyone is reporting something a bit different.

For us, the UART issue **only occurred after a cold boot** of the ESP32 ... but, very confusingly, could be resolved without requiring a cold boot.  So someone could:
- happy upgrade OTA to 'broken code'
- experience no problems (for months) until the ESP32 happen to get powered down and back up.
- they would try a previous version of ESPHome and report 'this version works' .. like ESPHome 2025.6.0 say
- but in that process may flash OTA a more recent version .. like ESPHome 2025.8.0 and report 'this versions works too'
- but it really doesn't.  It will fail after the next cold boot.

bigretromike, 2025-10-01:
@ssieb this is not my design its BoneIO ESP controller, I have no power over what is used I just know it use 1 and 3. 
``serial logging disabled `` how ?

@tigershadowclaw your solution is workaroud by downgrading the module that has issue.

GioMez, 2025-10-17:
> As suggested in [#11038 (comment)](https://github.com/esphome/esphome/issues/11038#issuecomment-3374955358) i tried this configuration, with the workaround of resetting the pins on boot:
> 
> ```
> esphome:
>   on_boot:
>     - priority: 1001
>       then:
>         - lambda: |-
>             gpio_reset_pin((gpio_num_t)17);
>             gpio_reset_pin((gpio_num_t)16);
> 
> esp32:
>   variant: esp32c6
>   framework:
>     type: esp-idf
> ```
> 
> and it seems to work correctly with default esp-idf version (5.4.2). I haven't tried cold boot yet.

I just performed a cold reboot on five ESP32 boards that control my split air conditioners.
Instead of unplugging each board individually, I turned the dedicated circuit breaker for the air conditioners off and on again from the electrical panel.
After reconnecting to the network, all boards successfully communicated with the air conditioners over UART using ESP-IDF 5.4.2 and ESPHome 2025.10.1.

This workaround seems much more reliable and stable than forcing an older version of ESP-IDF or ESPHome.

nikito7, 2025-11-23:
Thank you for the tips.

```
  id: modbus_serial
  rx_pin:
    number: 17
    mode:
      input: true
      pullup: true
  tx_pin: 15
  baud_rate: 4800
  stop_bits: 1
```

Working in 2025.11.0
Arduino
Esp32-s3

nikito7, 2025-12-07:
After 2025.11 all esphome are esp-idf based.

Arduino it's: IDF + arduino as component and some sort of partition compatibility. 

Use pull up on rx pin, it should fix.

exciton, 2025-12-07:
> in 2025.9 ESPhome it works fine
> i am not a developer i can do ninor canges in code but until the updates all runs fine

If you send a verbose log file I may be able to help.

exciton, 2025-12-07:
Thanks @concierge-at - now can you post one with the latest esphome version that's not working?

exciton, 2025-12-07:
@concierge-at can you turn on debug for your UART component? It seems that you're not receiving any responses at all.

concierge-at, 2025-12-07:
@exciton with a little help from you with a code snippet i think i can compile ist i only know the debug option wich i have send :/ 
i can change vom Verbose to debug when this helps didt you mean this ? sorry english is not my native language

exciton, 2025-12-07:
Add this to your uart component:
```yaml
uart:
  debug:
  #rest of config
```

concierge-at, 2025-12-07:
@exciton Thanks for info
i add it and here is the log of working 2025.9
```
INFO ESPHome 2025.9.3
INFO Reading configuration /config/esphome/sdm230-2grid-test.yaml...
INFO Detected timezone 'Europe/Vienna'
INFO Starting log output from 192.168.10.195 using esphome API
INFO Successfully resolved sdm2grid-test @ 192.168.10.195 in 0.000s
INFO Successfully connected to sdm2grid-test @ 192.168.10.195 in 0.006s
INFO Successful handshake with sdm2grid-test @ 192.168.10.195 in 0.076s
[20:19:40.312][I][app:185]: ESPHome version 2025.9.3 compiled on Dec  7 2025, 20:18:00
[20:19:40.312][C][wifi:661]: WiFi:
[20:19:40.316][C][wifi:444]:   Local MAC: 88:13:BF:03:B0:F4
[20:19:40.323][C][wifi:449]:   SSID: 'ZTE_53FFAE_2.4G'[redacted]
[20:19:40.323][C][wifi:452]:   IP Address: 192.168.10.195
[20:19:40.327][C][wifi:456]:   BSSID: F0:A7:31:2F:C0:58[redacted]
[20:19:40.327][C][wifi:456]:   Hostname: 'sdm2grid-test'
[20:19:40.327][C][wifi:456]:   Signal strength: -30 dB ▂▄▆█
[20:19:40.333][C][wifi:467]:   Channel: 1
[20:19:40.333][C][wifi:467]:   Subnet: 255.255.255.0
[20:19:40.333][C][wifi:467]:   Gateway: 192.168.10.1
[20:19:40.333][C][wifi:467]:   DNS1: 0.0.0.0
[20:19:40.333][C][wifi:467]:   DNS2: 0.0.0.0
[20:19:40.339][C][logger:273]: Logger:
[20:19:40.339][C][logger:273]:   Max Level: VERBOSE
[20:19:40.339][C][logger:273]:   Initial Level: VERBOSE
[20:19:40.340][C][logger:279]:   Log Baud Rate: 0
[20:19:40.340][C][logger:279]:   Hardware UART: UART0
[20:19:40.341][C][logger:286]:   Task Log Buffer Size: 768
[20:19:40.355][C][uart.arduino_esp32:150]: UART Bus 0:
[20:19:40.358][C][uart.arduino_esp32:151]:   TX Pin: GPIO17
[20:19:40.372][C][uart.arduino_esp32:152]:   RX Pin: GPIO16
[20:19:40.372][C][uart.arduino_esp32:154]:   RX Buffer Size: 512
[20:19:40.372][C][uart.arduino_esp32:156]:   Baud Rate: 9600 baud
[20:19:40.372][C][uart.arduino_esp32:156]:   Data Bits: 8
[20:19:40.372][C][uart.arduino_esp32:156]:   Parity: NONE
[20:19:40.372][C][uart.arduino_esp32:156]:   Stop bits: 1
[20:19:40.390][C][modbus:184]: Modbus:
[20:19:40.390][C][modbus:185]:   Flow Control Pin: GPIO4
[20:19:40.395][C][modbus:186]:   Send Wait Time: 2000 ms
[20:19:40.395][C][modbus:186]:   CRC Disabled: NO
[20:19:40.397][C][homeassistant.time:010]: Home Assistant Time:
[20:19:40.397][C][homeassistant.time:010]:   Timezone: 'CET-1CEST,M3.5.0,M10.5.0/3'
[20:19:40.427][C][modbus_controller:430]: ModbusController:
[20:19:40.427][C][modbus_controller:430]:   Address: 0x01
[20:19:40.427][C][modbus_controller:430]:   Max Command Retries: 4
[20:19:40.427][C][modbus_controller:430]:   Offline Skip Updates: 0
[20:19:40.427][C][modbus_controller:437]: sensormap
[20:19:40.427][C][modbus_controller:439]:  Sensor type=4 start=0x0 offset=0x0 count=2 size=4
[20:19:40.427][C][modbus_controller:439]:  Sensor type=4 start=0x6 offset=0x0 count=2 size=4
[20:19:40.429][C][modbus_controller:439]:  Sensor type=4 start=0xC offset=0x0 count=2 size=4
[20:19:40.432][C][modbus_controller:439]:  Sensor type=4 start=0x46 offset=0x0 count=2 size=4
[20:19:40.438][C][modbus_controller:439]:  Sensor type=4 start=0x4A offset=0x0 count=2 size=4
[20:19:40.440][C][modbus_controller:439]:  Sensor type=4 start=0x156 offset=0x0 count=2 size=4
[20:19:40.446][C][modbus_controller:443]: ranges
[20:19:40.446][C][modbus_controller:445]:   Range type=4 start=0x0 count=2 skip_updates=0
[20:19:40.449][C][modbus_controller:445]:   Range type=4 start=0x6 count=2 skip_updates=0
[20:19:40.455][C][modbus_controller:445]:   Range type=4 start=0xC count=2 skip_updates=0
[20:19:40.455][C][modbus_controller:445]:   Range type=4 start=0x46 count=2 skip_updates=0
[20:19:40.458][C][modbus_controller:445]:   Range type=4 start=0x4A count=2 skip_updates=0
[20:19:40.469][C][modbus_controller:445]:   Range type=4 start=0x156 count=2 skip_updates=0
[20:19:40.469][C][modbus_controller:448]: server registers
[20:19:40.475][C][modbus_controller.sensor:015]: modbus_controller.sensorModbus Controller Sensor 'SDM230 Voltage'
[20:19:40.475]modbus_controller.sensor  State Class: ''
[20:19:40.475]modbus_controller.sensor  Unit of Measurement: 'V'
[20:19:40.475]modbus_controller.sensor  Accuracy Decimals: 2
[20:19:40.489][C][modbus_controller.sensor:015]: modbus_controller.sensorModbus Controller Sensor 'SDM230 Frequency'
[20:19:40.489]modbus_controller.sensor  State Class: ''
[20:19:40.489]modbus_controller.sensor  Unit of Measurement: 'Hz'
[20:19:40.489]modbus_controller.sensor  Accuracy Decimals: 2
[... 478 lines cut ...]
[20:25:02.804][V][modbus_controller:297]: Range : 4A Size: 2 (4) skip: 0
[20:25:02.807][W][modbus_controller:284]: Duplicate modbus command found: type=0x4 address=74 count=2
[20:25:02.807][V][modbus_controller:297]: Range : 156 Size: 2 (4) skip: 0
[20:25:02.813][W][modbus_controller:284]: Duplicate modbus command found: type=0x4 address=342 count=2
[20:25:02.886][V][modbus:043]: Stop waiting for response from 1
[20:25:02.887][W][modbus_controller:027]: Modbus device=1 set offline
[20:25:02.887][D][modbus_controller:039]: Modbus command to device=1 register=0x00 no response received - removed from send queue
[20:25:02.889][V][modbus_controller:043]: Sending next modbus command to device 1 register 0x06 count 2
[20:25:02.901][V][modbus:258]: Modbus write: 01.04.00.06.00.02.91.CA (8)
[20:25:02.903][V][modbus_controller:676]: Command sent 4 0x6 2 send_count: 1
[20:25:02.907][D][uart_debug:114]: >>> 01:04:00:06:00:02:91:CA
[20:25:02.920][W][modbus:145]: Modbus CRC Check failed! 26C0!=3E00
[... cut ...]
[20:25:04.921][V][modbus_controller:676]: Command sent 4 0x6 2 send_count: 2
[20:25:04.925][D][uart_debug:114]: >>> 01:04:00:06:00:02:91:CA
[20:25:04.938][W][modbus:145]: Modbus CRC Check failed! FFF7!=603C
[20:25:04.939][V][modbus:027]: Clearing buffer of 9 bytes - parse failed
[20:25:05.048][D][uart_debug:114]: <<< 5A:01:04:04:3D:B6:B4:3C:60
```
A lot of lines wich i undersnad only IP adress :/
Thanks a lot that you spend time to fix my problem. This is my Testequipment the inverter and the real SDM wich is the same must be first shot hit when i get them to update

concierge-at, 2025-12-07:
@exciton 
it wars a long day 
i have done the canges look at the yaml 
```
substitutions:
  devicename: "sdm2grid-test"
  device_description: "SDM RS485 Logger USBC"
  friendly_name: "sdm230"

esphome:
  name: ${devicename}
  comment: "${device_description}"

esp32:
  board: esp32dev
  framework:
    type: arduino

###################################################################
# External Components – production-ready branch
###################################################################
external_components:
  - source: github://pr#8103
    components: [uart]
  - source: github://pr#8032
    components: [modbus, modbus_controller, growatt_solar]
    refresh: 1h

web_server:
  port: 80
  auth:
    username: sdm
    password: sdm
  version: 2

logger:
  level: VERBOSE
  baud_rate: 0

api:
  encryption:
    key: !secret api_key

ota:
  password: sdm_ota_update
  platform: esphome

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  manual_ip:
    static_ip: 192.168.10.195
    gateway: 192.168.10.1
    subnet: 255.255.255.0

  ap:
    ssid: "${devicename}"
    password: sdmsdmsdmsdm

  fast_connect: true
  power_save_mode: none

captive_portal:

time:
  - platform: homeassistant
    id: homeassistant_time

###################################################################
[... 250 lines cut ...]
[22:08:20.164][V][modbus:084]: Received first byte 151 (0X97) 13ms after last send
[22:08:20.165][W][modbus:193]: CRC check failed 16ms after last send
[22:08:20.168][W][modbus:363]: Clearing buffer of 9 bytes - parse failed 18ms after last send
[22:08:20.170][V][modbus_controller:329]: 5 modbus commands already in queue
[22:08:20.176][V][modbus_controller:297]: Range : 0 Size: 2 (4) skip: 0
[22:08:20.177][W][modbus_controller:284]: Duplicate modbus command found: type=0x4 address=0 count=2
[22:08:20.177][V][modbus_controller:297]: Range : 6 Size: 2 (4) skip: 0
[22:08:20.183][W][modbus_controller:284]: Duplicate modbus command found: type=0x4 address=6 count=2
[22:08:20.190][V][modbus_controller:297]: Range : C Size: 2 (4) skip: 0
[22:08:20.190][W][modbus_controller:284]: Duplicate modbus command found: type=0x4 address=12 count=2
[22:08:20.190][V][modbus_controller:297]: Range : 46 Size: 2 (4) skip: 0
[22:08:20.190][W][modbus_controller:284]: Duplicate modbus command found: type=0x4 address=70 count=2
[22:08:20.196][V][modbus_controller:297]: Range : 4A Size: 2 (4) skip: 0
[22:08:20.196][V][modbus_controller:297]: Range : 156 Size: 2 (4) skip: 0
[22:08:20.196][W][modbus_controller:284]: Duplicate modbus command found: type=0x4 address=342 count=2
[22:08:20.271][D][uart_debug:114]: <<< 97:01:04:04:3E:FE:F9:DB:95

```
to test i get back to 2025.9 hardware works well
i belive the MODBUS and UART HATE ME

i have done the changes from https://github.com/esphome/esphome/pull/8032

i have no idea what i can do i will look tomorrow fpr rs485 without flowcontrol pin
thank you and a good night now

exciton, 2025-12-07:
Very strange. The UART responses have the last byte of the modbus frame missing (0x97), and it shows up at the start of the next frame:

Log shows this:
[22:08:16.135][D][uart_debug:114]: <<< 97:01:04:04:3E:FE:F9:DB:95

Correct response is this: 01:04:04:3E:FE:F9:DB:95:97

I think this bug is in the uart component or driver somewhere.