esphome/esphome #10344: SwirchtBot Lock Pro ESP_GATTC_OPEN_EVT

TioBundy, 2025-08-20:
### The problem

Apparently the August update breaks or significantly affects the connections of Switchbot Lock Pro. I do not know if it is smart lock uses the V1 connection that is said in the documentation that has stopped offering support because I can't find documentation about it. But I understand that it is a relatively new product (2023-24) and should have something more modern than a first version of BLE, which I imagine will be somewhat older.

However, the manual openings and locks were still working correctly for me, because I spent months testing fixes so that the proxy plates work well with Bermuda, do not get saturated, The devices that need the proxy connect and everything works well, in the midst of all the changes that ESPHOME has undergone.

Apparently There is no way to have a bluetooth proxy without Bermuda Note the flashed inserts as a proxy, that's why I started using the option "on_ble_advertise", A ChatGPT genius that I did not know and helped me to the stability of the lock.

But I've gone from having several warnings per second of the lock to starting to see:

”ESP_GATTC” errors.

I have returned to the July version with a backup, but I have also deleted the integrations of SwitchBot because that part did not respond, I don't know if it's esphome, but everything has worked correctly with these changes.

[image]

I leave the piece of yaml of how I handle the smart lock now and the part scans in Bermuda, although no matter how much I reduce the values window and interval, Bermuda keeps detecting my mobile and sometimes it can cause problems because I don't have the lock pro exclusivity with that device.

### Which version of ESPHome has the issue?

2025.8

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### YAML Config

```yaml
esp32_ble_tracker:
  scan_parameters:
    interval: 400ms
    window: 10ms

  on_ble_advertise:
    mac_address: "MAC_ADRESS"
    then:
      - logger.log: "detectada"

ble client:
  - id: lock_pro_client
    mac address: "MAC_ADRESS"
    auto connect: True
```

### Anything in the logs that might be useful for us?

```txt

```

bdraco, 2025-08-20:
Hi @TioBundy 

Thanks for reporting this issue with the Switchbot Lock Pro. To help diagnose the ESP_GATTC errors you're experiencing after updating to ESPHome 2025.8, could you please provide:

**Text logs instead of screenshots** - Screenshots make it impossible to:
- Search for specific error patterns
- See the complete error messages (yours appear cut off)
- Copy error codes for investigation
- Analyze timing and sequence of events

You can copy the logs directly from:
- ESPHome dashboard → Click on your device → "LOGS" button → Select and copy the text
- Or from the command line: `esphome logs your-device.yaml`

Please include:
1. **At least 30-60 seconds of logs** showing the ESP_GATTC errors occurring
2. **The complete error messages** - ESP_GATTC errors include specific error codes (like `ESP_GATTC_OPEN_FAIL`, `ESP_GATTC_READ_TIMEOUT`, etc.) that tell us exactly what's failing
3. **Comparison logs** from the working July version if you still have access to them

I see you already have debug logging enabled, which is perfect. The text format of these debug logs will help us identify whether this is a regression in the BLE connection handling or a configuration issue.

Thanks for your patience in helping us track this down!

bdraco, 2025-08-20:
Also, I noticed something unusual in your configuration that might be contributing to the issue:

You have both:
1. **BLE Client** configured to connect directly to the lock:
   ```yaml
   ble_client:
     - id: lock_pro_client
       mac_address: "MAC_ADDRESS"
       auto_connect: true
   ```

2. **Bluetooth Proxy** enabled, which forwards BLE advertisements to Home Assistant

3. **on_ble_advertise trigger** listening for the same device

This creates a conflict - you're trying to both:
- **Connect directly** to the device (via `ble_client`) for local control
- **Proxy** the device to Home Assistant (via `bluetooth_proxy`) for remote control

These two modes are generally incompatible for the same device. The ESP32 can't simultaneously maintain a direct GATT connection to the lock AND proxy it to Home Assistant. This could explain the ESP_GATTC errors you're seeing.

The issue here is that you have `ble_client` configured for the lock, but ESPHome doesn't have a Switchbot Lock component that uses `ble_client`. The `ble_client` configuration alone doesn't do anything useful without a component that actually uses it to control the device.

**What you should do:**
- Remove the `ble_client` configuration entirely - it's not being used
- Keep `bluetooth_proxy` to forward the lock's advertisements to Home Assistant
- Use Home Assistant's native Switchbot integration to control the lock

The `ble_client` attempting to connect to a device that it can't actually control, while `bluetooth_proxy` is also trying to handle the same device, is likely causing the ESP_GATTC errors. The unused `ble_client` connection is interfering with the proxy functionality.

TioBundy, 2025-09-21:
Hi, it seems that with the latest updates everything is working without any problems. I don't know if there were any fixes, but the erroneous logs that led me to open the post don't seem to be repeating.