esphome/esphome #10346: I2C not working in 2025.8.0

j9brown, 2025-08-21:
### The problem

I'm observing NACK errors on the I2C bus in 2025.8.0 and am unable to communicate with my devices.  Everything worked fine in 2025.7.5.

At the suggestion of someone on Discord, I changed my config to use ESP-IDF 5.3.2 and I2C works fine again.  So there appears to be a regression in the ESP-IDF 5.4.2 I2C implementation.

### Which version of ESPHome has the issue?

2025.8.0

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Component causing the issue

i2c

### YAML Config

```yaml
See here: https://github.com/j9brown/minuet/blob/v3/esphome/minuet/core.yaml
```

### Anything in the logs that might be useful for us?

```txt
Hard resetting via RTS pin...
INFO Successfully uploaded program.
INFO Starting log output from /dev/tty.usbmodem11201 with baud rate 921600
[16:00:29]I (273) boot: Set actual ota_seq=1 in otadata[0]
[16:00:29]I (273) boot: Disabling RNG early entropy source...
[16:00:29][I][logger:165]: Log initialized
[16:00:29][C][safe_mode:082]: There have been 0 suspected unsuccessful boot attempts
[16:00:29][D][esp32.preferences:142]: Writing 1 items: 0 cached, 1 written, 0 failed
[16:00:29][I][app:090]: Running through setup()
[16:00:29][I][i2c.idf:437]: Performing bus recovery
[16:00:29][C][component:164]: Setup i2c took 16ms
[16:00:29][C][component:164]: Setup preferences took 0ms
[16:00:29][C][component:164]: Setup esphome.coroutine took 0ms
[16:00:29][D][pca9554:035]: Initialization complete. Warning: 0, Error: 0
[16:00:29][C][component:164]: Setup pca9554 took 4ms
[16:00:29][D][esp-idf:000]: E (593) i2c.master: I2C hardware NACK detected
[16:00:29][D][esp-idf:000]: E (594) i2c.master: I2C transaction unexpected nack detected
[16:00:29][D][esp-idf:000]: E (595) i2c.master: s_i2c_synchronous_transaction(945): I2C transaction failed
[16:00:29][D][esp-idf:000]: E (596) i2c.master: i2c_master_execute_defined_operations(1366): I2C transaction failed
[16:00:29][I][MCF8316:363]: Loading configuration shadow registers from EEPROM
[16:00:29][D][esp-idf:000]: E (824) i2c.master: s_i2c_synchronous_transaction(945): I2C transaction failed
[16:00:29][D][esp-idf:000]: E (824) i2c.master: i2c_master_execute_defined_operations(1366): I2C transaction failed
[16:00:29][E][MCF8316:560]: Failed to read register 0xea: ERROR_I2C code 2
[16:00:29][W][MCF8316:376]: Failed to confirm the configuration shadow registers were loaded from the EEPROM: ERROR_I2C
[16:00:29][I][MCF8316:333]: Reading configuration shadow registers
[16:00:29][D][esp-idf:000]: E (848) i2c.master: s_i2c_synchronous_transaction(945): I2C transaction failed
[16:00:29][D][esp-idf:000]: E (849) i2c.master: i2c_master_execute_defined_operations(1366): I2C transaction failed
[16:00:29][E][MCF8316:560]: Failed to read register 0x80: ERROR_I2C code 2
[16:00:29][E][MCF8316:338]: Failed to read configuration shadow registers: ERROR_I2C
[16:00:29][I][MCF8316:363]: Loading configuration shadow registers from EEPROM
[16:00:30][D][esp-idf:000]: E (1172) i2c.master: s_i2c_synchronous_transaction(945): I2C transaction failed
[16:00:30][D][esp-idf:000]: E (1173) i2c.master: i2c_master_execute_defined_operations(1366): I2C transaction failed
[16:00:30][E][MCF8316:560]: Failed to read register 0xea: ERROR_I2C code 2
[16:00:30][W][MCF8316:376]: Failed to confirm the configuration shadow registers were loaded from the EEPROM: ERROR_I2C
[16:00:30][I][MCF8316:333]: Reading configuration shadow registers
[16:00:30][D][esp-idf:000]: E (1196) i2c.master: s_i2c_synchronous_transaction(945): I2C transaction failed
[16:00:30][D][esp-idf:000]: E (1197) i2c.master: i2c_master_execute_defined_operations(1366): I2C transaction failed
[16:00:30][E][MCF8316:560]: Failed to read register 0x80: ERROR_I2C code 2
[16:00:30][E][MCF8316:338]: Failed to read configuration shadow registers: ERROR_I2C
[16:00:30][I][MCF8316:363]: Loading configuration shadow registers from EEPROM
[16:00:30][D][esp-idf:000]: E (1520) i2c.master: s_i2c_synchronous_transaction(945): I2C transaction failed
[16:00:30][D][esp-idf:000]: E (1521) i2c.master: i2c_master_execute_defined_operations(1366): I2C transaction failed
[16:00:30][E][MCF8316:560]: Failed to read register 0xea: ERROR_I2C code 2
[16:00:30][W][MCF8316:376]: Failed to confirm the configuration shadow registers were loaded from the EEPROM: ERROR_I2C
[16:00:30][I][MCF8316:333]: Reading configuration shadow registers
[16:00:30][D][esp-idf:000]: E (1544) i2c.master: s_i2c_synchronous_transaction(945): I2C transaction failed
[16:00:30][D][esp-idf:000]: E (1545) i2c.master: i2c_master_execute_defined_operations(1366): I2C transaction failed
[16:00:30][E][MCF8316:560]: Failed to read register 0x80: ERROR_I2C code 2
[16:00:30][E][MCF8316:338]: Failed to read configuration shadow registers: ERROR_I2C
[16:00:30][E][component:296]: mcf8316 set Error flag: Failed to load config from EEPROM
[16:00:30][E][component:204]: mcf8316 was marked as failed
[16:00:30][C][component:164]: Setup mcf8316 took 1057ms
[16:00:30][C][component:164]: Setup template.sensor took 1ms
```

### Additional information

The components on the I2C bus are a PCA9555 and a MCF8316D.  This I2C bus fault occurs immediately on boot and recurs after a reset.

I'm writing my own [component](https://github.com/j9brown/esphome-mcf8316) for the MCF8316D.  I don't think it's responsible for this issue.  Everything was working fine just before I upgraded ESPHome.

I should note that one of the GPIO pins of the PCA9555 drives the sleep/wake behavior of the MCF8316D.  So if the PCA9555 fails to initialize correctly then the MCF8316D could go to sleep and not respond to I2C requests.  However, we don't see any errors from the PCA9555 component being reported so my assumption is that the I2C bus driver is in a bad state.

swoboda1337, 2025-08-21:
Your component is uses the i2c driver directly not using the ESPHome I2C code.

j9brown, 2025-08-21:
After enabling VERY_VERBOSE logs for "i2c.idf", I noticed that some of my I2C write transactions succeeded and some failed.  The ones that succeeded all had stop = true whereas the ones that failed had stop = false.  When I changed my commands to use stop = true, the write succeeded and the MCF8316 component worked as expected.

Fortunately, the MCF8316 allows both usages: with or without a stop between the write and a read transactions for reading a register.

So there's been a change in behavior regarding I2C write transactions with stop = false.

---

More details on the debugging process for folks who want to dig deeper...

I2C log with ESP-IDF 5.3.2, using stop = false, and works:

```
[20:20:56][VV][i2c.idf:338]: 0x01 TX 5000EA000000404A
[20:20:56][VV][i2c.idf:338]: 0x01 TX D000EA
[20:20:56][VV][i2c.idf:313]: 0x01 RX 00000000B3
```

I2C log with ESP-IDF 5.4.2, using stop = false, and doesn't work:

```
[20:22:54][VV][i2c.idf:338]: 0x01 TX 5000EA000000404A
[20:22:55][VV][i2c.idf:338]: 0x01 TX D000EA
[20:22:55][VV][i2c.idf:374]: TX to 01 failed: not acked
```

I'll point out that "not acked" is actually a misnomer.  It means that `i2c_master_execute_defined_operations` returned ESP_ERR_INVALID_STATE.  Specifically, I'm fairly certain the error originates from here in `s_i2c_transaction_start`: https://github.com/espressif/esp-idf/blob/v5.4.2/components/esp_driver_i2c/i2c_master.c#L651

```
        if (atomic_load(&i2c_master->status) != I2C_STATUS_DONE) {
            ret = ESP_ERR_INVALID_STATE;
        }
```

The `I2C_STATUS_DONE` status is asserted when the interrupt routine observes the `I2C_LL_INTR_MST_COMPLETE` event.  And that happens when the `I2C_INT_STATUS_REG` hardware register contains the bit `I2C_TRANS_COMPLETE_INT_ST` which signals the `I2C_TRANS_COMPLETE_INT` interrupt.

According to the technical reference manual, here's how you send an I2C transaction at a low-level:

> After the command sequence is configured and data in RAM is ready, I2Cmaster enables the controller and
> initiates data transfer by setting the I2C_TRANS_START bit. The controller has four steps to take:
> 1. Wait for SCL to go high, to avoid SCL being used by other masters or slaves.
> 2. Execute a RSTART command and send a START bit.
> 3. Execute a WRITE command by taking N+1 bytes from the RAM in order and send them to I2Cslave in the
> same order. The first byte is the address of I2Cslave.
> 4. Send a STOP. Once the I2Cmaster transfers a STOP bit, an I2C_TRANS_COMPLETE_INT interrupt is
> generated.

And more concretely:

> If the command is a STOP, I2C stops transfer and generates an I2C_TRANS_COMPLETE_INT interrupt.

So I assume from this that the I2C transaction is only considered complete when it includes a STOP command.  So, I would expect that calling any `esphome::i2c::I2CBus` method with stop = false probably fails now.  (I haven't confirmed this yet.)

---

I recommend that we change the `esphome::i2c::I2CBus` API as follows:

- Remove the stop argument from `read()`, `readv()`, `write()`, and `writev()` because it is no longer possible to perform a transaction that does not end with a STOP.
- Add a new method, say `transact()`, that accepts an array of read and write command structures and sends the commands to ESP-IDF as a single operation consisting of a START, the provided reads/writes, and a STOP.
- Change the error messages that are logged to more accurately reflect the status reported by ESP-IDF.  For example, say "invalid state" instead of "nack".  Saying "nack" mistakenly led me believe the problem was with the I2C device or I2C bus (on an electrical level).

Similarly, I recommend updating the `esphome::i2c::I2CDevice` API as follows:

- Remove the stop argument from `write_register*()` because it doesn't work anymore.  Always stop after the write.
- Keep the stop argument in `read_register*()` and change its meaning such that it determines whether a stop happens between the write command that sets the requested register address and the read command that follows (use the new `transact()` method for this).  Always stop after the read.

Thanks for helping me out in Discord!  And I hope you find this information useful for improving the I2C stack.

nagyrobi, 2025-08-22:
https://github.com/esphome/esphome/issues/10352

PR #10389 (2025-08-27): [i2c] Perform register reads as single transactions
# What does this implement/fix?

Recent change in ESP-IDF seems to mean that i2c transfers aren't successful if STOP condition is not sent after the transfer.This is an issue when reading a register, since it means the write (of the register address) and the read often have to be in a single transaction, with no STOP condition between the write and read.

So refactor the `i2c` component and replace separate bus read and write functions with a single `write_readv` that takes a write buffer and read buffer.

Update all dependent components to u [... cut]
closed 2025-08-27 by clydebarrow