esphome/esphome #10352: jc3248w535en touch broken after update to 2025.8.0

oli737, 2025-08-21:
### The problem

After update my Display spams in log this error. Touch is not working anymore. i did not cahnge any code.

[11:25:12][I][esp-idf:000]: E (531383) i2c.master: s_i2c_synchronous_transaction(945): I2C transaction failed
[11:25:12][I][esp-idf:000]: E (531386) i2c.master: i2c_master_execute_defined_operations(1366): I2C transaction failed
[11:25:12][I][esp-idf:000]: E (531438) i2c.master: s_i2c_synchronous_transaction(945): I2C transaction failed
[11:25:12][I][esp-idf:000]: E (531441) i2c.master: i2c_master_execute_defined_operations(1366): I2C transaction failed
[11:25:13][I][esp-idf:000]: E (531493) i2c.master: s_i2c_synchronous_transaction(945): I2C transaction failed

### Which version of ESPHome has the issue?

2025.8.0 

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

i2c

### YAML Config

```yaml
....

esphome:
  name: ${device_name}
  friendly_name: ${friendly_name}
  comment: 192.168.178.76
  platformio_options:
    upload_speed: 115200
    board_build.flash_mode: dio
    board_build.f_flash: 40000000L
    board_build.f_cpu: 160000000L

esp32:
  board: esp32-s3-devkitc-1
  flash_size: 16MB
  framework:
    type: esp-idf
    version: recommended
    sdkconfig_options:
      CONFIG_ESP32S3_DEFAULT_CPU_FREQ_240: "y"
      CONFIG_ESP32S3_DATA_CACHE_64KB: "y"
      CONFIG_ESP32S3_DATA_CACHE_LINE_64B: "y"
      CONFIG_SPIRAM_FETCH_INSTRUCTIONS: y
      CONFIG_SPIRAM_RODATA: y

psram:
  mode: octal
  speed: 80MHz

...

# I2C for Touchscreen
i2c:
  sda:
    number: 4
  scl:
    number: 8
  id: touchscreen_bus
  frequency: 100kHz
  scan: true

touchscreen:
  - platform: axs15231
    id: my_touch
    display: my_display
    i2c_id: touchscreen_bus
    calibration:
      x_min: 0
      x_max: 480
      y_min: 0
      y_max: 320
    transform:
      swap_xy: true
      mirror_y: true
    on_touch:
      - lambda: |-
```

### Anything in the logs that might be useful for us?

```txt

```

### Additional information

I did try 50khz and scan false for i2c but no positive results.
I wanna avoide switch to arduino ide.

clydebarrow, 2025-08-21:
Confirmed; here is a workaround:

```
external_components:
  - source: github://esphome/esphome@2025.7.1
    components: [i2c]
```

clydebarrow, 2025-08-21:
Try setting the i2c clock speed to 400kHz.

oli737, 2025-08-21:
This worked for me but still at 100kHz. I cant say why is that different but working now, thanks.

external_components:
  - source: github://esphome/esphome@2025.7.1
    components: [i2c]
    refresh: never

swoboda1337, 2025-08-21:
FYI `CONFIG_ESP32S3_DEFAULT_CPU_FREQ_240: "y"` that doesnt do anything anymore you need to set `cpu_frequency`: https://esphome.io/components/esp32/

I think this has to do with https://github.com/esphome/esphome/issues/10346. I think any component that sets stop = false for I2C writes is broken.

KoKolaj, 2025-08-22:
The solution provided here got my device back up and running. Thanks a lot for the helpful advice!

nagyrobi, 2025-08-22:
Workaround still not working on `axs15231` touchscreen on Guition JC3248w535 display.
Downgrading ESPHome to 2025.7.5 is the only way to have a functional device.

AuldTimer, 2025-08-22:
> Workaround still not working on `axs15231` touchscreen on Guition JC3248w535 display. Downgrading ESPHome to 2025.7.5 is the only way to have a functional device.

Same here, platform: axs15231 not working with workaround, thanks

clydebarrow, 2025-08-22:
To test do the following (this works for me.)

1. Completely remove the .esphome folder (this removes all build artifacts, including downloads from `external_components`)
2. Ensure you are using ESPHome 2025.8.0 or the dev branch.
3. Compile and test the yaml below *unmodified!!*
4. Check the touchscreen.

```
substitutions:
  friendly_name: JC3248W535 with AXS15231

esphome:
  name: jc3248w535
  friendly_name: ${friendly_name]

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: esp-idf

logger:

psram:
  mode: octal
  speed: 80MHz

spi:
  id: display_qspi
  type: quad
  clk_pin: 47
  data_pins: [21,48,40,39]

power_supply:
  id: backlight_id
  pin: 1
  enable_on_boot: true

display:
  - platform: mipi_spi
    model: jc3248w535

i2c:
  sda: 4
  scl: 8
  frequency: 400kHz

external_components:
  - source: github://esphome/esphome@2025.7.1
    components: [i2c]

touchscreen:
  platform: axs15231
  update_interval: 30ms

lvgl:
```

nagyrobi, 2025-08-24:
Breaks, as soon as I change these, to set the display landscape:

```yaml
display:
  - platform: mipi_spi
    model: jc3248w535
    rotation: 90

touchscreen:
  - platform: axs15231
    update_interval: 30ms
    calibration:
      x_min: 0
      x_max: 479
      y_min: 0
      y_max: 319
    transform:
      swap_xy: true
      mirror_x: false
      mirror_y: true
```

clydebarrow, 2025-08-24:
> but picture breaks, as soon as I change these, to set the display landscape

Yeah, I see that too. Have not looked into it yet. Setting the display dimensions works around it.

clydebarrow, 2025-08-26:
> picture breaks, as soon as I change these, to set the display landscape

https://github.com/esphome/esphome/pull/10443

nagyrobi, 2025-08-27:
Tested this a few moments ago, display works fine now with https://github.com/esphome/esphome/pull/10443, but touch still needs the workaround above.

clydebarrow, 2025-08-27:
Should be:
```
    components: [i2c, axs15231]
```

It's merged into dev now.

closed 2025-08-27 by clydebarrow

nagyrobi, 2025-08-27:
I'm on 8.1.

PR #10389 (2025-08-27): [i2c] Perform register reads as single transactions
# What does this implement/fix?

Recent change in ESP-IDF seems to mean that i2c transfers aren't successful if STOP condition is not sent after the transfer.This is an issue when reading a register, since it means the write (of the register address) and the read often have to be in a single transaction, with no STOP condition between the write and read.

So refactor the `i2c` component and replace separate bus read and write functions with a single `write_readv` that takes a write buffer and read buffer.

Update all dependent components to u [... cut]