esphome/esphome #10358: Error when compiling with 2025.8.0 (it works fine on 2025.6.2)

warheat1990, 2025-08-21:
### The problem

I'm trying to update to version 2025.8.0 and it's broken (throwing bunch of error when compiling) but it works fine on 2025.6.2, device is Sonoff M5 (3 gang EU version), here's some of the code

```
binary_sensor:
  - platform: gpio
    name: "Left Button"
    pin:
      number: GPIO04
      mode: INPUT_PULLUP
      inverted: true
    on_multi_click:
      # single click
      - timing:
          - ON for at most 1s
          - OFF for at least 0.3s
        then:
          lambda: |-
            // if relay OFF or not connected to HASS, toggle relay
            if (!id(${eid}_left_relay).state || !global_api_server->is_connected())
            {
              id(${eid}_left_relay).toggle();
            }
            else
            {
              HomeassistantServiceResponse svc;
              svc.service = "pyscript.switch_single_click";

              HomeassistantServiceMap ent;
              ent.key = "ls_eid";
              ent.value = "binary_sensor.${eid}_left_button";
              svc.data.push_back(ent);
              global_api_server->send_homeassistant_service_call(svc);
            }
      # double click
      - timing:
          - ON for at most 1s
          - OFF for at most 1s
          - ON for at most 1s
          - OFF for at least 0.1s
        then:
          lambda: |-
         ..............
```

### Which version of ESPHome has the issue?

2025.8.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt
/config/esphome/common/sonoff-m5/m5-3c-86/binary_sensor.yaml: In lambda function:
/config/esphome/common/sonoff-m5/m5-3c-86/binary_sensor.yaml:22:9: error: 'HomeassistantServiceResponse' was not declared in this scope
   22 |               HomeassistantServiceResponse svc;
[... 79 lines cut ...]
      |         ^  
      |         int
/config/esphome/common/sonoff-m5/m5-3c-86/binary_sensor.yaml:147:28: error: 'class esphome::api::APIServer' has no member named 'send_homeassistant_service_call'
  147 |               global_api_server->send_homeassistant_service_call(svc);
      |                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/config/esphome/common/sonoff-m5/m5-3c-86/binary_sensor.yaml: In lambda function:
/config/esphome/common/sonoff-m5/m5-3c-86/binary_sensor.yaml:164:9: error: 'HomeassistantServiceResponse' was not declared in this scope
  164 |               HomeassistantServiceResponse svc;
      |         ^     ~~~~~~~~~~~~~~~~~~~~~~
/config/esphome/common/sonoff-m5/m5-3c-86/binary_sensor.yaml:165:9: error: 'svc' was not declared in this scope
  165 |               svc.service = "pyscript.switch_double_click";
      |         ^  
/config/esphome/common/sonoff-m5/m5-3c-86/binary_sensor.yaml:167:9: error: 'HomeassistantServiceMap' was not declared in this scope
  167 |               HomeassistantServiceMap ent;
      |         ^     ~~~~~~~~~~~~~~~~~
/config/esphome/common/sonoff-m5/m5-3c-86/binary_sensor.yaml:168:9: error: 'ent' was not declared in this scope; did you mean 'int'?
  168 |               ent.key = "ls_eid";
      |         ^  
      |         int
/config/esphome/common/sonoff-m5/m5-3c-86/binary_sensor.yaml:171:28: error: 'class esphome::api::APIServer' has no member named 'send_homeassistant_service_call'
  171 |               global_api_server->send_homeassistant_service_call(svc);
      |                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
*** [.pioenvs/ls-wk1fy/src/main.cpp.o] Error 1
========================= [FAILED] Took 145.18 seconds =========================
```

bdraco, 2025-08-21:
You'll need to turn on `homeassistant_services` manually if you aren't using any outside your custom lambdas

https://esphome.io/components/api/

```
homeassistant_services (Optional, boolean): Enable compilation of Home Assistant service call support for external components that use the C++ CustomAPIDevice::call_homeassistant_service() or CustomAPIDevice::fire_homeassistant_event() methods. This is automatically enabled when using homeassistant.service or homeassistant.event actions, or the homeassistant platform for number or switch components. Only needs to be manually set when external components call Home Assistant services without using the built-in actions. Defaults to false.
```

warheat1990, 2025-08-24:
fixed it by modifying the code to this

          HomeassistantServiceResponse svc;
          svc.set_service(StringRef("pyscript.switch_single_click"));

          HomeassistantServiceMap ent;
          ent.set_key(StringRef("ls_eid"));
          ent.value = "binary_sensor.${eid}_left_button";
          svc.data.push_back(ent);
          global_api_server->send_homeassistant_service_call(svc);