esphome/esphome #10367: [thermostat_climate] Climate entity incorrectly saves/restores preset and fan mode

j9brown, 2025-08-22:
### The problem

My device has a thermostat climate entity.  I see a warning on every boot: "Preset NONE is not supported by this device!"  That got me curious.

Indeed, my thermostat climate entity doesn't define a NONE preset in YAML because that would be silly (although I did experiment with that).  Upon closer inspection, I believe this warning is caused by some bugs in the Climate entity's state restore code.

`Climate::save_save_()` populates the `ClimateDeviceRestoreState` object with information about the currently selected preset  or custom preset in the `preset` / `custom_preset` union.  It sets `uses_custom_preset` as a discriminator to indicate which field of the union is used.  If the climate entity does not have a preset or custom preset, then the `preset` value will remain zero-initialized (equivalent to CLIMATE_PRESET_NONE) and `uses_custom_preset` will be false.

Overall, there are 3 important states to distinguish: has no preset, has a standard preset, has a custom preset.  Let's see what happens downstream.

`ClimateDeviceRestoreState::to_call()` dereferences the `preset` field of the union without checking `uses_custom_preset` and passes the value to `ClimateCall::set_preset()`.  This method fails to distinguish the 3 possible states.  First, the method should check `uses_custom_preset` and, if true, decode the custom preset index to set the ClimateCall correctly.  Second, the method should check whether there is no preset to restore (`uses_custom_preset` is false and `preset` is zero) and in that case, it should NOT call `ClimateCall::set_preset()` at all because there's no preset information in there.  This distinction matters because ClimateCall's preset field is `std::optional<...>` and there are behavioral differences depending on whether it contains `std::nullopt` or zero.

Note that `to_call()` also does the wrong thing for custom fan modes.

`ClimateDeviceRestoreState::apply()` has similar bugs.  Although it checks `uses_custom_preset`, it doesn't handle the case where there is no preset to restore.  So that will result in `Climate->preset` being the value of `CLIMATE_PRESET_NONE` instead of `std::nullopt`.

As for the spurious log message, I'm pretty sure it's because `ThermostatClimate::setup()` uses `ClimateDeviceRestoreState::to_call()` to restore its state and the `ClimateCall` object 's preset field erroneously contains `CLIMATE_PRESET_NONE` instead of `std::nullopt` (due to the bug I described above) and `ClimateCall::validate_()` yells because NONE is not a supported preset.

Although I haven't checked every case, I think this means that restoring custom presets and fan modes doesn't work at all for thermostat climate entities and perhaps other climate implementations.  Seems worth fixing.

### Which version of ESPHome has the issue?

2025.8.0

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Component causing the issue

climate

### YAML Config

```yaml
# Excerpt from: https://github.com/j9brown/minuet/blob/v3/esphome/minuet/core.yaml

  substitutions:
    minuet_thermostat_default_temperature: "78"
  climate:
    - id: minuet_thermostat
      name: "Thermostat"
      icon: mdi:thermostat-auto
      platform: thermostat
      sensor: minuet_ambient_temperature
      visual: # The IR remote uses Fahrenheit natively and expects the fan to do so as well
        min_temperature: 29°F
        max_temperature: 99°F
        temperature_step: 1°F
      cool_overrun: 2°F
      cool_deadband: 1°F
      min_idle_time: 0s
      min_cooling_off_time: 0s
      min_cooling_run_time: 0s
      min_fan_mode_switching_time: 0s
      idle_action:
        then:
          - script.execute: minuet_thermostat_update
      cool_action:
        then:
          - script.execute: minuet_thermostat_update
      fan_mode_auto_action:
        then:
          - script.execute: minuet_thermostat_update
      fan_mode_off_action:
        then:
          - script.execute: minuet_thermostat_update
      off_mode:
        then:
          - lambda: |-
              id(minuet_thermostat_reset_override).execute();
              if (!minuet::is_transient_operation()) {
                // When the user turns off the thermostat, turn the fan off but restore its
                // last commanded speed and direction.
                const auto& state = minuet::persistent_state();
                id(minuet_fan_set).execute(false, state.fan_speed, state.fan_exhaust,
                      /*suppress_cover_movement*/ false, /*force*/ false);
              }
      on_state:
        then:
          - script.execute: minuet_thermostat_update
      preset:
        - name: 'Default'
          mode: 'off'
          fan_mode: 'auto'
          default_target_temperature_high: ${minuet_thermostat_default_temperature} °F
      default_preset: 'Default'
      on_boot_restore_from: memory
```

### Anything in the logs that might be useful for us?

```txt
[19:10:07][D][climate:011]: 'Thermostat' - Setting
[19:10:07][W][climate:085]:   Preset NONE is not supported by this device!
[19:10:07][D][climate:015]:   Mode: OFF
[19:10:07][D][climate:024]:   Fan: AUTO
[19:10:07][D][climate:040]:   Target Temperature: 25.56
[19:10:07][D][climate:396]: 'Thermostat' - Sending state:
[19:10:07][D][climate:399]:   Mode: OFF
[19:10:07][D][climate:401]:   Action: OFF
[19:10:07][D][climate:404]:   Fan Mode: AUTO
[19:10:07][D][climate:419]:   Current Temperature: nan°C
[19:10:07][D][climate:425]:   Target Temperature: 25.56°C
```

ztox, 2025-09-09:
I'm running into the same issue. See also -- Related discussion, specific to the CN105 Mitsubishi heat pump component:  https://github.com/echavet/MitsubishiCN105ESPHome/issues/361

j9brown, 2025-10-05:
@ztox As it happens, I'm planning to use you CN105 component for a different project soon.  I guess I'll prepare a PR for this issue now.