esphome/esphome #10371: PM sensors are unknown

dhover, 2025-08-22:
### The problem

Since 2025.8.0 the PM sensors from luftdaten are unknown. The logfile shows the values are updated.

### Which version of ESPHome has the issue?

2025.8.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### YAML Config

```yaml
substitutions:
  devicename: luftdaten
  device_description: "Luftdaten"
  friendly_name: "Luftdaten"
  name: Luftdaten
  ip_address: 192.168.186.14

esphome:
  name: $devicename
  comment: "${device_description}"
  friendly_name: "${friendly_name}"

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    #type: arduino

wifi:
  ssid: !secret wifi_iot_ssid
  password: !secret wifi_iot_password
  fast_connect: on
  use_address: ${ip_address}
  power_save_mode: none
  ap:  
    ssid: ${name} !secret wifi_ap_ssid
    password: !secret wifi_ap_password

logger:
  level: DEBUG
  #baud_rate: 0
  hardware_uart: UART1
  logs:
    esp-idf: ERROR

api:
  encryption:
    key: !secret api_encryption_key

#web_server:
#  port: 80

ota:
  - platform: esphome
    password: !secret ota_password
#  - platform: web_server

http_request:
  useragent: esphome/${devicename}
  timeout: 10s
  verify_ssl: false
[... 309 lines cut ...]
INFO Successfully connected to luftdaten @ 192.168.186.14 in 0.002s
INFO Successful handshake with luftdaten @ 192.168.186.14 in 0.084s
[10:53:50][D][api:144]: Accept 192.168.186.9
[10:53:50][D][api.connection:1343]: Home Assistant 2025.8.3 (192.168.186.9) connected
[10:53:50][D][time:061]: Synchronized time: 2025-08-22 10:53:50
[10:54:20][I][safe_mode:042]: Boot seems successful; resetting boot loop counter
[10:54:20][D][esp32.preferences:142]: Writing 1 items: 0 cached, 1 written, 0 failed
[10:54:25][D][sensor:103]: 'wifi signal': Sending state -57.00000 dBm with 0 decimals of accuracy
[10:55:25][D][sensor:103]: 'wifi signal': Sending state -56.00000 dBm with 0 decimals of accuracy
[10:56:26][D][sensor:103]: 'wifi signal': Sending state -56.00000 dBm with 0 decimals of accuracy
[10:57:25][D][sensor:103]: 'wifi signal': Sending state -57.00000 dBm with 0 decimals of accuracy
[10:58:06][D][sds011:171]: Got PM2.5 Concentration: 0.8 µg/m³, PM10.0 Concentration: 3.4 µg/m³
[10:58:06][D][sensor:103]: '2.5µm': Sending state 0.80000 µg/m³ with 1 decimals of accuracy
[10:58:06][D][sensor:103]: '10.0µm': Sending state 3.40000 µg/m³ with 1 decimals of accuracy
INFO Processing unexpected disconnect from ESPHome API for luftdaten @ 192.168.186.14
WARNING Disconnected from API
INFO Successfully resolved luftdaten @ 192.168.186.14 in 0.000s
INFO Successfully connected to luftdaten @ 192.168.186.14 in 0.003s
INFO Successful handshake with luftdaten @ 192.168.186.14 in 0.089s
[10:59:13][I][safe_mode:042]: Boot seems successful; resetting boot loop counter
[10:59:13][D][esp32.preferences:142]: Writing 1 items: 0 cached, 1 written, 0 failed
[10:59:13][W][component:423]: safe_mode took a long time for an operation (53 ms)
[10:59:13][W][component:424]: Components should block for at most 30 ms
[10:59:15][D][sensor:103]: 'wifi signal': Sending state -56.00000 dBm with 0 decimals of accuracy
```

dhover, 2025-08-22:
After loading the same firmware once more it seems the API connection is lost.

`INFO ESPHome 2025.8.0
INFO Reading configuration /config/esphome/luftdaten.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing luftdaten (board: esp32dev; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32/releases/download/54.03.21-2/platform-espressif32.zip)
--------------------------------------------------------------------------------
INFO Version mismatch for tool-esptoolpy: 1.30000.201119 != 5.0.2
INFO Reinstalling tool-esptoolpy due to version mismatch
INFO Package configuration completed successfully
Tool Manager: Installing https://github.com/pioarduino/registry/releases/download/0.0.1/esptoolpy-v5.0.2.zip
INFO Installing https://github.com/pioarduino/registry/releases/download/0.0.1/esptoolpy-v5.0.2.zip
Unpacking  [####################################]  100%
Tool Manager: tool-esptoolpy@5.0.2 has been installed!
INFO tool-esptoolpy@5.0.2 has been installed!
Tool Manager: Installing file:///root/.platformio/tools/tool-esptoolpy
INFO Installing file:///root/.platformio/tools/tool-esptoolpy
Tool Manager: tool-esptoolpy@5.0.2 has been installed!
INFO tool-esptoolpy@5.0.2 has been installed!
INFO Tool tool-esptoolpy successfully installed
INFO Package configuration completed successfully
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50402.0 (5.4.2) 
 - tool-cmake @ 3.30.2 
 - tool-esp-rom-elfs @ 2024.10.11 
 - tool-esptoolpy @ 5.0.2 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.13.1 
 - tool-scons @ 4.40801.0 (4.8.1) 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp-elf @ 14.2.0+20241119
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.10
|-- ArduinoJson @ 7.4.2
RAM:   [=         ]  11.0% (used 35900 bytes from 327680 bytes)
Flash: [======    ]  55.7% (used 1022562 bytes from 1835008 bytes)
========================= [SUCCESS] Took 7.74 seconds =========================
INFO Successfully compiled program.
INFO Uploading to 192.168.186.14
INFO Connecting to 192.168.186.14 port 3232...
INFO Connected to 192.168.186.14
INFO Uploading /data/build/luftdaten/.pioenvs/luftdaten/firmware.bin (1022960 bytes)
Uploading: [============================================================] 100% Done...

INFO Upload took 5.86 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.186.14 using esphome API
INFO Successfully resolved luftdaten @ 192.168.186.14 in 0.000s
INFO Successfully connected to luftdaten @ 192.168.186.14 in 2.063s
INFO Successful handshake with luftdaten @ 192.168.186.14 in 0.092s
[11:08:00][I][app:200]: ESPHome version 2025.8.0 compiled on Aug 21 2025, 10:07:25
[11:08:00][C][wifi:659]: WiFi:
[11:08:00][C][wifi:442]:   Local MAC: 4C:EB:D6:74:D7:50
[11:08:00][C][wifi:447]:   SSID: [redacted]
[11:08:00][C][wifi:450]:   IP Address: 192.168.186.14
[11:08:00][C][wifi:454]:   BSSID: [redacted]
[11:08:00][C][wifi:454]:   Hostname: 'luftdaten'
[11:08:00][C][wifi:454]:   Signal strength: -56 dB ▂▄▆█
[11:08:00][C][wifi:465]:   Channel: 1
[11:08:00][C][wifi:465]:   Subnet: 255.255.255.0
[11:08:00][C][wifi:465]:   Gateway: 192.168.186.1
[11:08:00][C][wifi:465]:   DNS1: 192.168.186.1
[11:08:00][C][wifi:465]:   DNS2: 0.0.0.0
[11:08:00][C][logger:252]: Logger:
[11:08:00][C][logger:252]:   Max Level: DEBUG
[11:08:00][C][logger:252]:   Initial Level: DEBUG
[11:08:00][C][logger:258]:   Log Baud Rate: 115200
[11:08:00][C][logger:258]:   Hardware UART: UART1
[11:08:00][C][logger:265]:   Task Log Buffer Size: 768
[11:08:00][C][logger:270]:   Level for 'esp-idf': ERROR
[11:08:00][C][i2c.idf:133]: I2C Bus:
[11:08:00][C][i2c.idf:134]:   SDA Pin: GPIO21
[11:08:00][C][i2c.idf:134]:   SCL Pin: GPIO22
[11:08:00][C][i2c.idf:134]:   Frequency: 50000 Hz
[11:08:00][C][i2c.idf:144]:   Recovery: bus successfully recovered
[11:08:00][C][i2c.idf:154]: Results from bus scan:
[11:08:00][C][i2c.idf:160]: Found device at address 0x76
[11:08:00][C][uart.idf:153]: UART Bus 0:
[11:08:00][C][uart.idf:154]:   TX Pin: GPIO17
[11:08:00][C][uart.idf:155]:   RX Pin: GPIO16
[11:08:00][C][uart.idf:157]:   RX Buffer Size: 256
[11:08:00][C][uart.idf:159]:   Baud Rate: 9600 baud
[11:08:00][C][uart.idf:159]:   Data Bits: 8
[11:08:00][C][uart.idf:159]:   Parity: NONE
[11:08:00][C][uart.idf:159]:   Stop bits: 1
[11:08:00][D][sensor:103]: 'temperature': Sending state 18.71555 °C with 1 decimals of accuracy
[11:08:00][D][sensor:103]: 'pressure': Sending state 1018.33795 hPa with 1 decimals of accuracy
[11:08:00][D][sensor:103]: 'humidity': Sending state 55.01172 % with 1 decimals of accuracy
INFO Processing unexpected disconnect from ESPHome API for luftdaten @ 192.168.186.14
WARNING Disconnected from API
INFO Successfully resolved luftdaten @ 192.168.186.14 in 0.000s
INFO Successfully connected to luftdaten @ 192.168.186.14 in 0.003s`

dhover, 2025-08-22:
After making some small changes I see some warning:

```
INFO ESPHome 2025.8.0
INFO Reading configuration /config/esphome/luftdaten.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing luftdaten (board: esp32dev; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32/releases/download/54.03.21-2/platform-espressif32.zip)
--------------------------------------------------------------------------------
INFO Package configuration completed successfully
INFO Package configuration completed successfully
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50402.0 (5.4.2) 
 - tool-cmake @ 3.30.2 
 - tool-esp-rom-elfs @ 2024.10.11 
 - tool-esptoolpy @ 5.0.2 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.13.1 
 - tool-scons @ 4.40801.0 (4.8.1) 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp-elf @ 14.2.0+20241119
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.10
|-- ArduinoJson @ 7.4.2
Compiling .pioenvs/luftdaten/src/main.cpp.o
/config/esphome/luftdaten.yaml: In lambda function:
/config/esphome/luftdaten.yaml:139:58: warning: 'ArduinoJson::V742HB22::JsonArray ArduinoJson::V742HB22::JsonObject::createNestedArray(TChar*) const [with TChar = const char]' is deprecated: use obj[key].to<JsonArray>() instead [-Wdeprecated-declarations]
  139 |             JsonArray sensordatavalues = root.createNestedArray("sensordatavalues");
      |                                    ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
In file included from .piolibdeps/luftdaten/ArduinoJson/src/ArduinoJson.hpp:39,
                 from .piolibdeps/luftdaten/ArduinoJson/src/ArduinoJson.h:9,
                 from src/esphome/components/json/json_util.h:11,
                 from src/esphome/core/string_ref.h:11,
                 from src/esphome/core/entity_base.h:5,
                 from src/esphome/components/sensor/sensor.h:4,
                 from src/esphome/core/application.h:29,
                 from src/esphome/components/api/api_frame_helper.h:12,
                 from src/esphome/components/api/api_connection.h:5,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
.piolibdeps/luftdaten/ArduinoJson/src/ArduinoJson/Object/JsonObject.hpp:197:13: note: declared here
  197 |   JsonArray createNestedArray(TChar* key) const {
      |             ^~~~~~~~~~~~~~~~~
/config/esphome/luftdaten.yaml:140:60: warning: 'ArduinoJson::V742HB22::JsonObject ArduinoJson::V742HB22::JsonArray::createNestedObject() const' is deprecated: use add<JsonObject>() instead [-Wdeprecated-declarations]
  140 |             JsonObject pm10 = sensordatavalues.createNestedObject();
      |                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
In file included from .piolibdeps/luftdaten/ArduinoJson/src/ArduinoJson.hpp:65:
.piolibdeps/luftdaten/ArduinoJson/src/ArduinoJson/compatibility.hpp:140:19: note: declared here
  140 | inline JsonObject JsonArray::createNestedObject() const {
      |                   ^~~~~~~~~
/config/esphome/luftdaten.yaml:143:60: warning: 'ArduinoJson::V742HB22::JsonObject ArduinoJson::V742HB22::JsonArray::createNestedObject() const' is deprecated: use add<JsonObject>() instead [-Wdeprecated-declarations]
  143 |             JsonObject pm25 = sensordatavalues.createNestedObject();
      |                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
.piolibdeps/luftdaten/ArduinoJson/src/ArduinoJson/compatibility.hpp:140:19: note: declared here
  140 | inline JsonObject JsonArray::createNestedObject() const {
      |                   ^~~~~~~~~
/config/esphome/luftdaten.yaml: In lambda function:
/config/esphome/luftdaten.yaml:156:58: warning: 'ArduinoJson::V742HB22::JsonArray ArduinoJson::V742HB22::JsonObject::createNestedArray(TChar*) const [with TChar = const char]' is deprecated: use obj[key].to<JsonArray>() instead [-Wdeprecated-declarations]
  156 |             JsonArray sensordatavalues = root.createNestedArray("sensordatavalues");
      |                                    ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
.piolibdeps/luftdaten/ArduinoJson/src/ArduinoJson/Object/JsonObject.hpp:197:13: note: declared here
  197 |   JsonArray createNestedArray(TChar* key) const {
      |             ^~~~~~~~~~~~~~~~~
/config/esphome/luftdaten.yaml:157:60: warning: 'ArduinoJson::V742HB22::JsonObject ArduinoJson::V742HB22::JsonArray::createNestedObject() const' is deprecated: use add<JsonObject>() instead [-Wdeprecated-declarations]
  157 |             JsonObject pm10 = sensordatavalues.createNestedObject();
      |                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
.piolibdeps/luftdaten/ArduinoJson/src/ArduinoJson/compatibility.hpp:140:19: note: declared here
  140 | inline JsonObject JsonArray::createNestedObject() const {
      |                   ^~~~~~~~~
[... 140 lines cut ...]
Info: ota_data_initial.bin not found - skipping
Info: luftdaten.bin not found - skipping
Using FLASH_EXTRA_IMAGES from PlatformIO environment
Merging binaries into /data/build/luftdaten/.pioenvs/luftdaten/firmware.factory.bin
Merging binaries with esptool
SHA digest in image updated.
Wrote 0x109c00 bytes to file '/data/build/luftdaten/.pioenvs/luftdaten/firmware.factory.bin', ready to flash to offset 0x0.
Successfully created /data/build/luftdaten/.pioenvs/luftdaten/firmware.factory.bin
esp32_copy_ota_bin([".pioenvs/luftdaten/firmware.bin"], [".pioenvs/luftdaten/firmware.elf"])
[... cut ...]
INFO Connected to 192.168.186.14
INFO Uploading /data/build/luftdaten/.pioenvs/luftdaten/firmware.bin (1022976 bytes)
Uploading: [============================================================] 100% Done...

INFO Upload took 5.56 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.186.14 using esphome API
INFO Successfully resolved luftdaten @ 192.168.186.14 in 0.000s
INFO Successfully connected to luftdaten @ 192.168.186.14 in 4.139s
```

dhover, 2025-08-24:
Sensors are still now and then unknown or unavailable.

ssieb, 2025-09-13:
Try with the latest release and see if it was fixed by that linked PR.

dhover, 2025-09-14:
With the latest ESPhome version the disconnects are not solved.Also the status of the two PM sensors is still sometimes unknown.

`INFO ESPHome 2025.8.4
INFO Reading configuration /config/esphome/luftdaten.yaml...
INFO Starting log output from 192.168.186.14 using esphome API
INFO Successfully resolved luftdaten @ 192.168.186.14 in 0.000s
INFO Successfully connected to luftdaten @ 192.168.186.14 in 0.010s
INFO Successful handshake with luftdaten @ 192.168.186.14 in 0.089s
[08:21:28][I][app:200]: ESPHome version 2025.8.4 compiled on Sep 10 2025, 10:15:15
[08:21:28][C][wifi:661]: WiFi:
[08:21:28][C][wifi:444]:   Local MAC: 4C:EB:D6:74:D7:50
[08:21:28][C][wifi:449]:   SSID: [redacted]
[08:21:28][C][wifi:452]:   IP Address: 192.168.186.14
[08:21:28][C][wifi:456]:   BSSID: [redacted]
[08:21:28][C][wifi:456]:   Hostname: 'luftdaten'
[08:21:28][C][wifi:456]:   Signal strength: -52 dB ▂▄▆█
[08:21:28][C][wifi:467]:   Channel: 1
[08:21:28][C][wifi:467]:   Subnet: 255.255.255.0
[08:21:28][C][wifi:467]:   Gateway: 192.168.186.1
[08:21:28][C][wifi:467]:   DNS1: 192.168.186.1
[08:21:28][C][wifi:467]:   DNS2: 0.0.0.0
[08:21:28][C][logger:252]: Logger:
[08:21:28][C][logger:252]:   Max Level: DEBUG
[08:21:28][C][logger:252]:   Initial Level: DEBUG
[08:21:28][C][logger:258]:   Log Baud Rate: 115200
[08:21:28][C][logger:258]:   Hardware UART: UART1
[08:21:28][C][logger:265]:   Task Log Buffer Size: 768
[08:21:28][C][logger:270]:   Level for 'esp-idf': ERROR
[08:21:28][C][i2c.idf:081]: I2C Bus:
[08:21:28][C][i2c.idf:082]:   SDA Pin: GPIO21
[08:21:28][C][i2c.idf:082]:   SCL Pin: GPIO22
[08:21:28][C][i2c.idf:082]:   Frequency: 50000 Hz
[08:21:28][C][i2c.idf:092]:   Recovery: bus successfully recovered
[08:21:28][C][i2c.idf:102]: Results from bus scan:
[08:21:28][C][i2c.idf:108]: Found device at address 0x76
[08:21:28][C][uart.idf:153]: UART Bus 0:
[08:21:28][C][uart.idf:154]:   TX Pin: GPIO17
[08:21:28][C][uart.idf:155]:   RX Pin: GPIO16
[08:21:28][C][uart.idf:157]:   RX Buffer Size: 256
[08:21:28][C][uart.idf:159]:   Baud Rate: 9600 baud
[08:21:28][C][uart.idf:159]:   Data Bits: 8
[08:21:28][C][uart.idf:159]:   Parity: NONE
[08:21:28][C][uart.idf:159]:   Stop bits: 1
[08:21:28][C][homeassistant.time:010]: Home Assistant Time:
[08:21:28][C][homeassistant.time:010]:   Timezone: 'CET-1CEST,M3.5.0,M10.5.0/3'
[08:21:28][C][sds011:070]: SDS011:
[08:21:28][C][sds011:070]:   Update Interval: 5 min
[08:21:28][C][sds011:070]:   RX-only mode: OFF
[08:21:28][C][sds011:075]:   PM2.5 'PM 2.5'
[08:21:28][C][sds011:075]:     State Class: 'measurement'
[08:21:28][C][sds011:075]:     Unit of Measurement: 'µg/m³'
[08:21:28][C][sds011:075]:     Accuracy Decimals: 1
[08:21:28][C][sds011:075]:     Device Class: 'pm25'
[08:21:28][C][sds011:075]:     Icon: 'mdi:chemical-weapon'
[08:21:28][C][sds011:076]:   PM10.0 'PM 10'
[08:21:28][C][sds011:076]:     State Class: 'measurement'
[08:21:28][C][sds011:076]:     Unit of Measurement: 'µg/m³'
[08:21:28][C][sds011:076]:     Accuracy Decimals: 1
[08:21:28][C][sds011:076]:     Device Class: 'pm10'
[08:21:28][C][sds011:076]:     Icon: 'mdi:chemical-weapon'
[08:21:28][C][bme280_i2c.sensor:025]:   Address: 0x76
[08:21:28][C][bme280.sensor:180]: BME280:
[08:21:28][C][bme280.sensor:192]:   IIR Filter: OFF
[08:21:28][C][bme280.sensor:193]:   Update Interval: 300.0s
[08:21:28][C][bme280.sensor:195]:   Temperature 'Temperature'
[08:21:28][C][bme280.sensor:195]:     State Class: 'measurement'
[08:21:28][C][bme280.sensor:195]:     Unit of Measurement: '°C'
[08:21:28][C][bme280.sensor:195]:     Accuracy Decimals: 1
[08:21:28][C][bme280.sensor:195]:     Device Class: 'temperature'
[08:21:28][C][bme280.sensor:196]:     Oversampling: 16x
[... 44 lines cut ...]
INFO Processing unexpected disconnect from ESPHome API for luftdaten @ 192.168.186.14
WARNING Disconnected from API
INFO Successfully resolved luftdaten @ 192.168.186.14 in 0.000s
INFO Successfully connected to luftdaten @ 192.168.186.14 in 0.003s
INFO Successful handshake with luftdaten @ 192.168.186.14 in 1.209s
[08:22:30][E][component:323]: http_request cleared Error flag
[08:23:18][I][safe_mode:042]: Boot seems successful; resetting boot loop counter
[08:23:18][D][esp32.preferences:142]: Writing 1 items: 0 cached, 1 written, 0 failed
[08:23:21][D][sensor:103]: 'Wifi signal': Sending state -52.00000 dBm with 0 decimals of accuracy
[08:24:21][D][sensor:103]: 'Wifi signal': Sending state -51.00000 dBm with 0 decimals of accuracy
[08:25:21][D][sensor:103]: 'Wifi signal': Sending state -51.00000 dBm with 0 decimals of accuracy
[08:26:21][D][sensor:103]: 'Wifi signal': Sending state -53.00000 dBm with 0 decimals of accuracy
[08:26:55][D][sds011:171]: Got PM2.5 Concentration: 16.0 µg/m³, PM10.0 Concentration: 35.2 µg/m³
[08:26:55][D][sensor:103]: 'PM 2.5': Sending state 16.00000 µg/m³ with 1 decimals of accuracy
[08:26:55][D][sensor:103]: 'PM 10': Sending state 35.20000 µg/m³ with 1 decimals of accuracy
[08:26:56][D][http_request.idf:043]: Received response header, name: content-type, value: application/json
[08:26:57][D][http_request.idf:043]: Received response header, name: content-type, value: text/html; charset=UTF-8
INFO Processing unexpected disconnect from ESPHome API for luftdaten @ 192.168.186.14
WARNING Disconnected from API
INFO Successfully resolved luftdaten @ 192.168.186.14 in 0.000s
INFO Successfully connected to luftdaten @ 192.168.186.14 in 0.002s
INFO Successful handshake with luftdaten @ 192.168.186.14 in 0.081s
[08:28:04][I][safe_mode:042]: Boot seems successful; resetting boot loop counter
[08:28:04][D][esp32.preferences:142]: Writing 1 items: 0 cached, 1 written, 0 failed
[08:28:12][D][sensor:103]: 'Wifi signal': Sending state -51.00000 dBm with 0 decimals of accuracy`

ssieb, 2025-09-14:
You'll need to get serial logs to see why it's disconnecting.

dhover, 2025-09-14:
Ok, I'm not an expert. How do I get those serial.logs?

dhover, 2025-09-14:
Ok I understand. This will take a while. End of September I"ll post the log.

dhover, 2025-10-06:
This is de serial logging after the first disconnect.

`=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2025.10.06 16:09:31 =~=~=~=~=~=~=~=~=~=~=~=
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7112
load:0x40078000,len:15624
load:0x40080400,len:4
load:0x40080404,len:3876
entry 0x4008064c
I (29) boot: ESP-IDF 5.1.5 2nd stage bootloader
I (29) boot: compile time May 20 2025 22:23:30
I (29) boot: Multicore bootloader
I (33) boot: chip revision: v1.0
I (37) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (46) boot.esp32: SPI Flash Size : 4MB
I (51) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 otadata          OTA data         01 00 00009000 00002000
I (75) boot:  1 phy_init         RF data          01 01 0000b000 00001000
I (82) boot:  2 app0             OTA app          00 10 00010000 001c0000
I (89) boot:  3 app1             OTA app          00 11 001d0000 001c0000
I (97) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
I (104) boot: End of partition table
I (109) esp_image: segment 0: paddr=001d0020 vaddr=3f400020 size=274e0h (160992) map
I (175) esp_image: segment 1: paddr=001f7508 vaddr=3ff80000 size=0001ch (    28) load
I (176) esp_image: segment 2: paddr=001f752c vaddr=3ffb0000 size=03f04h ( 16132) load
I (188) esp_image: segment 3: paddr=001fb438 vaddr=40080000 size=04be0h ( 19424) load
I (198) esp_image: segment 4: paddr=00200020 vaddr=400d0020 size=b8758h (755544) map
I (471) esp_image: segment 5: paddr=002b8780 vaddr=40084be0 size=12150h ( 74064) load
I (513) boot: Loaded app from partition at offset 0x1d0000
I (513) boot: Disabling RNG early entropy source...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7112
load:0x40078000,len:15624
load:0x40080400,len:4
load:0x40080404,len:3876
entry 0x4008064c
I (29) boot: ESP-IDF 5.1.5 2nd stage bootloader
I (29) boot: compile time May 20 2025 22:23:30
I (29) boot: Multicore bootloader
I (33) boot: chip revision: v1.0
I (37) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (46) boot.esp32: SPI Flash Size : 4MB
I (51) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 otadata          OTA data         01 00 00009000 00002000
I (75) boot:  1 phy_init         RF data          01 01 0000b000 00001000
I (82) boot:  2 app0             OTA app          00 10 00010000 001c0000
I (89) boot:  3 app1             OTA app          00 11 001d0000 001c0000
I (97) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
I (104) boot: End of partition table
I (109) esp_image: segment 0: paddr=001d0020 vaddr=3f400020 size=274e0h (160992) map
I (175) esp_image: segment 1: paddr=001f7508 vaddr=3ff80000 size=0001ch (    28) load
I (176) esp_image: segment 2: paddr=001f752c vaddr=3ffb0000 size=03f04h ( 16132) load
I (188) esp_image: segment 3: paddr=001fb438 vaddr=40080000 size=04be0h ( 19424) load
I (198) esp_image: segment 4: paddr=00200020 vaddr=400d0020 size=b8758h (755544) map
I (471) esp_image: segment 5: paddr=002b8780 vaddr=40084be0 size=12150h ( 74064) load
I (513) boot: Loaded app from partition at offset 0x1d0000
I (513) boot: Disabling RNG early entropy source...
`

LorenzoCattai, 2025-11-25:
Try to update `Home Assistant Core`, in `Home Assistant` go to `Settings` -> `System` -> `Updates`, it could be that you are running a version that is quite old, I had this exact problem and this fixed it for me.