esphome/esphome #10403: Hostname isn't set to match the YAML config file.

ngist, 2025-08-24:
### The problem

I'm seeing the same issue that occurred here(https://github.com/esphome/issues/issues/6014), with some minor differences I'm using ethernet instead of wifi.

I pretty much copied the latest config from here and made a couple tweeks:
https://github.com/esphome/bluetooth-proxies/blob/main/olimex/olimex-esp32-poe-iso.yaml

It's my first time setting up this device so I'm not sure if this is new or a long standing issue.

Anyway the config has ble-proxy-1 in it, and ESPHome looks for it at ble-proxy-1.local, however looking at tcpdump I can see the device identifying as ble-proxy-1-38e690, and it doesn't show up in home assistant and says it's offline, and I can't connect. If i modify the config to match the host without reuploading then it starts working.  

The workaround is easy if you know the problem.

### Which version of ESPHome has the issue?

2025.08.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

ethernet

### YAML Config

```yaml
substitutions:
  name: ble-proxy-1-38e690
  friendly_name: Bluetooth Proxy Front

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  min_version: 2025.7.0
  name_add_mac_suffix: true

esp32:
  board: esp32-poe-iso
  framework:
    type: esp-idf

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk:
    pin: GPIO17
    mode: CLK_OUT
  phy_addr: 0
  power_pin:
    number: GPIO12
    ignore_strapping_warning: true

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xxx"

ota:
  - platform: esphome
    password: "xxx"

esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

bluetooth_proxy:
  active: true
  connection_slots: 3

button:
  - platform: safe_mode
    id: button_safe_mode
    name: Safe Mode Boot

  - platform: factory_reset
    id: factory_reset_btn
    name: Factory reset
```

### Anything in the logs that might be useful for us?

```txt

```

ssieb, 2025-08-24:
Your description and the config don't match.  Even worse, you also have the mac suffix enabled in the config.
It sounds like you flashed the default proxy firmware from the website and then created a config in the builder and somehow expected it to match.  Did you install the config that had `ble-proxy-1` to the device?

ngist, 2025-08-24:
Let me explain more clearly.

I compiled a config and then uploaded it to the device. The compiled config was identical to what I attached here except that that name listed 'ble-proxy-1' instead of 'ble-proxy-1-38e690'. The device showed offline and I could not connect to access logs.

I then changed the name to 'ble-proxy-1-38e690' in the config, but did **not** recompile or re-uploaded to the device. Then the device reports online, and I was able to connect and access logs.

I hadn't noticed the `name_add_mac_suffix: true` setting before, which would now seem to explain why the random digits are added to the hostname. I would note though I didn't come up with it, it's in the example config here https://github.com/esphome/bluetooth-proxies/blob/main/olimex/olimex-esp32-poe-iso.yaml is it deprecated or something?

It does seems strange to me that this option prevents esphome from connecting for logs, is seems like the device makes use of the flag but the ESPHome UI doesn't. I'll try removing this option as I plan to manually name devices with my own 1,2,3,... suffixes. It also doesn't seem to be a MAC suffix, as my router doesn't report this in the MAC anywhere, and even with some basic bit manipulations I can't get it to match.

ssieb, 2025-08-24:
The default config has it, but when you "take control" of a device, the dashboard will disable the option.
The dashboard has no way to know what the device mac address is, so it can't know what the name is.  You should not turn on that option for devices that you're configuring yourself.  I think the suffix should be the last part of the mac address.

ngist, 2025-08-25:
> The default config has it, but when you "take control" of a device, the dashboard will disable the option. The dashboard has no way to know what the device mac address is, so it can't know what the [...]

Got it, so if I had programmed without credentials then I could have "taken control", and it would have setup the credentials and removed this flag. If configuring it manually I should remove the flag. 

> The Ethernet MAC address will be a couple of digits higher than the MAC suffix appended to the name. The MAC we suffix is the Wi-Fi/base MAC address

Thanks for the clarification, I was looking at the wrong MAC address, I was looking at 192.168.1.6 instead of 192.168.2.6.

I'm going to close as it looks like this was all user error after all. Thanks for the help!

closed 2025-08-25 by ngist