esphome/esphome #10432: Little/big endian issue

HennieBosch, 2025-08-25:
### The problem

After compiling exisitng ESPHome yaml configuration, I successfully compile but then experience multiple linking errors stating the file's endian does not match the target's endian. I tried my other ESPHome devices without editing the yaml and experience the same problem

### Which version of ESPHome has the issue?

2025.8.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### YAML Config

```yaml
esphome:
  name: winestore
  friendly_name: Winestore

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "zzzzzz"

ota:
  - platform: esphome
    password: "zzzz"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Winestore Fallback Hotspot"
    password: "zzzz"

captive_portal:

binary_sensor:
  - platform: gpio
    pin: 
      number: 23
      mode:
        input: true
        pullup: True
    name: Door Switch

output:
  - platform: gpio
    pin: 2
    id: 'generic_out'

switch:
  - platform: output
    name: "Generic Output"
    output: 'generic_out'

one_wire:
  - platform: gpio
    pin: 15
    
sensor:
  - platform: dallas_temp  
    name: temperature
    update_interval: 10s

remote_transmitter:
  pin: 26
  carrier_duty_percent: 50%

button:
  - platform: template
    name: "On/Off Samsung"
    on_press:
      - remote_transmitter.transmit_samsung:
          data: 0x00FF
```

### Anything in the logs that might be useful for us?

```txt
INFO Package configuration completed successfully
INFO Package configuration completed successfully
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-arduinoespressif32 @ 3.2.1 
 - framework-arduinoespressif32-libs @ 5.4.0+sha.858a988d6e 
 - tool-esptoolpy @ 5.0.2 
 - tool-mklittlefs @ 3.2.0 
 - toolchain-xtensa-esp-elf @ 14.2.0+20241119
Dependency Graph
|-- Networking @ 3.2.1
|-- AsyncTCP @ 3.4.5
|-- WiFi @ 3.2.1
|-- FS @ 3.2.1
|-- Update @ 3.2.1
|-- ESPAsyncWebServer @ 3.7.10
|-- ESP32 Async UDP @ 3.2.1
|-- DNSServer @ 3.2.1
|-- ESPmDNS @ 3.2.1
|-- noise-c @ 0.1.10
Linking .pioenvs/winestore/firmware.elf
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld: /data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/lib/crt0.o: compiled for a big endian system and target is little endian
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld: failed to merge target specific data of file /data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/lib/crt0.o
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld: /data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/lib/libm.a(libm_a-w_exp.o): compiled for a big endian system and target is little endian
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld: failed to merge target specific data of file /data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/lib/libm.a(libm_a-w_exp.o)
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld: /data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/lib/libm.a(libm_a-w_fmod.o): compiled for a big endian system and target is little endian
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld: failed to merge target specific data of file /data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/lib/libm.a(libm_a-w_fmod.o)
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld: /data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/lib/libm.a(libm_a-w_log.o): compiled for a big endian system and target is little endian
```

swoboda1337, 2025-08-25:
What device are you building on?

HennieBosch, 2025-08-26:
Removing the ESPHome add-on and re-installing it worked - thank you

closed 2025-08-26 by HennieBosch