esphome/esphome #10433: Failed compiling LD2410

BorisFR, 2025-08-25:
### Version

2025.8.0

### What type of installation are you using?

Home Assistant Add-on

### Browser

Edge

### What happened?

`INFO ESPHome 2025.8.0
INFO Reading configuration /config/esphome/presencechambre.yaml...
INFO Detected timezone 'Europe/Paris'
INFO Generating C++ source...
INFO Compiling app...
Processing presencechambre (board: lolin_s2_mini; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32/releases/download/54.03.21-2/platform-espressif32.zip)
--------------------------------------------------------------------------------
INFO Package configuration completed successfully
INFO Package configuration completed successfully
HARDWARE: ESP32S2 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50402.0 (5.4.2) 
 - tool-cmake @ 3.30.2 
 - tool-esp-rom-elfs @ 2024.10.11 
 - tool-esptoolpy @ 5.0.2 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.13.1 
 - tool-scons @ 4.40801.0 (4.8.1) 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp-elf @ 14.2.0+20241119
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.10
|-- ArduinoJson @ 7.4.2
Compiling .pioenvs/presencechambre/src/esphome/components/ld2410/number/gate_threshold_number.cpp.o
Compiling .pioenvs/presencechambre/src/esphome/components/ld2410/number/max_distance_timeout_number.cpp.o
Compiling .pioenvs/presencechambre/src/esphome/components/ld2410/select/baud_rate_select.cpp.o
Compiling .pioenvs/presencechambre/src/esphome/components/ld2410/select/distance_resolution_select.cpp.o
src/esphome/components/ld2410/number/gate_threshold_number.cpp: In member function 'virtual void esphome::ld2410::GateThresholdNumber::control(float)':
src/esphome/components/ld2410/number/gate_threshold_number.cpp:10:18: error: 'class esphome::ld2410::LD2410Component' has no member named 'set_gate_threshold'
   10 |   this->parent_->set_gate_threshold(this->gate_);
      |                  ^~~~~~~~~~~~~~~~~~
src/esphome/components/ld2410/number/max_distance_timeout_number.cpp: In member function 'virtual void esphome::ld2410::MaxDistanceTimeoutNumber::control(float)':
src/esphome/components/ld2410/number/max_distance_timeout_number.cpp:8:18: error: 'class esphome::ld2410::LD2410Component' has no member named 'set_max_distances_timeout'
    8 |   this->parent_->set_max_distances_timeout();
      |                  ^~~~~~~~~~~~~~~~~~~~~~~~~
*** [.pioenvs/presencechambre/src/esphome/components/ld2410/number/gate_threshold_number.cpp.o] Error 1
*** [.pioenvs/presencechambre/src/esphome/components/ld2410/number/max_distance_timeout_number.cpp.o] Error 1
========================== [FAILED] Took 3.96 seconds ==========================`

### How to reproduce

Create a component with LD2410

### Expected behavior

Compile must complete

### Relevant log output

```shell

```

BorisFR, 2025-08-25:
I have used them initialy but remove detail to have lower transmission of data.
I now removed the **number** and **select** section completly and it is now ok! 
**Thanks a lot!**

```
#select:
#  - platform: ld2410
    #distance_resolution:
    #  name: "Distance resolution"
    #light_function:
    #  name: Light function
#number:
#  - platform: ld2410
    #timeout:
    #  name: timeout
    #light_threshold:
    #  name: light threshold
    #max_move_distance_gate:
    #  name: max move distance gate
    #max_still_distance_gate:
    #  name: max still distance gate
    #g0:
    #  move_threshold:
    #    name: g0 move threshold
    #  still_threshold:
    #    name: g0 still threshold
...

ssieb, 2025-08-25:
I'll leave it open.  It shouldn't be crashing the compile anyway.

PR #10436 (2025-08-25): [core] Dont copy platform source files if there are no entities of that type
# What does this implement/fix?

Optimizations in #10023 meant that the entity defines eg `USE_NUMBER` are now only defined if there is actually an entity to be added. This meant that the platform files for components were being copied to the project, but parts were pre-processed out. 

Rather than put `#ifdef USE_NUMBER` inside platform files, we can simply not copy the source files if there is no chance they are going to be used.

## Types of changes

- [x] Bugfix (non-breaking change which fixes an issue)
- [ ] New [... cut]
closed 2025-08-25 by kbx81