esphome/esphome #10434: LVGL Image Color Channels swapped

ss89, 2025-08-25:
### The problem

Hello,

I am running into an issue where the color channels R and B are swapped, which results in red images looking blue, yellow looks cyan and green is fine.
I have tried out changing `byte_order` to `little_endian`, but that shifted the channels only - so not the solution.
Then I have tried enabling border colors for the lvgl widgets and they are shown correctly, so i assume it must be something with the conversion of the png to the internal RGB565 format.
Here's the demo pictures i used.
They are plain PNG files generated with GIMP 3:
[image]
[image]
[image]

### Which version of ESPHome has the issue?

2025.8.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

Either lvgl or image

### YAML Config

```yaml
esphome:
  name: display-test
  friendly_name: display-test

esp32:
  board: esp32-c6-devkitc-1
  framework:
    type: esp-idf

# Enable logging
logger:

spi:
  clk_pin: GPIO6
  mosi_pin: GPIO7
  miso_pin: GPIO4

text:
  - platform: template
    id: status
    name: "Status"
    min_length: 0
    max_length: 20
    mode: text
    lambda: |-
      return id(status).state;
    update_interval: 5s
    set_action: 
      then:
        - lambda: |-
            id(status).publish_state(x);
        - lvgl.widget.refresh: status_widget

output:
  - platform: ledc
    pin: GPIO5
    #level: 0.25
    id: display_brightness
    inverted: false
    min_power: 0.0
    max_power: 1.0

number:
  - platform: template
    name: "Display Brightness"
    optimistic: true
    min_value: 0
    max_value: 100
    step: 1
    initial_value: 25
    restore_value: True
    on_value:
      then:
        - output.set_level:
            id: display_brightness
            level: !lambda "return x/100;"

image:
  - file: images/green-square.png
    id: green_square
    resize: 22x22
    type: RGB565
  - file: images/yellow-square.png
    id: yellow_square
    resize: 22x22
    type: RGB565
  - file: images/red-square.png
    id: red_square
    resize: 22x22
    type: RGB565

lvgl:
  default_font: montserrat_20

  pages:
    - id: main_page
      
      widgets:
        - obj:
            width: 240
            height: 320
            bg_opa: 0
            border_opa: 0
            layout: # enable the GRID layout for the children widgets
              type: GRID # split the rows and the columns proportionally
              grid_columns: [FR(108), FR(24), FR(108)] # equal
              grid_rows: [FR(7), FR(7), FR(7), FR(7), FR(7), FR(7), FR(7), FR(7), FR(7), FR(7), FR(7), FR(7), FR(7)] # like percents
              pad_column: 0
            widgets:

              - label:
                  text: "My label"
                  grid_cell_column_pos: 0
                  grid_cell_row_pos: 0
                  grid_cell_x_align: STRETCH
                  grid_cell_y_align: START
              - image:
                  id: status_widget
                  src: !lambda |-
                    auto circle = id(red_square);
                    auto status = id(status).state.c_str();
                    ESP_LOGD("lambda", "checking %s", status);
                    if (strcmp(status, "yellow") == 0){
                      ESP_LOGD("lambda", "found to be yellow :-/");
                      circle = id(yellow_square);
                    }
                    else if (strcmp(status, "green") == 0){
                      ESP_LOGD("lambda", "found to be green :-)");
                      circle = id(green_square);
                    } else {
                      ESP_LOGD("lambda", "found to be red :-())");
                    }
                    return circle;
                  border_color: 0x0
                  border_side: LEFT
                  border_width: 1
                  grid_cell_column_pos: 1
                  grid_cell_row_pos: 0
                  grid_cell_x_align: CENTER
                  grid_cell_y_align: START
              - label:
                  text: "Status"
                  grid_cell_column_pos: 2
                  grid_cell_row_pos: 0
                  grid_cell_x_align: STRETCH
                  grid_cell_y_align: START

display:
  - platform: mipi_spi
    id: lcd_display
    auto_clear_enabled: false
    update_interval: never
    model: ILI9341
    cs_pin: GPIO10
    dc_pin: GPIO0
    reset_pin: GPIO1
    #enable_pin: GPIO5
    dimensions:
      height: 320
      width: 240
    rotation: 90
```

### Anything in the logs that might be useful for us?

```txt
Nope, nothing i have seen.
```

ss89, 2025-08-26:
@clydebarrow the color order is fine, as i mentioned that the border colors for components are indeed correct. Please re-open.

clydebarrow, 2025-08-26:
You'll need to provide the images you are using then, there's obviously something wrong with them. Have you tried other images, other than ones you created yourself?

ss89, 2025-08-26:
Yes, i have tried other PNGs (3) from search engines. They all show this behavior.

I will try the test card thing later today, however if the borders of components show the right color then i do not see why it should be the color_order.