esphome/esphome #10435: After updating to:  2025.8.0  missing package.json

stefancn, 2025-08-25:
### The problem

INFO ESPHome 2025.8.0
INFO Reading configuration /config/esphome/sureflap.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing sureflap (board: esp32-c6-devkitc-1; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32/releases/download/54.03.21-2/platform-espressif32.zip)
--------------------------------------------------------------------------------
INFO Package configuration completed successfully
MissingPackageManifestError: Could not find one of 'package.json' manifest files in the package

### Which version of ESPHome has the issue?

2025-8.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

stefancn, 2025-08-27:
Thanks, good to know this workaround if the normal upgrade path ever will be fixed. Hope tou that a proper patch/upgrade will come in the near future...

ssieb, 2025-08-28:
It's an issue with platformio that esphome doesn't have any control over.

closed 2025-08-28 by ssieb