esphome/esphome #10440: WiFi doesn't connect with hidden SSID

marcellogentile, 2025-08-25:
### The problem

Upgrading to 2025.8.0 breaks WiFi connect with hidden SSID.  

### Which version of ESPHome has the issue?

2025.8.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

ssieb, 2025-08-25:
You should have `fast_connect` or `hidden` set if you're using a hidden SSID.  I thought the fallback for checking for hidden SSIDs has been fixed, but I can't find a PR for it.

ssieb, 2025-08-25:
Can you get serial logs?

marcellogentile, 2025-08-25:
> ssieb
>  left a comment
> (esphome/esphome#10440)
> Can you get serial logs?

17:35][E][wifi:383]: wifi_sta_connect_ failed

[14:17:35][D][wifi:767]: Retrying with hidden networks

[14:17:40][W][wifi_esp32:464]: esp_wifi_connect failed: ESP_ERR_WIFI_SSID

[14:17:40][E][wifi:383]: wifi_sta_connect_ failed

[14:17:40][W][wifi:128]: Restarting adapter

ssieb, 2025-08-25:
Try with verbose log level.

closed 2025-08-26 by ssieb