esphome/esphome #10448: Logger Warnung at Esp32 deep sleep

CrazyUs3r, 2025-08-26:
### The problem

Hello 

I have update to 8.1 and become after wake up from deep sleep this logger error

Logger: aioesphomeapi.reconnect_logic
Quelle: runner.py:156
Erstmals aufgetreten: 13:13:12 (2 Vorkommnisse)
Zuletzt protokolliert: 13:47:21

Can't connect to ESPHome API for wetterstation @ 192.168.178.233: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.178.233', port=6053))]: [Errno 113] Connect call failed ('192.168.178.233', 6053) (SocketAPIError)

### Which version of ESPHome has the issue?

2025.8.1 

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

ssieb, 2025-08-26:
You need to provide more info.  Why is this a problem?  HA isn't able to connect to the device at that point.  Is that before or after the device wakes up?

CrazyUs3r, 2025-08-27:
Sure, the warning comes after the ESP goes into deep sleep about 30 seconds after disconnection. The deep sleep configuration is as in the documentation.

ssieb, 2025-08-27:
Why is this a problem?

ssieb, 2025-08-27:
esphome does do an MDNS broadcast when it connects to wifi, so HA will connect immediately.  I don't know if HA is supposed to stop trying to connect if it knows the ESP is sleeping.
Do the device entities keep their state or go to unavailable?

bdraco, 2025-08-27:
Thank you for reporting this issue. This is a Home Assistant integration issue, not an ESPHome firmware issue.

**Please open an issue with Home Assistant Core:** https://github.com/home-assistant/core/issues/new?template=bug_report.yml

**In your Home Assistant issue, please include:**

1. **Where you're seeing these warnings** (Home Assistant logs, ESPHome logs, or elsewhere)

2. **Debug logs for the ESPHome integration** - Enable using either method:
   - **Via UI:** Settings > Devices & services > ESPHome > (three dots menu) > Enable debug logging
   - **Via configuration.yaml:**
     ```yaml
     logger:
       default: info
       logs:
         aioesphomeapi: debug
     ```
   
   Then follow the guide at: https://www.home-assistant.io/docs/configuration/troubleshooting/#debug-logs-and-diagnostics

3. **Complete logs covering:**
   - Your device preparing for and entering deep sleep
   - Any connection attempts while sleeping  
   - The wake-up and reconnection sequence
   - After reproducing the issue, disable debug logging and download the logs when prompted

4. **Download diagnostics** (if available): ESPHome integration > (three dots menu) > Download diagnostics

5. Your ESPHome YAML configuration (especially the deep_sleep section)

**Important context:** If your ESP32 is cleanly disconnecting before deep sleep, you shouldn't see warnings. These warnings suggest either the device isn't disconnecting cleanly, or Home Assistant isn't handling the disconnection properly.

The `aioesphomeapi` debug logs will help determine exactly what's happening with the API connection during sleep transitions.

Closing this issue as it belongs in the Home Assistant repository. Please link the HA issue here once created. We can reopen if the investigation reveals an ESPHome firmware problem, but that seems unlikely at this point.

closed 2025-08-27 by bdraco

ssieb, 2025-08-29:
That's just the retries.  We need to see what happens when the connection gets closed.

CrazyUs3r, 2025-08-30:
2025-08-29 17:37:57.568 DEBUG (MainThread) [aioesphomeapi.connection] wetterstation @ 192.168.178.233: Got message of type TextSensorStateResponse (len=28): key: 2199499128
state: "Schlafmodus Aktiviert"
2025-08-29 17:37:58.488 DEBUG (MainThread) [aioesphomeapi.connection] wetterstation @ 192.168.178.233: Got message of type DisconnectRequest (len=0): 
2025-08-29 17:37:58.488 DEBUG (MainThread) [aioesphomeapi.connection] wetterstation @ 192.168.178.233: Sending DisconnectResponse: 
2025-08-29 17:37:58.488 DEBUG (MainThread) [aioesphomeapi._frame_helper.base] wetterstation @ 192.168.178.233: Sending frame: [0100143db58f801113061d75a9e6894f4facf69fff6aa5]
2025-08-29 17:37:58.489 DEBUG (MainThread) [aioesphomeapi.connection] Cleaning up connection to wetterstation @ 192.168.178.233
2025-08-29 17:37:58.489 INFO (MainThread) [aioesphomeapi.reconnect_logic] Processing expected disconnect from ESPHome API for wetterstation @ 192.168.178.233
2025-08-29 17:37:58.489 DEBUG (MainThread) [homeassistant.components.esphome.manager] wetterstation: 192.168.178.233 disconnected (expected=True), running disconnected callbacks
2025-08-29 17:37:58.489 DEBUG (MainThread) [aioesphomeapi.reconnect_logic] Scheduling new connect attempt in 5.00 seconds
2025-08-29 17:38:03.490 DEBUG (MainThread) [aioesphomeapi.reconnect_logic] Trying to connect to wetterstation @ 192.168.178.233
2025-08-29 17:38:03.490 DEBUG (MainThread) [aioesphomeapi.reconnect_logic] Connected lock acquired for wetterstation @ 192.168.178.233
2025-08-29 17:38:03.490 DEBUG (MainThread) [aioesphomeapi.reconnect_logic] Starting zeroconf listener for wetterstation
2025-08-29 17:38:03.491 INFO (MainThread) [aioesphomeapi.reconnect_logic] Successfully resolved wetterstation @ 192.168.178.233 in 0.000s
2025-08-29 17:38:03.491 DEBUG (MainThread) [aioesphomeapi.connection] wetterstation @ 192.168.178.233: Connecting to IPv4Sockaddr(address='192.168.178.233', port=6053)
2025-08-29 17:38:17.742 DEBUG (MainThread) [aioesphomeapi.connection] Cleaning up connection to wetterstation @ 192.168.178.233
2025-08-29 17:38:17.742 WARNING (MainThread) [aioesphomeapi.reconnect_logic] Can't connect to ESPHome API for wetterstation @ 192.168.178.233: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.178.233', port=6053))]: [Errno 113] Connect call failed ('192.168.178.233', 6053) (SocketAPIError)

state: 4.79492712
2025-08-29 17:37:57.428 DEBUG (MainThread) [aioesphomeapi.connection] wetterstation @ 192.168.178.233: Got message of type SensorStateResponse (len=10): key: 2326860773