esphome/esphome #10450: bluetooth_proxy boot loop on 2025.8.0 2025.8.1

corrado-c, 2025-08-26:
### The problem

bluetooth_proxy boot loop on 2025.8.0 2025.8.1

yaml was working fine with the latest esphome, then I updated to 2025.8 and the reboot loop started, I tried disabling several things and found that the loop goes away if I disable bluetooth_proxy

### Which version of ESPHome has the issue?

2025.8.x

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

bluetooth_proxy

### YAML Config

```yaml
esphome:
  name: esp32c3bleproxy
  friendly_name: ESP32c3BLEProxy

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf

logger:
  level: DEBUG

api:
  encryption:
    key: !secret esp_api_key

ota:
  platform: esphome
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true 
#  captive_portal:
  
i2c:
  sda: GPIO4
  scl: GPIO3
  scan: false

bluetooth_proxy:
  active: true

esp32_ble_tracker:
  scan_parameters:
    interval: 300ms
    window: 40ms

light:
  - platform: esp32_rmt_led_strip
    chipset: WS2812        # <-- specifica il chipset WS2812 per i NeoPixel
    pin: GPIO8
    num_leds: 1
    rgb_order: GRB
    name: "NeoPixel Light"

sensor:
  - platform: am2320
    address: 0x5C
    temperature:
      name: "AM2320 Temperature"
      id: am_temp_sensor
      accuracy_decimals: 1
    humidity:
      name: "AM2320 Humidity"
      id: am_hum_sensor
      accuracy_decimals: 1
    update_interval: 61s

  - platform: wifi_signal
    name: "WiFi Signal dB c3"
    id: wifi_signal_db
    update_interval: 235s
    entity_category: diagnostic

  - platform: copy
    source_id: wifi_signal_db
    name: "WiFi Signal Percent c3"
    filters:
    #  - lambda: return min(max(2 * (x + 100.0), 0.0), 100.0);
       - lambda: return min(max(2.06 * (x + 100.0), 0.0), 100.0); #filtro migliore
    unit_of_measurement: "%"
    entity_category: diagnostic

  - platform: internal_temperature  # Monitora la temperatura interna dell'ESP32
    name: "ESP32 bleproxy Internal Temperature"
    id: esp32_temperature
    update_interval: 181s
    entity_category: "diagnostic"

text_sensor:
  - platform: wifi_info
    ip_address:
      name: "EspHome Node IP Address"
      
interval:
  - interval: 300s
    then:
      - if:
          condition:
            not:
              wifi.connected:
          then:
            - logger.log: "WiFi perso, riavvio..."
            - button.press: restart_button

button:
  - platform: restart
    id: restart_button
    name: "Restart"
    entity_category: config

switch:
  - platform: template
    name: "Debug Logging"
    id: debug_logging_switch
    turn_on_action:
      - logger.set_level: DEBUG #VERBOSE
      - logger.log: "Logger livello impostato su DEBUG"
    turn_off_action:
      - logger.set_level: ERROR
      - logger.log: "Logger livello impostato su INFO"
```

### Anything in the logs that might be useful for us?

```txt

```

corrado-c, 2025-09-01:
I tried with the 2025.9 dev version but nothing changes, it always crashes if the Bluetooth proxy is active, it doesn't like the C3 process

bdraco, 2025-09-02:
Please follow the steps at https://esphome.io/guides/troubleshooting/

derpeter, 2025-09-02:
Same issue here, sadly i could not find any usefull information in the log via serial cable and the OTA log isn working until i remove bt proxy

speedysk1, 2025-09-05:
Same here.  Crashed 2 esp32s and I had to erase and reflash them via USB.

edwardtfn, 2025-09-15:
@derpeter, this one is crashing due to lack of memory.

derpeter, 2025-09-15:
Oh I'm sorry, it worked in the with older esphome releases and i was not able to find memory issue in the logs. my bad

bkappler69, 2025-09-22:
I guess I have the same issue. I'm running an ESPhome Bluetooth Proxy on an AZ Delivery ESP32 D1 Mini:

```
esphome:
  name: esp-bluetooth-proxy
  friendly_name: esp-bluetooth-proxy

esp32:
  board: esp32dev
  #board: wemos_d1_mini32
  framework:
    type: esp-idf

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxxxxxxxxxxxxxxxxxxxxxx"

ota:
  - platform: esphome
    password: "xxxxxxxxxxxxxxxxxxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: True
  #output_power: 20.5dB

bluetooth_proxy:
  active: True
```

The boot process loops when trying to connect to the WiFi:

```
[14:04:57][C][component:163]: Setup esp32_ble_tracker took 0ms
[14:04:57][C][component:163]: Setup bluetooth_proxy took 0ms
[14:04:57][C][component:163]: Setup bluetooth_proxy took 0ms
[14:04:57][C][component:163]: Setup bluetooth_proxy took 0ms
[14:04:57][C][wifi:060]: Starting
[14:04:57][C][wifi:060]:  Local MAC: 3C:61:05:14:00:68
[14:04:57]ets Jun  8 2016 00:22:57
[14:04:57]
[14:04:57]rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[14:04:57]configsip: 0, SPIWP:0xee
[14:04:57]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[14:04:57]mode:DIO, clock div:2
[14:04:57]load:0x3fff0030,len:6276
[14:04:57]load:0x40078000,len:15736
[14:04:57]load:0x40080400,len:4
[14:04:57]load:0x40080404,len:3860
[14:04:57]entry 0x40080634
[14:04:57]I (56) boot: ESP-IDF 5.4.2 2nd stage bootloader
[14:04:57]I (56) boot: compile time Sep 22 2025 13:56:12
[14:04:57]I (57) boot: Multicore bootloader
[14:04:57]I (59) boot: chip revision: v1.0
[14:04:57]I (61) boot.esp32: SPI Speed      : 40MHz
[14:04:57]I (65) boot.esp32: SPI Mode       : DIO
[14:04:57]I (69) boot.esp32: SPI Flash Size : 4MB
[14:04:57]I (72) boot: Enabling RNG early entropy source...
[14:04:57]I (77) boot: Partition Table:
[14:04:57]I (79) boot: ## Label            Usage          Type ST Offset   Length
[14:04:57]I (86) boot:  0 otadata          OTA data         01 00 00009000 00002000
[14:04:57]I (92) boot:  1 phy_init         RF data          01 01 0000b000 00001000
[14:04:57]I (99) boot:  2 app0             OTA app          00 10 00010000 001c0000
[14:04:57]I (105) boot:  3 app1             OTA app          00 11 001d0000 001c0000
[14:04:57]I (112) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
[14:04:57]I (118) boot: End of partition table
[14:04:57]I (122) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=2f5ach (193964) map
[14:04:57]I (196) esp_image: segment 1: paddr=0003f5d4 vaddr=3ff80000 size=0001ch (    28) load
[14:04:57]I (196) esp_image: segment 2: paddr=0003f5f8 vaddr=3ffbdb60 size=00a20h (  2592) load
[14:04:57]I (201) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=d86c4h (886468) map
[14:04:57]I (510) esp_image: segment 4: paddr=001186ec vaddr=3ffbe580 size=04d50h ( 19792) load
[14:04:57]I (517) esp_image: segment 5: paddr=0011d444 vaddr=40080000 size=1d9f0h (121328) load
[14:04:57]I (579) boot: Loaded app from partition at offset 0x10000
[14:04:57]I (579) boot: Disabling RNG early entropy source...
[14:04:57][I][logger:165]: Log initialized
[14:04:57][C][safe_mode:084]: Unsuccessful boot attempts: 2
[14:04:57][D][esp32.preferences:143]: Writing 1 items: 0 cached, 1 written, 0 failed
[14:04:57][I][app:073]: Running through setup()
[14:04:57][C][component:163]: Setup preferences took 1ms
[14:04:57][C][component:163]: Setup bluetooth_proxy took 0ms
[14:04:57][C][component:163]: Setup esp32_ble took 0ms
[14:04:57][C][component:163]: Setup esp32_ble_tracker took 0ms
[14:04:57][C][component:163]: Setup bluetooth_proxy took 0ms
[14:04:57][C][component:163]: Setup bluetooth_proxy took 0ms
[14:04:57][C][component:163]: Setup bluetooth_proxy took 0ms
[14:04:57][C][wifi:060]: Starting
[14:04:57][C][wifi:060]:  Local MAC: 3C:61:05:14:00:68
[14:04:57]ets Jun  8 2016 00:22:57
[14:04:57]
[14:04:57]rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[14:04:57]configsip: 0, SPIWP:0xee
[14:04:57]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[14:04:57]mode:DIO, clock div:2
[14:04:57]load:0x3fff0030,len:6276
[14:04:57]load:0x40078000,len:15736
[14:04:57]load:0x40080400,len:4
[14:04:57]load:0x40080404,len:3860
[14:04:57]entry 0x40080634
.....
```
I'm using `ESPHome version 2025.9.1`. The ESP has been running stable with `ESPHome version 2025.7.x` versions.

Strangely, the boot succeeds, as soon as I touch the the onboard WLAN antenna (keep a finger on the Antenna). 
I tried experimenting with different Wifi options, but no impact:

```
fast_connect: True
output_power: 18.0dB
power_save_mode: none
```
I also tried different power supplies to exclude low voltage issues. 
I disabled / removed the bluetooth proxy, same issue....
Tried another ESP32 board - same version - same issue.... Touching the antenna resolves the issue..

Are there any other WiFi related options I could try? Any additional debugs I should provide?

bdraco, 2025-09-24:
If its running out of RAM, https://github.com/esphome/esphome/pull/10862 will save another 2k

closed 2025-12-31 by github-actions[bot]