esphome/esphome #10451: ESP32-S3-N16R8 sometimes crashes when playing through media_player

popy2k14, 2025-08-26:
### The problem

I am using the newest main yaml from here combined with the device config in the readme.txt.
Link: https://github.com/Skons/SOAS/issues/14

The playback starts (mostly you hear a crack in speaker), sometimes (50/50) it plays, sometimes it crashes the esp32/esphome.
An older version of the yaml (which was based on arduino and NOT espidf) does not crash!

What i have tried:

- different medias incl mp3's (mono, stereo, diff. bitrates), internet streams like this (http://stream1.liferadio.at/), flac files. Stored also locally in HA www/media folder.
  It doesnt matter, it crashes sometimes.
- lowered psram speed to 40Mhz -> Same issue
- reflashed over usb to get clean flash-> same issue
- tried on two another devices with same esp32 -> same issue
- tried this esp-idf configuration (like it is on voice pe) -> same issue: https://github.com/Skons/SOAS/issues/14#issuecomment-3221522860
- tried different supply's  -> same issue
- power the amp from ESP 3V instead -> same issue
- add a capacitor (10µF, 100µF) to GND & +5V -> same issue
- bridge OTG pads to get +5V instead of ~4,6V -> Same issue

I have found something interesting here: https://esp32.com/viewtopic.php?t=38977
Maybe it's relevant?

### Which version of ESPHome has the issue?

2025.8.0, 2025.8.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

esp_wifi, lwip, tcp_transport, esp_http_client, audio & speaker/media_player are involved, see stacktrace

### YAML Config

```yaml
substitutions:
  display_model: "SH1107 128x128" #use "SH1106 128x64" for SH1106
  display_rotation: "270" #use 0 for SH1106
  sun_latitude: 48.33°
  sun_longitude: 14.31°
  i2c_sda: GPIO48
  i2c_scl: GPIO47
  i2s_lrclk_pin: GPIO3
  i2s_bclk_pin: GPIO1
  i2s_dout_pin: GPIO2
  pin_a: GPIO9
  pin_b: GPIO10
  alarm_off_button_pin: GPIO4
  rotary_button_pin: GPIO8
  alarm_file: /config/esphome/sounds/alarm.flac
  language: "DE" #NL and DE are also supported
  alarm_off_button_single_click_time: 3s

#include device_base yaml
packages:
  #remote_package_shorthand: github://skons/soas/alarm-clock-soas.yaml@main

# WiFi connection
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
[... 282 lines cut ...]

ELF file SHA256: 99fe95cef

Decoded:

0x420e757f: wifi_free at /data/cache/platformio/packages/framework-espidf/components/esp_wifi/src/wifi_netif.c:59
0x420e023d: esp_netif_free_rx_buffer at /data/cache/platformio/packages/framework-espidf/components/esp_netif/lwip/esp_netif_lwip.c:1252
0x420c44ed: esp_pbuf_free at /data/cache/platformio/packages/framework-espidf/components/esp_netif/lwip/netif/esp_pbuf_ref.c:36
0x42060e1a: pbuf_free at /data/cache/platformio/packages/framework-espidf/components/lwip/lwip/src/core/pbuf.c:770
0x42060e1a: pbuf_free at /data/cache/platformio/packages/framework-espidf/components/lwip/lwip/src/core/pbuf.c:727
0x4205d4da: lwip_recv_tcp at /data/cache/platformio/packages/framework-espidf/components/lwip/lwip/src/api/sockets.c:1049
0x4205dfa6: lwip_recvfrom at /data/cache/platformio/packages/framework-espidf/components/lwip/lwip/src/api/sockets.c:1257
0x4205e09a: lwip_recv at /data/cache/platformio/packages/framework-espidf/components/lwip/lwip/src/api/sockets.c:1322
0x420700e9: recv at /data/cache/platformio/packages/framework-espidf/components/lwip/include/lwip/sockets.h:38
0x420700e9: tcp_read at /data/cache/platformio/packages/framework-espidf/components/tcp_transport/transport_ssl.c:309
0x420e0e9f: esp_transport_read at /data/cache/platformio/packages/framework-espidf/components/tcp_transport/transport.c:140
0x420502e2: esp_http_client_read at /data/cache/platformio/packages/framework-espidf/components/esp_http_client/esp_http_client.c:1275
0x420090e5: esphome::audio::AudioReader::http_read_() at /data/build/schlafzimmer-wecker-tobi/src/esphome/components/audio/audio_reader.cpp:288
0x4200914d: esphome::audio::AudioReader::read() at /data/build/schlafzimmer-wecker-tobi/src/esphome/components/audio/audio_reader.cpp:224
0x4201432b: esphome::speaker::AudioPipeline::read_task(void*) at /data/build/schlafzimmer-wecker-tobi/src/esphome/components/speaker/media_player/audio_pipeline.cpp:418
```

### Additional information

Download of the local alarm files: https://drive.google.com/file/d/1wX5OCg1_Z96KZPoBAjR8XxkuDsuijpk0/view?usp=drive_link

ZDBioHazard, 2025-10-19:
I'm having the same problem on 2025.10.1 with just an I2S board connected to an ESP32-S3 board.

  * Sometimes it will stream from HA/MA just fine for quite a while as long as the stream doesn't get interrupted.
  * Sometimes it crashes as soon as I try to play something.
  * Sometimes it plays for a second or two before crashing.
  * Sometimes it will play tiny bits of audio for several seconds before eventually crashing.
  * Adding a `voice_assistant:` component (and mic) and triggering an announcement from HA will crash sometimes.
  * Adding a local WAV audio file and playing it will sometimes crash.
  * Switching to the `i2s_audio` `media_player` platform and `framework: arduino` seems to work well for music playback in my limited testing, but then I obviously lose all the assistant functions.

After a few evenings trying to figure this out including trying all the same tests and remedies as OP, I'm stuck...

I was initially having this problem with my ESP32-S3FH4R2 (S3 supermini) players, but I also did all this testng with a minimal ESP32-S3-devkitc-1 N16R8 setup too since it seemed to be a memory issue.
Despite their limited PSRAM, the ESP32-S3FH4R2 players were working great back on the ~2025.07-ish firmware. I guess I should have been keeping firmware backups. :<

I'm using the virtualenv/pip build environment on a fresh build VM to rule out any sort of cached build files.

Here's the whole config:
```yaml
esphome:
  name: media-2f-bedroom
  friendly_name: "Bedroom Sound Bar"
  platformio_options:
    board_build.flash_mode: dio
esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  flash_size: 16MB
  framework:
    type: esp-idf
    version: dev  # also tried with 'recommended' and 'latest'. The stack traces below are with 'dev'
psram:
  mode: octal
  speed: 80MHz
logger:
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
api:
  encryption:
    key: !secret api_key
ota:
  - platform: esphome
    password: !secret ota_key
i2s_audio:
  - id: i2s_out
    i2s_lrclk_pin: GPIO1
    i2s_bclk_pin: GPIO3
speaker:
  - platform: i2s_audio
    id: speaker_out
    dac_type: external
    i2s_audio_id: i2s_out
    i2s_dout_pin: GPIO2
    sample_rate: 44100
    channel: stereo
  - platform: mixer
    id: speaker_mixer
    output_speaker: speaker_out
    source_speakers:
      - id: speaker_announce
      - id: speaker_media
  - platform: resampler
    id: speaker_announce_resampler
    output_speaker: speaker_announce
  - platform: resampler
    id: speaker_media_resampler
    output_speaker: speaker_media
media_player:
  - platform: speaker
[... 119 lines cut ...]
    0x4038320e: multi_heap_free_impl at ~/.platformio/packages/framework-espidf/components/heap/multi_heap.c:229
    0x40376d1f: heap_caps_free at ~/.platformio/packages/framework-espidf/components/heap/heap_caps_base.c:86
    0x403854e9: free at ~/.platformio/packages/framework-espidf/components/newlib/src/heap.c:34
    0x42013ad7: esphome::RAMAllocator::deallocate(unsigned char*, unsigned int) at ~/esphome/.esphome/build/media-2f-bedroom/src/esphome/core/helpers.h:933
    0x42013ad7: esphome::RingBuffer::~RingBuffer() at ~/esphome/.esphome/build/media-2f-bedroom/src/esphome/core/ring_buffer.cpp:18
    0x4200a05a: std::default_delete::operator()(esphome::RingBuffer*) const at ~/.platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/unique_ptr.h:93
    0x4200a05a: std::default_delete::operator()(esphome::RingBuffer*) const at ~/.platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/unique_ptr.h:87
    0x4200a05a: std::_Sp_counted_deleter, std::allocator, (__gnu_cxx::_Lock_policy)2>::_M_dispose() at ~/.platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/shared_ptr_base.h:527
    0x420077bd: std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release_last_use() at ~/.platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/shared_ptr_base.h:175
    0x420077bd: std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release() at ~/.platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/shared_ptr_base.h:361
    0x420077bd: std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count() at ~/.platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/shared_ptr_base.h:1069
    0x420077bd: std::__shared_ptr::~__shared_ptr() at ~/.platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/shared_ptr_base.h:1525
    0x420077bd: std::shared_ptr::~shared_ptr() at ~/.platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/shared_ptr.h:175
    0x420077bd: esphome::audio::AudioTransferBuffer::~AudioTransferBuffer() at ~/esphome/.esphome/build/media-2f-bedroom/src/esphome/components/audio/audio_transfer_buffer.cpp:10
    0x4200a5fd: esphome::audio::AudioSourceTransferBuffer::~AudioSourceTransferBuffer() at ~/esphome/.esphome/build/media-2f-bedroom/src/esphome/components/audio/audio_transfer_buffer.h:118
    0x4200a5fd: std::default_delete::operator()(esphome::audio::AudioSourceTransferBuffer*) const at ~/.platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/unique_ptr.h:93
    0x4200a5fd: std::default_delete::operator()(esphome::audio::AudioSourceTransferBuffer*) const at ~/.platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/unique_ptr.h:87
[... cut ...]
    0x4200a5fd: esphome::i2s_audio::I2SAudioSpeaker::speaker_task(void*) at ~/esphome/.esphome/build/media-2f-bedroom/src/esphome/components/i2s_audio/speaker/i2s_audio_speaker.cpp:459
```

There we a couple other types of crashes I saw during my testing, but I didn't know about the backtrace decoder at the time, so I didn't save the logs.

An amusing thing that happens sometimes with the voice assistant is the system crashing when it plays the STT-ended sound effect, and coming back online in time to catch the response and crashing again on that, haha.

popy2k14, 2025-10-19:
@ZDBioHazard as a workaround on the SOAS project we are using arduino framework.

You can give it a try until players on esp-idf are stable.

popy2k14, 2026-01-18:
Issue not solved!