esphome/esphome #10453: Since 2025.8 and 2025.8.1 received compilation errors with ESP32 S3 and ESP32 CAM.: sh: 1: xtensa-esp32-elf-g++: not found. Boards  ESP32 C3 and ESP8266 are OK.

fcortijos, 2025-08-26:
### The problem

I have 5 x ESP32 S3 boards with distinct ESPHOME projects for more than two years with no problem, until came the new release. 
And the ESP32 CAM also.

Other projects with ESP32-C3 and ESP8266 compiled without errors in this new release.

I tried several workarounds without success:
- Delete the compilation folder on config/esphome
- Uninstall ESPHOME e installed again.

### Which version of ESPHome has the issue?

2025.8.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

Compilation 

### YAML Config

```yaml
esphome:
  name: corrente-s3-2
  friendly_name: Corrente-S3-2
  comment: Medidor de corrente da casa
  build_path: "../config/esphome/corrente-s3_2_build"

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "5Tsln4+UmxG05g/nAHwtu8abc6sDsuDw/hyWnZgNR4A="

ota:
  - platform: esphome
    password: "96a71f45883c3d8456693692cc212448"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2025.8.1
INFO Reading configuration /config/esphome/Corrente S3.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing corrente-s3-2 (board: esp32-s3-devkitc-1; framework: arduino; platform: https://github.com/pioarduino/platform-espressif32/releases/download/54.03.21-2/platform-espressif32.zip)
--------------------------------------------------------------------------------
INFO Package configuration completed successfully
INFO Package configuration completed successfully
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
 - framework-arduinoespressif32 @ 3.2.1 
 - framework-arduinoespressif32-libs @ 5.4.0+sha.858a988d6e 
 - tool-esptoolpy @ 5.0.2 
 - tool-mklittlefs @ 3.2.0 
 - toolchain-xtensa-esp-elf @ 14.2.0+20241119
Dependency Graph
|-- Networking @ 3.2.1
|-- WiFi @ 3.2.1
|-- ESPmDNS @ 3.2.1
|-- Update @ 3.2.1
|-- noise-c @ 0.1.10
Compiling .pioenvs/corrente-s3-2/lib4c5/Network/NetworkClient.cpp.o
sh: 1: xtensa-esp32s3-elf-g++: not found
Compiling .pioenvs/corrente-s3-2/lib4c5/Network/NetworkEvents.cpp.o
*** [.pioenvs/corrente-s3-2/lib4c5/Network/NetworkClient.cpp.o] Error 127
sh: 1: xtensa-esp32s3-elf-g++: not found
*** [.pioenvs/corrente-s3-2/lib4c5/Network/NetworkEvents.cpp.o] Error 127
========================== [FAILED] Took 1.23 seconds ==========================
```

### Additional information

I reduced the YAML file to the bare minimal and the error still occur.
With this same installation (2025.8.0 or 2025.8.1) I can compile and install in ESP32-C3 and ESP8266 boards with no problem.
I think that the problem is on 'board definition/configuration", what had changed in this release regarding board definitions?

Tks 

P.S. I have HAOS 2025.8.3.

ssieb, 2025-08-26:
Try reinstalling the addon.  You probably missed the original error in a previous compile.

swoboda1337, 2025-08-26:
What is HA running on pi or pc? Is everything else up to date?

[image]

fcortijos, 2025-08-27:
I made another ADDON uninstall, this time with delete data option. Restarted HA.
Then added the ADDON again, and the compilation error vanished.
Please add to the release notes that these steps must be taken.

Tks