esphome/esphome #10455: AS3935 does not write registers correctly (both SPI and i2c)

illigtr, 2025-08-26:
### The problem

I have been trying to get these lightning sensors to behave for a couple of years, with no luck. I have noticed in the HA community  that I am not alone. Although the sensors work FINE under Arduino IDE using either DFROBOT or SPARKFUN libraries, they do not behave well under ESPHOME. Typical symptoms are continuous disturber interrupts regardless of settings, incorrect values written for capacitance, noise level, etc., failure to fire HA automation on storm alert binary sensor, amongst many others.

I finally decided to review the entire code, as3935, as3935_spi and as3935_i2c. I have compared this code with code available for Arduino platform from SPARKFUN and DFROBOT. Although most of the code is similiar, there are significant modifications in some areas.

Here is what I have found so far:

1. Binary sensor for storm alert is set to only 10ms ON before it turns OFF. In my environment, this will simply never trigger an automation. If I change the code to 100ms, HA automation triggers as expected.

Line 60 in as3935.cpp has this:
```C
      this->set_timeout(10, [this]() { this->thunder_alert_binary_sensor_->publish_state(false); });
```

Increasing the timeout resolves the issue.

2. Writing registers with SPI or I2C under ESPHOME uses reverse logic than other Arduino libraries.

ESPHome library applies a reverse mask to the register... 

```C
  uint8_t write_reg = this->read_register(reg);
  write_reg &= (~mask);
  write_reg |= (bits << start_pos);
```

while SPARKFUN and DFROBOT do not.

```C
        _spiWrite = _readRegister(_wReg);       // Get the current value of the register
        _spiWrite &= _mask;                     // Mask the position we want to write to
        _spiWrite |= (_bits << _startPosition); // Write the given bits to the variable
```

3. ESPHome library mixes two different methods to write to the registers, I am not sure why, but clearly some are improper writing values.

Here are two examples: The indoor boolean is using the classic write method found in SPARKFUN and DFROBOT libraries  This is probably incompatible with the reverse mask method specified in the as3935_i2c and as_3935_spi libraries. See item 2.

```C

void AS3935Component::write_indoor(bool indoor) {
  ESP_LOGV(TAG, "Setting indoor to %d", indoor);
  if (indoor) {
    this->write_register(AFE_GAIN, GAIN_MASK, INDOOR, 1);
  } else {
    this->write_register(AFE_GAIN, GAIN_MASK, OUTDOOR, 1);
  }
}

void AS3935Component::write_mask_disturber(bool enabled) {
  ESP_LOGV(TAG, "Setting mask disturber to %d", enabled);
  if (enabled) {
    this->write_register(INT_MASK_ANT, (1 << 5), 1, 5);
  } else {
    this->write_register(INT_MASK_ANT, (1 << 5), 0, 5);
  }
}
```
The second example for the mask distuber is using a different method to prepare the MASK that will be sent to the write_register function. I presume this one will work with the reverse mask.

4. Finally, here is an example of where setting the capacitance to any ODD number will result in the value NOT being applied, while even numbers appear to work. It is as if the LSB is always set to zero.

A good YAML example is here:

```YAML

as3935_i2c:
  address: 0x03
  irq_pin: GPIO04
  indoor: true
  noise_level: 3
  watchdog_threshold: 2
  spike_rejection: 1
  lightning_threshold: 1
  mask_disturber: true
  tune_antenna: true
  #calibration: true
  div_ratio: 16
  capacitance: 5
```

Purposely setting tune_antenna: true will allow us to see the result of the capacitance setting (which should be 40pF), but in fact, we can see it is only 32pF. Here is part of the log file:

```
[I][as3935:308]: Starting antenna tuning
[I][as3935:311]: Division Ratio is set to: 16
[I][as3935:312]: Internal Capacitor is set to: 32

```

### Which version of ESPHome has the issue?

2025.8.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

as3935, as3935_spi, as3935_i2c

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt
Testing environment:

SBC and libraries:

Arduino IDE v2.3.6
Adruino UNO for Arduino IDE testing (both sensors)
SparkFun AS3935 Library v1.4.9
DFROBOT AS3935 Library v1.0.2
ESP-WROOM-32 for ESPHome and Arduino IDE (both sensors)

Sensors:
DF Robot Gravity AS3935 I2C
CJMCU-3935 Sensor (I2C and SPI)

Tests: Many TL;DR

Under Arduino I can write and read back all registers correctly, both SPI and I2C. Sensors behave as described in their technical documentation.

Under ESPHome, sensors are erratic. Values to registers either do not get written correctly or reading them gives incorrect results, regarless of SPI or I2C.
```

### Additional information

Willing to assist in any testing to finally get these sensors to work as well as they do under Arduino IDE.

artarus, 2025-08-31:
Hi,
I had a similar issue with the SPI configuration and, after checking with a logic analyser, found that the SPI mode was incorrect.
Just add the following to your config:
as3935_spi:
  spi_mode: MODE1
This will fix the communication, and the sensor will be read and written correctly.

illigtr, 2025-08-31:
@artarus Thanks for that suggestion. I will retest SPI mode, although I seem to see the issue with i2c as well, but will recheck that.

illigtr, 2025-08-31:
@artarus 

Ok, I added the MODE1 for spi as suggested. This still does not resolve writing the div ratio and capacitance correctly to the sensor. Reading both back shows these were not changed. In addition, the calibration now fails.

```YAML

as3935_spi:
  cs_pin: GPIO33
  spi_mode: MODE1
  irq_pin: GPIO04
  indoor: true
  noise_level: 2
  watchdog_threshold: 2
  spike_rejection: 1
  lightning_threshold: 1
  mask_disturber: true
  tune_antenna: false
  calibration: true
  div_ratio: 32
  capacitance: 5
```

Logs show this:

```

[I][as3935:024]: Division Ratio is set to: 16
[I][as3935:025]: Internal Capacitor is set to: 56
[I][as3935:287]: Starting oscillators calibration
[W][as3935:306]: Calibration was NOT succesful

```

Removing the SPI mode results in this:

```
[I][as3935:024]: Division Ratio is set to: 16
[I][as3935:025]: Internal Capacitor is set to: 0
[I][as3935:287]: Starting oscillators calibration
[W][as3935:306]: Calibration was NOT succesful
```

So, yes, it does appear to affect the LSB, as odd values for capacitance appear to get written with MODE1, but they are STILL wrong values. Capcitance = 5 should equal 40pF, but what is read back is 56pF with MODE1 and 0pF in default (MODE3?). So the problem is  not resolved with the change in SPI mode. Also, registers written with i2c are still problematic.

artarus, 2025-08-31:
Thanks for the logs. I’ll test it today with a logic analyzer.

illigtr, 2025-08-31:
@artarus I decided to test all 4 SPI modes. Interestingly MODE2 wrote the registers CORRECTLY! This is really odd.

According to the as3935_spi.h the default for the sensor is MODE2

```C
class SPIAS3935Component : public as3935::AS3935Component,
                           public spi::SPIDevice<spi::BIT_ORDER_MSB_FIRST, spi::CLOCK_POLARITY_LOW,
                                                 spi::CLOCK_PHASE_LEADING, spi::DATA_RATE_2MHZ> {
```

However, if one does not explicitly specifiy the SPI MODE, it defaults to MODE0!!!! This is the logs:

```
[D][spi_device:393]: mode 0, data_rate 2000kHz
```

This of course explains why data writes to registers are hit and miss. Explicitly setting SPI MODE2 results in the registers finally being written correctly. I will test all the registers now with SPI MODE2.

Does ANYONE know what should be the default SPI MODE if it is explicitly stated in the sensor_spi.h file?

artarus, 2025-08-31:
> [@artarus](https://github.com/artarus) I decided to test all 4 SPI modes. Interestingly MODE2 wrote the registers CORRECTLY! This is really odd.
> 
> According to the as3935_spi.h the default for the sensor is MODE2
> 
> class SPIAS3935Component : public as3935::AS3935Component,
>                            public spi::SPIDevice<spi::BIT_ORDER_MSB_FIRST, spi::CLOCK_POLARITY_LOW,
>                                                  spi::CLOCK_PHASE_LEADING, spi::DATA_RATE_2MHZ> {
> 
> However, if one does not explicitly specifiy the SPI MODE, it defaults to MODE0!!!! This is the logs:
> 
> ```
> [D][spi_device:393]: mode 0, data_rate 2000kHz
> ```
> 
> This of course explains why data writes to registers are hit and miss. Explicitly setting SPI MODE2 results in the registers finally being written correctly. I will test all the registers now with SPI MODE2.
> 
> Does ANYONE know what should be the default SPI MODE if it is explicitly stated in the sensor_spi.h file?

CLOCK_POLARITY_LOW CLOCK_PHASE_LEADING = thats MODE0. Chip is using correct MODE1

artarus, 2025-08-31:
Also tested with capacitance set to 5. The logic analyzer shows that the value is correctly written to the correct register. It's possible that the LOG is misinterpreting it differently.

illigtr, 2025-09-01:
@artarus div_ratio code error: ok! 

spi mode: Like I said, I tested all 4 SPI modes and only MODE2 worked in getting the capacitance value to "stick", especially odd values with an  lsb=1. However, I can test again. I copied the component locally and added routines to read back ALL settings for testing purposes. I know these work, because they work under Arduino IDE using the SparkFun library. I will retest again.

illigtr, 2025-09-01:
@artarus Here is some more initial results from detailed testing. 

First, I should add that I was using the **dev** branch of ESPHome and copied the components locally in order to add additional logging. (The div_ratio typo was already fixed).

I also created specific read functions for all settings, using the known, good SparkFun library as a guide.

Here is the rewritten setup() function

```C++

void AS3935Component::setup() {
  this->irq_pin_->setup();
  LOG_PIN("  IRQ Pin: ", this->irq_pin_);

  // reset sensor
  reset_allsettings_();
  
  // Write properties to sensor
  this->write_indoor(this->indoor_);
  read_indoor_();
  
  this->write_noise_level(this->noise_level_);
  read_noise_level_();

  this->write_watchdog_threshold(this->watchdog_threshold_);
  read_watchdog_threshold_();
  
  this->write_spike_rejection(this->spike_rejection_);
  read_spike_rejection_();
  
  this->write_lightning_threshold(this->lightning_threshold_);
  read_lightning_threshold_();
  
  this->write_mask_disturber(this->mask_disturber_);
  read_mask_disturber_();
  
  this->write_div_ratio(this->div_ratio_);
  read_div_ratio_();

  this->write_capacitance(this->capacitance_);
  read_capacitance_();
```

And here is the code for the new functions:

```C++
void AS3935Component::reset_allsettings_() {
	//    RESET_LIGHT = 0x3C,
	// this constant was not in the as3935.h file
    this->write_register(0x3C, WIPE_ALL, DIRECT_COMMAND, 0);
    ESP_LOGI(TAG, "Reset all as3935 settings");
	delay(20);
}
void AS3935Component::read_noise_level_() {
  
  uint8_t reg_val = this->read_register_(THRESHOLD, NOISE_FLOOR_MASK);
  ESP_LOGI(TAG, "READING noise level reg_val:  %d", reg_val);
   uint8_t reg_val2 = reg_val >> 4;
   ESP_LOGI(TAG, "CONVERTING noise level value to :  %d", reg_val2);
  
}

void AS3935Component::read_indoor_() {
  
  uint8_t reg_val = this->read_register_(AFE_GAIN, GAIN_MASK);
  ESP_LOGI(TAG, "READING indoor:  %d", reg_val);
  
}
void AS3935Component::read_watchdog_threshold_() {
  
[... 169 lines cut ...]
[I][as3935:205]: Setting noise level to 5
[I][as3935:065]: READING noise level reg_val:  2
[I][as3935:067]: CONVERTING noise level value to :  0
[I][as3935:193]: Setting watchdog sensitivity to 3
[I][as3935:080]: READING watchdog threshold:  80
[I][as3935:217]: Setting spike rejection to 3
[I][as3935:086]: READING spike rejection:  192
[I][as3935:228]: Setting lightning threshold to 5
[I][as3935:092]: READING lightning threshold:  16
[I][as3935:098]: CONVERTING lightning threshold :  5
[I][as3935:249]: Setting mask disturber to 0
[I][as3935:109]: READING mask distruber:  0
[I][as3935:261]: Setting div ratio to 0
[I][as3935:118]: Division Ratio is set to: 16
[I][as3935:285]: Setting tune cap to 40 pF
[I][as3935:130]: Internal Capacitor is set to: 40 pF
[I][as3935:389]: Starting oscillators calibration
[W][as3935:408]: Calibration was NOT succesful
```

EARLY CONCLUSIONS
1. Per your testing, MODE1 seems to be the right one.
2. Using MODE1 alone does not guarantee that values are read are those that were written... It appears that resetting the as3935 makes a big difference. I did not find an initial reset/initialize  sequence in the ESPHome code.

More testing underway.

billmaterial, 2025-09-01:
I have been using the as3935 for around a year using the arduino framework, recently my bme688 sensor has stopped working with the arduino framework but works with the esp-idf framework. I switched to esp-idf and now the as3935 refuses to work. I can see in this thread that some people are working on this. Is there anyway I can get this working?
PS. on some boards the capacitors on the PCB are wrong, mine was so I changed them and it started working perfectly under the arduino framework.

illigtr, 2025-09-01:
@billmaterial Hi. not sure why the as3935 is not working with esp-idf, all I can say is that I believe the as3935 library needs some updates based on what I have found. If you are using the sensor under SPI, you might want to try to force SPI MODE1, per @artarus.

I still believe there are lots of issues where register values are simply not written correctly. That said, sometimes the "default" configuration works fine. I've had these sensors under ESPHome for 2 years and never got them working well enough, my main issue was they would not trigger HA automation, but I believe I have found the software code issue for that, at least.

artarus, 2025-09-01:
@illigtr 
Thanks for the information.
The thing is, I'm testing all communication not through software, but directly in hardware using a logic analyzer, and I haven’t found any other issues. All register reads and writes are correct — the values written match the values read back.
My hardware is an ESP32 ETH01 module, using the ESP-IDF framework, compiled via ESPHome in Home Assistant.
I haven’t investigated the I²C interface yet, as it doesn’t work at all — there seems to be something fundamentally wrong with it.
esp32:
  board: esp32dev
  framework:
    type: esp-idf

illigtr, 2025-09-02:
@artarus Thanks for your reply. I am curious how you can validate the value of the registers using a logic analyzer. I understand how one could verify if the bit stream is formatted correctly and watch for an ACK, but that doesn't necessarily mean the value in the register is the value expected. Just curious.

All I can say is that under Arduino IDE with 2 different libraries I can write and read all the values correctly, every time, under SPI and I2C. I cannot get to that point under ESPHome with the current component library. I will continue some tests though.

artarus, 2025-09-02:
@illigtr Most of the written registers are successfully read back, so I simply compare what was written with what was read.
Anyway, my board is now working after setting SPI MODE1.
If you'd like, I can test your binary on my hardware to see what actually happens.
What do you mean by "RESET"?
As far as I can tell, there's no requirement to perform any reset on the chip — all register values should be appear to be at their defaults after power-up. 
As for I²C, there seems to be an issue where the ESP32 writes to the 0x03 register instead read it when an interrupt arrives from the AS3935. Register 0x03 last 4 bits read only and as a result, the IRQ pin stays high all the time. I plan to investigate this further later this week.
@billmaterial Please refer to the official datasheet for the chip:
SPI mode pins (ADD1, ADD0, SI, EN_VREG) must be connected to GND.
Power supply 2.4 - 3.6 V, as the internal regulator is disabled by default on these boards. The capacitor value can be tested for correctness via the output signal on the IRQ pin, but you’ll need to determine the proper method for measuring that signal. If you have oscilloscope there no problem at all.

illigtr, 2025-09-02:
@artarus Ok. I *finally* have something concrete and I am quite sure this is a *real* issue.

As previously mentioned, comparing the ESPHome library with SparkFun shows that the actual register writes in ESPHome use this logic (as3935_spi.cpp):

```C+

void SPIAS3935Component::write_register(uint8_t reg, uint8_t mask, uint8_t bits, uint8_t start_pos) {
  uint8_t write_reg = this->read_register(reg);

  write_reg &= (~mask);
  write_reg |= (bits << start_pos);

  this->enable();
  this->write_byte(reg);
  this->write_byte(write_reg);
  this->disable();
}
```

You will note that the register value is ANDed with the reverse MASK. SparkFun libary does not reverse the MASK. Ok. I then checked a number of the constants for masks in the ESPHome library and they have indeed been reversed as well... BUT NOT ALL OF THEM.

So, reversing a reversed mask will lead to the exact same actual SPI write. So far, so good.

In the ESPHome library, not ALL defined masks were reversed. A good example is CAP_MASK. It is set to 0xF0, just like SparkFun.

If you use this mask for writing a value, you are just not going to the get value that you want, and I can confirm this by reading the value back! I have tested this multiple times.

As a quick "correction" test I created this:

```C+
  CAP_MASK_READ = 0xF0,
  CAP_MASK =  0xF,
```

CAP_MASK is now an inverse mask of the original value (per SparkFun). I renamed the original to CAP_MARK_READ, because I really need that for the routine that reads that value.

Aha. It works, ALL values are now written correctly and read correctly. 

My guess is there are more... I will check them all.

illigtr, 2025-09-02:
Here are the results of checking each YAML setting and what the code actually does. I'd be curious to know from anyone why the masks and the write logic using double mask inversion in in the first place. There must be some reason for this and it appears the code simply wasn't completely converted.

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:\Users\tony_\AppData\Local\Temp\msohtmlclip1\01\clip.htm">
<link rel=File-List
href="file:///C:\Users\tony_\AppData\Local\Temp\msohtmlclip1\01\clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

Function | mask | Comments | Test Results
-- | -- | -- | --
write_indoor() | inversed |   | Ok
read_indoor() |   | does   not exist in ESPHome library | would   fail
write_watchdog_threshold() | inversed |   | Ok.
read_watchdog_threshold() |   | does   not exist in ESPHome library | would   fail
write_noise_level() | inversed |   | Ok.
read_noise_level() |   | does   not exist in ESPHome library | would   fail
write_spike_rejection() | inversed |   | Ok.
read_spike_rejection() |   | does   not exist in ESPHome library, BUT there is an unsed R_SPIKE_MASK constant!!! | Ok,   if using R_SPIKE_MASK
write_lightning_threshold() | original/but   uses custom bitwise mask |   | Ok
read_lightning_threshold() |   | does   not exist in ESPHome library | It   would pass
write_mask_disturber() | original/but   uses custom bitwise mask |   | Ok
read_mask_disturber() |   | does   not exist in ESPHome library | It   would pass
write_div_ratio() | INVERSED/but   uses custom bitwise mask | although   an inversed mask exists, it is not used in the write command | Ok
read_div_ratio() | inversed | inversed   mask causes a incorrect read values | FAIL!
write_capacitance() | original |   | FAIL!
read_capacitance() | original |   | OK

</body>

</html>

illigtr, 2025-09-03:
Ok. I finally tracked down what is happening with the CJHMC-3935 in I2C mode. It is, in fact, writing registers correctly. The problem when I try to read them back is that I am relying on this code which causes all the problems and confusion.

For some reason, whomever decided to convert the original SparkFun library to ESPHome, decided to inverse nearly all register masks for writes. The SPI and I2C write routines would the inverse the mask AGAIN before performing an OR operation on the left shifted write values.

```C++
write_reg &= (~mask);
write_reg |= (bits << start_pos);
```

The problem is when one wants to read the register value back, one cannot use the inversed mask. Ok, so for testing, I added the original masks for read register operations. This works perfectly under SPI, but returns the same value, regardless in I2C.

What gives? The problem is the read_register() in the as3935_i2c.cpp

\\\C++
uint8_t I2CAS3935Component::read_register(uint8_t reg) {
  uint8_t value;
  if (write(&reg, 1) != i2c::ERROR_OK) {
    ESP_LOGW(TAG, "Writing register failed!");
    return 0;
  }
  if (read(&value, 1) != i2c::ERROR_OK) {
    ESP_LOGW(TAG, "Reading register failed!");
    return 0;
  }
  return value;
}

```

Note that it attempts a write, then a read. Well, that's no good. I replaced that with this temporarily:

```C++
uint8_t I2CAS3935Component::read_register(uint8_t reg) {
  uint8_t value;
  if (!this->read_byte(reg, &value)) {
    this->mark_failed();
    ESP_LOGW(TAG, "read_byte failed - increase log level for more details!");
    return 0;
  }
  return value;
}
```

Aha! All the read register operations work as expected and I can read all the custom YAML settings back.

Note that read_register() is actually never used in the write_register() function. It is actually only used in this function in as_3935.cpp

```C+
uint8_t AS3935Component::read_register_(uint8_t reg, uint8_t mask) {
  uint8_t value = this->read_register(reg);
  value &= (~mask);
  return value;
}
```

The read_register_() function is only used in the as3935.cpp in these locations:
```
Line 197:   return this->read_register_(INT_MASK_ANT, INT_MASK);
Line 216:   return this->read_register_(DISTANCE, DISTANCE_MASK);
Line 224:   temp = this->read_register_(ENERGY_LIGHT_MMSB, ENERGY_MASK);
Line 244:   uint8_t reg_val = this->read_register_(INT_MASK_ANT, DIV_MASK);
Line 262:   uint8_t reg_val = this->read_register_(FREQ_DISP_IRQ, CAP_MASK) * 8;
Line 291:   uint8_t reg_val_srco = this->read_register_(CALIB_SRCO, CALIB_MASK_NOK);
Line 292:   uint8_t reg_val_trco = this->read_register_(CALIB_TRCO, CALIB_MASK_NOK);
```
Of course if any of these are using a modified (inversed) mask, they will fail to return the correct value. Some are using a custom mask. And of course if used under I2C all reads will FAIL due to what I discovered above.

Are you one of the authors or maintainers of this library? How can we get all this cleaned up? I also noticed, in reading the i2c lib in ESPHome that the ic2 calls are legacy format and should be avoided altogether.

Willing to help test any PR#.

artarus, 2025-09-03:
@illigtr SPI mode uses a different protocol — that behaviour is specific to I²C.
Almost all I²C memory devices use the read method as described above.
I haven’t tested this with the Arduino platform. From datasheet: "Whenever events happen, the AS3935 pulls the IRQ high and
displays the interrupt in the REG0x03[3:0]. Figure 44 shows the
interrupt register. After the signal IRQ goes high the external
unit should wait 2ms before reading the interrupt register. The
interrupt bus IRQ is set back to low whenever the interrupt
register is read out."

artarus, 2025-09-04:
@illigtr Just tested under the Arduino framework - everything is working, including storm and disturbance detection.
Conclusion: both ESP-IDF and Arduino I²C implementations are working correctly now.