esphome/esphome #10461: How to log crash when connected to main power?

warheat1990, 2025-08-27:
### The problem

I have bunch of Sonoff M5 1C, 2C, 3C (EU version) flashed with ESPHome, however one 2C seems to have problem which will keep crashing (I suspect it's crashing until it goes into safe mode and repeat again and again), it will get stable sometimes for hours until it happened again.

I pulled it apart and connect it to 5v power, use the TX and RX to check the log and it seems stable (I leave it for days and it never crashes), so it only occurs when connected to main power.

I also tried flashing Tasmota and it is also stable whether on 5v or main power.

So my question is how do I check the log when connected to main power? Because after it crashes I can't view it anymore, is there any way to write the log/crashes into its flash?

Thank you

### Which version of ESPHome has the issue?

2025.8.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

dekiesel, 2025-10-22:
> I got this one a while back and it says it's not available now, but that's what I mean. https://www.aliexpress.com/item/1005006374497996.html

I am also seeing an issue with a device connected to mains, so thanks for your debug input!

Just to clarify, when the device (shelly in my case) is connected to mains I attach RX/TX from the shelly to the isolated uart adapters TX/RX and nothing else, correct?

ssieb, 2025-10-22:
You also need GND connected.  It is isolated as well.  You don't need to connect to the Shelly RX because you're only reading from it, not sending anything.  You would need it if you wanted to flash though.