esphome/esphome #10467: I2C software timeout errors and reconnections

mariusz-schimke, 2025-08-27:
### The problem

I updated one of my ESP32 boards yesterday to the latest ESPHome version 2025.8.1. I don't do it every time a new version is released, so previously I stayed on 2025.7.5 for a couple of weeks. Since yesterday, I'm observing some I2C timeout errors and the device becomes unavailable every minute or so for a second or two.

I checked a several-year log history on my syslog server, and there are no such errors before yesterday. Is it a change in some default params that I need to change manually now or is there a chance it's some kind of a bug in the latest release?

### Which version of ESPHome has the issue?

2025.8.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt
26.08.2025,16:39:22,Notice,entrance-gate-and-gas-meter,local0,esp-idf,[I][esp-idf:000]: E (141585) i2c.master: i2c_master_receive(1261): I2C transaction failed
26.08.2025,16:39:22,Notice,entrance-gate-and-gas-meter,local0,esp-idf,[I][esp-idf:000]: E (141572) i2c.master: s_i2c_synchronous_transaction(945): I2C transaction failed
26.08.2025,16:39:22,Notice,entrance-gate-and-gas-meter,local0,esp-idf,[I][esp-idf:000]: E (141569) i2c.master: I2C software timeout
26.08.2025,16:39:05,Notice,entrance-gate-and-gas-meter,local0,esp-idf,[I][esp-idf:000]: E (124922) i2c.master: i2c_master_receive(1261): I2C transaction failed
26.08.2025,16:39:05,Notice,entrance-gate-and-gas-meter,local0,esp-idf,[I][esp-idf:000]: E (124909) i2c.master: s_i2c_synchronous_transaction(945): I2C transaction failed
26.08.2025,16:39:05,Notice,entrance-gate-and-gas-meter,local0,esp-idf,[I][esp-idf:000]: E (124906) i2c.master: I2C software timeout
```

swoboda1337, 2025-09-20:
Is this still an issue?

mariusz-schimke, 2025-09-21:
I'm a bit hesitant as one of my other devices started behaving strangely (my garage door would randomly open and close at night). But it could have as well been my mistake in connections. Let me try the new release. Will let you know about the result.

mariusz-schimke, 2025-09-21:
@swoboda1337 , I'm sorry, but there still seems to be some issue (see the screenshot). Also, I noticed that one of my relays connected to an I2C GPIO expander just got randomly triggered (I heard the doorbell connected to it). What's strange, the sound was shorter than it should be. There is a switch component that controls the duration of the relay, and it was much shorter than it should last. As though it was triggered for a very short moment and then released.

In my previous comment I mentioned a garage door being randomly opened/closed by my other board, but there are no I2C elements in use. Probably the same reason as, similarly, a relay controls the door.

I have an impression that the 2025.8.x or 2025.9.x release introduces major issues that haven't been noticed or fixed yet.

[image]

swoboda1337, 2025-09-21:
Can you share you config, I can't tell what device you are using. It might be an issue with that component not a general I2C issue.

mariusz-schimke, 2025-09-21:
@swoboda1337 but the thing is that it never happened before. I made the update today morning, and the issue happened around the time of the logs I attached today here.

mariusz-schimke, 2025-09-21:
@swoboda1337, sorry. Here you go (I updated the script).

mariusz-schimke, 2025-11-05:
@swoboda1337 , does anything come to your mind based on this? I can see again disconnections every few minutes as before. I set the i2c timeout to 50ms now, but it has no impact on this behavior. I didn't specify the framework version you suggested. Is that necessary?

[image]

swoboda1337, 2025-11-05:
You can also try to use Arduino instead of IDF. Arduino is still using the legacy I2C drivers I believe. Not that I recommend you use Arduino but  I am curious if it changes anything.

swoboda1337, 2025-11-05:
There is not very much logging, there is 8 minutes between the I2C error and the reboot:
```
05.11.2025,12:12:38,Notice,entrance-gate-controller,local0,app,[I][app:120]: setup() finished successfully!
05.11.2025,12:04:39,Notice,entrance-gate-controller,local0,esp-idf,[I][esp-idf:000]: E (645118) i2c.master: i2c_master_execute_defined_operations(1366): I2C transaction failed
```

akx, 2025-11-21:
I'm seeing the same symptoms with an [M5Stack 8Encoder](https://github.com/m5stack/M5Unit-8Encoder) board with ESPHome version 2025.11.0, M5NanoC6 hardware.

Writing to it works fine and LEDs start to blink (see configuration in details block below), but as soon as I try to read anything (commented-out code in the block), it's

```
[22:15:41.000][D][esp-idf:000]: E (8234) i2c.master: clear bus failed.
[22:15:41.000][D][esp-idf:000]: E (8235) i2c.master: s_i2c_transaction_start(617): reset hardware failed
[22:15:41.000][D][esp-idf:000]: E (8236) i2c.master: s_i2c_synchronous_transaction(945): I2C transaction failed
[22:15:41.000][D][esp-idf:000]: E (8237) i2c.master: i2c_master_execute_defined_operations(1401): I2C transaction failed
[22:15:41.052][D][esp-idf:000]: E (8290) i2c.master: clear bus failed.
[22:15:41.053][D][esp-idf:000]: E (8291) i2c.master: s_i2c_transaction_start(617): reset hardware failed
[22:15:41.055][D][esp-idf:000]: E (8292) i2c.master: s_i2c_synchronous_transaction(945): I2C transaction failed
[22:15:41.056][D][esp-idf:000]: E (8293) i2c.master: i2c_master_execute_defined_operations(1401): I2C transaction failed
```

all the way. I then have to unplug the thing for the I2C device to come back to life. 

<details>

```yaml

esphome:
  name: encoders

esp32:
  board: esp32-c6-devkitc-1
  framework:
    type: esp-idf

logger:
  level: DEBUG

api:
  password: ""

ota:
  - platform: esphome
    password: ""

wifi: ...

i2c:
  sda: GPIO2
  scl: GPIO1
  scan: true
  id: bus_a
  frequency: 50kHz
  timeout: 5ms
    
output:
  - platform: gpio
    pin:
      number: 7
      mode: output
    id: LED

interval:
  - interval: 500ms
    then:
      - output.turn_on: LED
      - delay: 250ms
      - output.turn_off: LED
  - interval: 1000ms
    then:
    - lambda: !lambda |-
        // uint8_t state[4];
        // id(enc8).read_register(0x0, state, 4);
        uint8_t leds[24] = {0x00};
        random_bytes(leds, 24);
        id(enc8).write_register(0x70, leds, 24);

i2c_device:
  id: enc8
  address: 0x41
```

bzumik1, 2026-01-04:
I am on **2025.12.4** and I am seeing this issue as well, not sure if it was there before, I have noticed it after update. Now I have full log of i2c errors.

nepozs, 2026-01-30:
Maybe not related to this issue, but there is known hardware issue with MCP23017 
https://github.com/adafruit/Adafruit_CircuitPython_MCP230xx/issues/57
PA7 and PB7 should not be used as inputs, but component `mcp23017` do not block such use