esphome/esphome #10468: buttons from multiple packages not showing up over mqtt (but work fine in web server)

szupi-ipuzs, 2025-08-27:
### The problem

I have a smoke sensor which I want to control from HA over mqtt, but it also has the "api" part (maybe it's relevant).
The device is supposed to have 2 buttons:
1) a "Restart" button (from the restart platform) - added via my "device_base" package
2) a "Battery replaced" button - a template button added via my "battery_notes" package

The problem is that the "Restart" button does NOT show up in HA and I have checked using MQTT explorer that it is not reported as part of the device.

[image]

What's interesting that both buttons are visible via the web server:

[image]

The device's yaml is complicated as it is multi-layered and uses multiple packages. It looks something like this:

smoke-sensor-yg400a-1.yaml:
```yaml
substitutions:
   # this part defines names and networks

<<: !include smoke-sensor/yg400a-cbu.yaml
```

yg400a-cbu.yaml:
```yaml
packages:
  wifi: !include ../common/wifi_no_ip.yaml
  device_base: !include ../common/device_base_mqtt.yaml
  web_server: !include ../common/web_server.yaml
  battery_notes: !include
    file: ../shared_public/battery_notes.yaml
    vars:
      battery_notes_time_id: homeassistant_time
      battery_notes_type: AA
      battery_notes_count: 2

# the actual sensors follow
```

device_base_mqtt.yaml:
```yaml
# this defines common part like uptime sensors and api connection

mqtt:
  broker: xx.xx.xx.xx
  birth_message:
  will_message:

api:
  reboot_timeout: 120min

logger:

ota:
  - platform: esphome
  - platform: web_server

# and here's the button that does not show up
button:
  - platform: restart
    name: Restart
    id: restart_button

```

battery_notes.yaml:
```yaml

button:
  - platform: template
    id: battery_replaced_button
    name: "Battery replaced"
    entity_category: diagnostic
    on_press: 
# the rest is not relevant
```

### Which version of ESPHome has the issue?

2025.9.0-dev

### What type of installation are you using?

Docker

### What platform are you using?

BK72XX

### Component causing the issue

core

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

ssieb, 2025-08-27:
Buttons shouldn't show up.  I think that other button topic is there because you pressed it some time.  That's the command.  What are you expecting to see?  A button doesn't have any state.  The only place you would see anything would be in the discovery message.

closed 2025-08-27 by ssieb

ssieb, 2025-08-27:
Most likely.  Try searching for the entity.

szupi-ipuzs, 2025-08-27:
Ok, the problem seemed to be the unique_id - it was not really unique.
The problem disappeared after setting:
```yaml
mqtt:
   discovery_unique_id_generator: mac
```

Thanks!