esphome/esphome #10473: Libretiny conpile issues.   Seems like python site-packages missing???

blindguynar, 2025-08-28:
### The problem

unable to compile yaml for libretiny device.

### Which version of ESPHome has the issue?

2025.8.1

### What type of installation are you using?

pip

### What platform are you using?

BK72XX

### YAML Config

```yaml
esphome:
  name: kaden-powerstrip
  friendly_name: Kaden-Powerstrip

bk72xx:
  board: generic-bk7231t-qfn32-tuya

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xxx"

ota:
  - platform: esphome
    password: "xxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Optional manual IP
  manual_ip:
    static_ip: 192.168.40.155
    gateway: 192.168.40.1
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Kaden-Powerstrip"
    password: "xxx"

captive_portal:
    
#web_server:
#  port: 80

# Device Specific Config
text_sensor:
  - platform: version
    name: "ESPHome Version"

button:
  - platform: restart
    name: "Restart" # optional device restart switch

binary_sensor:
  - platform: status
    name: "Status"
  - platform: gpio
    pin:
      number: GPIO08
      mode: INPUT
      inverted: True # Normally off /Press on
    name: "Button"
    on_multi_click: # Button Behavior (Click Outlets On/Off)
    - timing:
        - ON for at least 1s
        - OFF for at least 0.2s
      then:
        # Single Click Toggle Outlet 1 On/Off
        # (One long press)
        - logger.log: "Single Click"
        - switch.toggle: outlet_1
    - timing:
        - ON for at most 1s
        - OFF for at most 1s
        - ON for at least 1s
        - OFF for at least 0.2s
      then:
        # Double Click Toggle Outlet 2 On/Off
        # (One press followed by a long press)
        - logger.log: "Double Click"
        - switch.toggle: outlet_2
    - timing:
        - ON for at most 1s
        - OFF for at most 1s
        - ON for at most 1s
        - OFF for at most 1s
        - ON for at least 1s
        - OFF for at least 0.2s
      then:
        # Triple Click Toggle Outlet 3 On/Off 
        # (Two presses followed by a long press)
        - logger.log: "Triple Click"
        - switch.toggle: outlet_3
    - timing:
        - ON for at most 1s
        - OFF for at most 1s
        - ON for at most 1s
        - OFF for at most 1s
        - ON for at most 1s
        - OFF for at most 1s
        - ON for at least 1s
        - OFF for at least 0.2s
      then:
        # Quad Click Toggle Outlet 4 On/Off 
        # (Three presses followed by a long press)
        - logger.log: "Quad Click"
        - switch.toggle: outlet_4
    - timing:
        - ON for at most 1s
        - OFF for at most 1s
        - ON for at most 1s
        - OFF for at most 1s
        - ON for at most 1s
        - OFF for at most 1s
        - ON for at most 1s
        - OFF for at most 1s
        - ON for at least 1s
        - OFF for at least 0.2s
      then:
        # Penta Click Toggle Outlet USB On/Off 
        # (Four presses followed by a long press)
        - logger.log: "Penta Click"
        - switch.toggle: usb

sensor:
  - platform: wifi_signal
    name: "WiFi signal"
    update_interval: 60s
  - platform: uptime
    name: "Uptime"
    
switch:
  - platform: gpio
    name: "Outlet 1"
    pin: GPIO06
    id: outlet_1
    restore_mode: RESTORE_DEFAULT_OFF
  - platform: gpio
    name: "Outlet 2"
    pin: GPIO24
    id: outlet_2
    restore_mode: RESTORE_DEFAULT_OFF
  - platform: gpio
    name: "Outlet 3"
    pin: GPIO26
    id: outlet_3
    restore_mode: ALWAYS_ON
  - platform: gpio
    name: "Outlet 4"
    pin: GPIO14
    id: outlet_4
    restore_mode: ALWAYS_ON
  - platform: gpio
    name: "USB"
    pin: GPIO09
    id: usb
    restore_mode: ALWAYS_ON
  
status_led:
  pin:
    number: GPIO1
    inverted: no # no for on/yes for off
```

### Anything in the logs that might be useful for us?

```txt

```

### Additional information

INFO ESPHome 2025.8.1
INFO Reading configuration /root/config/kaden-powerstrip.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing kaden-powerstrip (board: generic-bk7231t-qfn32-tuya; framework: arduino; platform: libretiny @ 1.9.1)
--------------------------------------------------------------------------------
AssertionError: Error: Missing site-packages directory `/root/.platformio/penv/.libretiny/lib/python3.12/site-packages`:
  File "/opt/esphome/.venv/lib/python3.12/site-packages/platformio/builder/main.py", line 173:
    env.SConscript("$BUILD_SCRIPT")
  File "/root/.platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py", line 620:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/root/.platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/root/.platformio/platforms/libretiny/builder/main.py", line 24:
    env.ConfigurePythonVenv()
  File "/root/.platformio/packages/tool-scons/scons-local-4.8.1/SCons/Util/envs.py", line 252:
    return self.method(*nargs, **kwargs)
  File "/root/.platformio/platforms/libretiny/builder/python-venv.py", line 74:
    site_path.is_dir()
========================== [FAILED] Took 0.26 seconds ==========================

blindguynar, 2025-08-28:
The compiler is calling for python 3.12, but in 2025.8.0 the min python is 3.11 so i'm unsure why libretiny is calling for 3.12

`AssertionError: Error: Missing site-packages directory /root/.platformio/penv/.libretiny/lib/python3.12/site-packages:`

```
(.esphome) root@esphome:~/.platformio/penv/.libretiny/lib# ll
total 4
drwxr-xr-x 3 root root 4096 Feb 19  2025 python3.11
(.esphome) root@esphome:~/.platformio/penv/.libretiny/lib# 
```

mmstano, 2025-08-30:
I experienced this briefly, too. I cleaned the build files then I I did a hard reboot on HA and when I tried after that it worked.

blindguynar, 2025-09-09:
Just me, myself, and i?  Must be.  Built an entire new lxc and installed esphome, same issue, so at a loss.

blindguynar, 2025-09-09:
Tried to build lxc again and this time it worked. not shure what i did wrong the first time, but i built it on a diff host and moved it over this time as i noticed some issues resolving dns entries during previous lxc create.  

Closing issue.

closed 2025-09-09 by blindguynar