esphome/esphome #10487: Cannot Compile 2025.8.x for esp32-c3-devkitm-1 board

ziolelle, 2025-08-29:
### The problem

Errors in compilation of project for a esp32-c3-devkitm-1 board. This started with 2025.8.x. 
Up to version 2025.7.5 no errors were present.

### Which version of ESPHome has the issue?

2025.8.2

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

board: esp32-c3-devkitm-1

### YAML Config

```yaml
substitutions:
  nodename: mmwstudio
  devicename: mmwstudio
  friendly_devicename: "MMWave Studio"

esphome:
  name: mmwstudio
  friendly_name: mmwstudio

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:

ota:
  - platform: esphome

wifi:
  ssid: "dmtIoT"
  password: xxxxxxxxxxxxxxxxxxxxxxxx
  fast_connect: on

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: xxxxxxxxxxxxxxxxxxxx
    password: xxxxxxxxxxxxxxxxxxxx

captive_portal:

light:
  - platform: status_led
    name: "${friendly_devicename} Led"
    id: ${devicename}_led
    pin: 
      number: GPIO8
      inverted: True

ld2410:
  id: ld2410_studio
  
uart:
  tx_pin: GPIO4
  rx_pin: GPIO3
  baud_rate: 256000
  parity: NONE
  stop_bits: 1

button:
  - platform: ld2410
    factory_reset:
      name: "factory reset"
    restart:
      name: "restart"
    query_params:
      name: query params

text_sensor:
  - platform: ld2410
    version:
      name: "firmware version"
    mac_address:
      name: "mac address"

binary_sensor:
  - platform: ld2410
    has_target:
      name: Presence
    has_moving_target:
      name: Moving Target
    has_still_target:
      name: Still Target

switch:
  - platform: ld2410
    engineering_mode:
      name: "Engineering mode"
    bluetooth:
      name: "Bluetooth Control"

sensor:
  - platform: ld2410
    light:
      name: light
    moving_distance:
      name : Moving Distance
    still_distance:
      name: Still Distance
    moving_energy:
      name: Move Energy
    still_energy:
      name: Still Energy
    detection_distance:
      name: Detection Distance
    g0:
      move_energy:
        name: g0 move energy
      still_energy:
        name: g0 still energy
    g1:
      move_energy:
        name: g1 move energy
      still_energy:
        name: g1 still energy
    g2:
      move_energy:
        name: g2 move energy
      still_energy:
        name: g2 still energy
    g3:
      move_energy:
        name: g3 move energy
      still_energy:
        name: g3 still energy
    g4:
      move_energy:
        name: g4 move energy
      still_energy:
        name: g4 still energy
    g5:
      move_energy:
        name: g5 move energy
      still_energy:
        name: g5 still energy
    g6:
      move_energy:
        name: g6 move energy
      still_energy:
        name: g6 still energy
    g7:
      move_energy:
        name: g7 move energy
      still_energy:
        name: g7 still energy
    g8:
      move_energy:
        name: g8 move energy
      still_energy:
        name: g8 still energy

number:
  - platform: ld2410
    timeout:
      name: timeout
    light_threshold:
      name: light threshold
    max_move_distance_gate:
      name: max move distance gate
    max_still_distance_gate:
      name: max still distance gate
    g0:
      move_threshold:
        name: g0 move threshold
      still_threshold:
        name: g0 still threshold
    g1:
      move_threshold:
        name: g1 move threshold
      still_threshold:
        name: g1 still threshold
    g2:
      move_threshold:
        name: g2 move threshold
      still_threshold:
        name: g2 still threshold
    g3:
      move_threshold:
        name: g3 move threshold
      still_threshold:
        name: g3 still threshold
    g4:
      move_threshold:
        name: g4 move threshold
      still_threshold:
        name: g4 still threshold
    g5:
      move_threshold:
        name: g5 move threshold
      still_threshold:
        name: g5 still threshold
    g6:
      move_threshold:
        name: g6 move threshold
      still_threshold:
        name: g6 still threshold
    g7:
      move_threshold:
        name: g7 move threshold
      still_threshold:
        name: g7 still threshold
    g8:
      move_threshold:
        name: g8 move threshold
      still_threshold:
        name: g8 still threshold
```

### Anything in the logs that might be useful for us?

```txt
Processing mmwstudio (board: esp32-c3-devkitm-1; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32/releases/download/54.03.21-2/platform-espressif32.zip)
--------------------------------------------------------------------------------
INFO Package configuration completed successfully
INFO Package configuration completed successfully
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50402.0 (5.4.2) 
 - tool-cmake @ 3.30.2 
 - tool-esp-rom-elfs @ 2024.10.11 
 - tool-esptoolpy @ 5.0.2 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.13.1 
 - tool-scons @ 4.40801.0 (4.8.1) 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 14.2.0+20241119
Reading CMake configuration...
-- git rev-parse returned 'fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).'
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- The ASM compiler identification is GNU
-- Found assembler: /data/cache/platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-gcc - broken
-- Configuring incomplete, errors occurred!

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/tool-cmake/share/cmake-3.30/Modules/CMakeTestCCompiler.cmake:67 (message):
  The C compiler

    "/data/cache/platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: '/data/build/mmwstudio/.pioenvs/mmwstudio/CMakeFiles/CMakeScratch/TryCompile-ok97rg'
    
    Run Build Command(s): /data/cache/platformio/packages/tool-ninja/ninja -v cmTC_4302a
    [1/2] /data/cache/platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-gcc   -march=rv32imc_zicsr_zifencei -o CMakeFiles/cmTC_4302a.dir/testCCompiler.c.obj -c /data/build/mmwstudio/.pioenvs/mmwstudio/CMakeFiles/CMakeScratch/TryCompile-ok97rg/testCCompiler.c
    FAILED: [code=1] CMakeFiles/cmTC_4302a.dir/testCCompiler.c.obj 
    /data/cache/platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-gcc   -march=rv32imc_zicsr_zifencei -o CMakeFiles/cmTC_4302a.dir/testCCompiler.c.obj -c /data/build/mmwstudio/.pioenvs/mmwstudio/CMakeFiles/CMakeScratch/TryCompile-ok97rg/testCCompiler.c
    riscv32-esp-elf-gcc: fatal error: cannot execute 'as': posix_spawnp: No such file or directory
    compilation terminated.
    ninja: build stopped: subcommand failed.
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:571 (__project)
  CMakeLists.txt:3 (project)

========================== [FAILED] Took 1.96 seconds ==========================
```

swoboda1337, 2025-08-31:
https://github.com/pioarduino/platform-espressif32/releases/tag/54.03.21-2

Timeout was removed, platform updated, may have to wipe your platformio. Please open a new issue if you still can't build

closed 2025-08-31 by swoboda1337

ziolelle, 2025-09-01:
> https://github.com/pioarduino/platform-espressif32/releases/tag/54.03.21-2
> 
> Timeout was removed, platform updated, may have to wipe your platformio. Please open a new issue if you still can't build

Really sorry., but I don't get what you write... 
What timeout are you writing? 
Moreover: "wipe your platformio"... apart cleaning biuild files from the ESPHome Dashboard in Homeassistant, I cannot wipe anything else.
However the issue is still present

ssieb, 2025-09-01:
Re-install the addon with the erase data checkbox checked.

ziolelle, 2025-09-02:
> Re-install the addon with the erase data checkbox checked.

It worked, thank you so much for your support.