esphome/esphome #10498: I2C still not working 2025.9dev

latonita, 2025-08-29:
### The problem

Hi,
my i2c device stopped working after 2025.7.5.
then i see major change was done https://github.com/esphome/esphome/pull/10389 

unfortunately this did not fix the issue with my device (CST328, Touchscreen in Waveshare LCD 2.8 https://github.com/esphome/esphome/pull/8011) .

I took a look at the bus itself with logic analyzer of `read_register` operation: 
1) 2025.7.5 - write reg number, stop, start, read data - all ok
2) 2025.8.1 - initialy looks similar but somehow breaks after "address read", data is correpted.
3) 2025.8.2 and current 2025.9.dev - repeated start after write, data is corrupted.
4) then i though i will ensure STOP after writing register number so i made this:
```c++

#define TWO_TRANSACTIONS
i2c::ErrorCode CST328Touchscreen::read_touch_register_(uint16_t reg, uint8_t *data, size_t len) {
#ifndef TWO_TRANSACTIONS
  return this->read_register16(reg, data, len);
#endif

#ifdef TWO_TRANSACTIONS
  reg = convert_big_endian(reg);
  i2c::ErrorCode err = this->write_read((uint8_t *) &reg, 2, nullptr, 0);
  if (err != i2c::ERROR_OK) {
    ESP_LOGW(TAG, "write_read W err %d", err);
    return err;
  }
  err = this->write_read(nullptr, 0, data, len);
  if (err != i2c::ERROR_OK) {
    ESP_LOGW(TAG, "write_read R err %d", err);
  }
  return err;
#endif
}
```

5) i even tried to update function `write_readv` and insert one more job to STOP after writing data... no luck - esp-idf tells me 'too many jobs'. there is a limit of 8 jobs, but with my additional one it is 9.

here is the picture for 1...4

[image]

### Which version of ESPHome has the issue?

2025.9.0-dev

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Component causing the issue

i2c

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

clydebarrow, 2025-08-30:
So looking at the first two traces, they are identical except for the data being returned by the device. Since the request sent by the master is the same, but the response is different, there must be another factor at work, e.g. some earlier initialisation that is different so the chip is in a different state.

Pel1can111, 2025-08-30:
Not sure if its the same issue but I have a kmeteriso which works over i2c set up on a esp32-s3-devkitc-1. fails to report temperatures on 2025.8.2 but reverting to 2025.8.1 works.

latonita, 2025-08-30:
i downloaded waveshare esp-idf example, compiled it with esp-idf 5.5 and it works okay.
they are using special methods from esp-idf called `esp_lcd_panel_io_rx_param` and `esp_lcd_panel_io_tx_param`.
I will try to analyze what is the difference.

latonita, 2025-09-03:
@swoboda1337 Looks like there is some subtle difference, since simple call to `i2c_master_multi_buffer_transmit` works properly with cst328, but `write_readv` doesn't.

closed 2025-09-03 by swoboda1337