esphome/esphome #10499: ENS160 not working after updated to 2025.8.x

georgea62, 2025-08-30:
### The problem

It was working for months until the latest update. 
I have two devices with the same problem.
I have AHT20 and BMP085 on same device they work fine.

### Which version of ESPHome has the issue?

2025.8.x

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

ENS160_i2c

### YAML Config

```yaml
...
...
i2c:
  - id: bus_a
    sda: GPIO21
    scl: GPIO22
    scan: true

  - id: bus_b
    sda: GPIO18
    scl: GPIO19
    scan: True

sensor:
...
....
 - platform: ens160_i2c
    i2c_id: bus_a
    eco2:
      name: "ENS160 eCO2"
    tvoc:
      name: "ENS160 Total Volatile Organic Compounds"
    aqi:
      name: "ENS160 Air Quality Index"
      id: ENS160_Air_Quality_Index   
    address: 0x53
    compensation:
      temperature: Temperature
      humidity: AHT_humidity

  - platform: aht10
    variant: AHT20
    i2c_id: bus_a
    temperature:
      name: "AHT Temperature"
      id: AHT_temperature      
    humidity:
      name: "AHT Humidity"
      id: AHT_humidity

  - platform: bmp085
    i2c_id: bus_b
    temperature:
      name: "Temperature"
      id: Temperature
[... 184 lines cut ...]
[18:57:22][D][sensor:103]: 'AHT Humidity': Sending state 56.67820 % with 2 decimals of accuracy
[18:57:23][D][ens160:170]: ENS160 readings unavailable - Normal Operation but readings not ready
[18:57:24][D][bmp085.sensor:078]: Got Temperature=23.0 °C
[18:57:24][D][sensor:103]: 'Temperature': Sending state 23.03922 °C with 2 decimals of accuracy
[18:57:24][D][bmp085.sensor:124]: Got Pressure=985.2 hPa
[18:57:24][D][sensor:103]: 'Pressure': Sending state 985.18744 hPa with 1 decimals of accuracy
[18:57:29][D][sensor:103]: 'WiFi Signal dB': Sending state -37.00000 dBm with 0 decimals of accuracy
[18:57:29][D][sensor:103]: 'WiFi Signal Percent': Sending state 100.00000 % with 0 decimals of accuracy
[18:57:59][D][sensor:103]: 'WiFi Signal dB': Sending state -46.00000 dBm with 0 decimals of accuracy
[18:57:59][D][sensor:103]: 'WiFi Signal Percent': Sending state 100.00000 % with 0 decimals of accuracy
[18:58:21][D][sensor:103]: 'Wifi Channel': Sending state 11.00000  with 1 decimals of accuracy
[18:58:22][D][sensor:103]: 'Uptime Sensor': Sending state 302.11200 s with 0 decimals of accuracy
[18:58:22][D][text_sensor:069]: 'ENS160 Air Quality Rating': Sending state 'Not Available'
[18:58:22][D][sensor:103]: 'AHT Temperature': Sending state 23.64292 °C with 2 decimals of accuracy
[18:58:22][D][sensor:103]: 'AHT Humidity': Sending state 55.48649 % with 2 decimals of accuracy
[18:58:23][D][ens160:170]: ENS160 readings unavailable - Normal Operation but readings not ready
[18:58:24][D][bmp085.sensor:078]: Got Temperature=23.2 °C
[18:58:24][D][sensor:103]: 'Temperature': Sending state 23.23565 °C with 2 decimals of accuracy
[18:58:24][D][bmp085.sensor:124]: Got Pressure=985.2 hPa
[18:58:24][D][sensor:103]: 'Pressure': Sending state 985.21796 hPa with 1 decimals of accuracy
[18:58:29][D][sensor:103]: 'WiFi Signal dB': Sending state -33.00000 dBm with 0 decimals of accuracy
[18:58:29][D][sensor:103]: 'WiFi Signal Percent': Sending state 100.00000 % with 0 decimals of accuracy
[18:58:59][D][sensor:103]: 'WiFi Signal dB': Sending state -34.00000 dBm with 0 decimals of accuracy
[18:58:59][D][sensor:103]: 'WiFi Signal Percent': Sending state 100.00000 % with 0 decimals of accuracy
```

georgea62, 2025-08-30:
The same device after reset:

[11:32:14][E][ens160:286]: Error writing to register
[11:32:14][I][ens160:301]: Firmware Version: 0.0.0
...
[11:32:14][C][ens160_i2c.sensor:028]:   Address: 0x53
[11:32:14][E][component:145]:   ens160_base is marked FAILED: unspecified

I did another reset and I got logs similar to my previous message:

[11:35:25][C][ens160:276]: ENS160:
[11:35:25][D][ens160:298]: Setup successful
[11:35:25][I][ens160:301]: Firmware Version: 5.4.6

Still readings unavailable.

gomaed, 2025-09-01:
Having the exact same issue. Also tested both frameworks. Was Arduino, changed to esp-idf just to be sure and up-tp-date. Both getting the exact same logs on my side.

Also since 2025.08. Was working just fine with 2025.7.5 ans works now again with the older version.

georgea62, 2025-09-03:
I found (temporary) fix.
Add this to yaml:

external_components:
  - source:
      type: git
      url: https://github.com/esphome/esphome
      ref: 2025.7.1 
    components: [i2c]    
  

This confirms the problem is in i2c module.

hic-sol, 2025-09-06:
> I found (temporary) fix. Add this to yaml:
> 
> ```
> external_components:
>   - source:
>       type: git
>       url: https://github.com/esphome/esphome
>       ref: 2025.7.1 
>     components: [i2c]    
> ```
> 
> This confirms the problem is in i2c module.

I've been pulling my hair out over the same issue all day, and this finally fixed it for me. Thanks!

MondoElectronics, 2025-09-19:
I have the same issue, ESPHome Device Builder (Current version: 2025.9.0). I have tried the fix above, but experienced no change:

06:49:26.289][C][logger:273]: Logger:
[06:49:26.289][C][logger:273]:   Max Level: DEBUG
[06:49:26.289][C][logger:273]:   Initial Level: DEBUG
[06:49:26.291][C][logger:279]:   Log Baud Rate: 115200
[06:49:26.291][C][logger:279]:   Hardware UART: UART0
[06:49:26.299][C][logger:286]:   Task Log Buffer Size: 768
[06:49:26.302][C][i2c.idf:083]: I2C Bus:
[06:49:26.304][C][i2c.idf:084]:   SDA Pin: GPIO22
[06:49:26.304][C][i2c.idf:084]:   SCL Pin: GPIO21
[06:49:26.304][C][i2c.idf:084]:   Frequency: 50000 Hz
[06:49:26.309][C][i2c.idf:094]:   Recovery: bus successfully recovered
[06:49:26.319][I][i2c.idf:104]: Results from bus scan:
[06:49:26.324][I][i2c.idf:110]: Found device at address 0x38
[06:49:26.330][I][i2c.idf:110]: Found device at address 0x53
[06:49:26.333][C][spi:067]: SPI bus:
[06:49:26.340][C][spi:068]:   CLK Pin: GPIO18
[06:49:26.342][C][spi:069]:   SDI Pin: 
[06:49:26.345][C][spi:070]:   SDO Pin: GPIO23
[06:49:26.350][C][spi:075]:   Using HW SPI: SPI2_HOST
[06:49:26.370][C][ledc.output:150]: Output:
[06:49:26.372][C][ledc.output:151]:   Pin GPIO13
[06:49:26.376][C][ledc.output:152]:   Channel: 0
[06:49:26.376][C][ledc.output:152]:   PWM Frequency: 1000.0 Hz
[06:49:26.376][C][ledc.output:152]:   Phase angle: 0.0°
[06:49:26.376][C][ledc.output:152]:   Bit depth: 16
[06:49:26.379][C][rtttl:030]: Rtttl:
[06:49:26.379][C][rtttl:030]:   Gain: 0.600000
[06:49:26.397][C][ens160:276]: ENS160:
[06:49:26.399][E][ens160:280]: Communication failed
[06:49:26.402][I][ens160:301]: Firmware Version: 0.0.0
[06:49:26.406][C][ens160:352]:   Update Interval: 60.0s
[06:49:26.409][C][ens160:015]:   CO2 Sensor: 'ENS160 eCO2'
[06:49:26.409][C][ens160:015]:     State Class: 'measurement'
[06:49:26.409][C][ens160:015]:     Unit of Measurement: 'ppm'
[06:49:26.409][C][ens160:015]:     Accuracy Decimals: 0
[06:49:26.411][C][ens160:025]:     Device Class: 'carbon_dioxide'
[06:49:26.422][C][ens160:029]:     Icon: 'mdi:molecule-co2'
[06:49:26.426][C][ens160:015]:   TVOC Sensor: 'ENS160 Total Volatile Organic Compounds'
[06:49:26.426][C][ens160:015]:     State Class: 'measurement'
[06:49:26.426][C][ens160:015]:     Unit of Measurement: 'ppb'
[06:49:26.426][C][ens160:015]:     Accuracy Decimals: 0
[06:49:26.433][C][ens160:025]:     Device Class: 'volatile_organic_compounds_parts'
[06:49:26.443][C][ens160:029]:     Icon: 'mdi:radiator'
[06:49:26.446][C][ens160:015]:   AQI Sensor: 'ENS160 Air Quality Index'
[06:49:26.446][C][ens160:015]:     State Class: 'measurement'
[06:49:26.446][C][ens160:015]:     Unit of Measurement: ''
[06:49:26.446][C][ens160:015]:     Accuracy Decimals: 0
[06:49:26.453][C][ens160:025]:     Device Class: 'aqi'
[06:49:26.459][C][ens160:029]:     Icon: 'mdi:chemical-weapon'
[06:49:26.474][C][ens160:015]:     Temperature Compensation: 'Temperature °C'
[06:49:26.474][C][ens160:015]:     State Class: 'measurement'
[06:49:26.474][C][ens160:015]:     Unit of Measurement: '°C'
[06:49:26.474][C][ens160:015]:     Accuracy Decimals: 2
[06:49:26.477][C][ens160:025]:     Device Class: 'temperature'
[06:49:26.495][C][ens160:015]:     Humidity Compensation: 'Relative Humidity %'
[06:49:26.495][C][ens160:015]:     State Class: 'measurement'
[06:49:26.495][C][ens160:015]:     Unit of Measurement: '%'
[06:49:26.495][C][ens160:015]:     Accuracy Decimals: 2
[06:49:26.496][C][ens160:025]:     Device Class: 'humidity'
[06:49:26.499][C][ens160_i2c.sensor:028]:   Address: 0x52
[06:49:26.514][E][component:144]:   ens160_base is marked FAILED: unspecified
[06:49:26.517][C][aht10:156]: AHT10:
[06:49:26.520][C][aht10:157]:   Address: 0x38
[06:49:26.525][C][aht10:015]:   Temperature 'Temperature °C'
[06:49:26.525][C][aht10:015]:     State Class: 'measurement'
[06:49:26.525][C][aht10:015]:     Unit of Measurement: '°C'
[06:49:26.525][C][aht10:015]:     Accuracy Decimals: 2
[06:49:26.535][C][aht10:025]:     Device Class: 'temperature'
[06:49:26.545][C][aht10:015]:   Humidity 'Relative Humidity %'
[06:49:26.545][C][aht10:015]:     State Class: 'measurement'
[06:49:26.545][C][aht10:015]:     Unit of Measurement: '%'
[06:49:26.545][C][aht10:015]:     Accuracy Decimals: 2
[06:49:26.548][C][aht10:025]:     Device Class: 'humidity'

The temperature and humidity readings are OK, but the communication with the VOC/CO2/Aiq sensor fails.

marky852, 2025-09-19:
> The temperature and humidity readings are OK, but the communication with the VOC/CO2/Aiq sensor fails.
"[06:49:26.514][E][component:144]: ens160_base is marked FAILED: unspecified"

I have two of these sensors and they are both configured the same.  One works and the other doesn't and throws that error.  I'm beginning to suspect the latter sensor has gone kaput and needs to be replaced.   I'll try and inspect it this weekend.

gomaed, 2025-09-20:
For me the external components "temporary fix" does not work. Maybe because I have more than one sensor on the i2c bus. But only the ENS160 does not want to work. Also with 2025.9.1.

georgea62, 2025-10-16:
I tried the latest version 2025.10.0, ENS160 still has problem.

Joshyakadamien, 2025-10-25:
> 2025.10.3 still works, without messing around with archival components. Although this time I needed to clean build files and re-update (OTA). This is a method that has helped me and my friends with several updates, several times.

Can confirm, removed all external packages and after the initial hour readings started to come in.
Will let it roll to see stability...

B48D81EFCC, 2025-10-28:
> [...] I notice adding frequency: 400kHz to i2c works most times, but maybe this is just for me or the chip variety.

**Using 400kHz bus frequency fixed the issue for me,** I have multiple boards from different vendors, none worked.
I tried everything (esp-idf vs arduino, older esphome versions), nothing worked.

Using the latest esphome version wird 400kHz made all boards work.

Joshyakadamien, 2025-10-30:
Can also confirm frequency change resolves this.
Migrated the sensor to a board where i run i2c @800kHz
It reports values now on each boot, as mine already did the 24hr initial calibration within 3 minutes numbers are flowing in.
Only strange thing is, it always reports 0.0.0. version for the firmware. But regardless it is reporting properly.
Thanks for the tip,
Might be a good idea to update the sensor documentation too with this bit of info if one might suggest :) ;)

hrw, 2025-11-13:
I have ESP8266 with AHT20, BMP280 and ENS160 added to IKEA Vindriktning sensor. Also stopped working with 2025.10.5 version (it spent lot of time in a drawer so no idea which version was there earlier. Probably 2023.x)

```yaml
esp8266:
  board: d1_mini
  framework: 
    version: latest

i2c:
  sda: D1
  scl: D2
  scan: true
  id: bus_a

sensor:
  - platform: bmp280_i2c
#    address: 0x77
    temperature:
      name: TemperatureBMP
      oversampling: 16x
    pressure:
      name: Pressure
    update_interval: 60s

  - platform: aht10
#    address: 0x38
    variant: AHT20
    temperature:
      name: TemperatureAHT20
      id: aht20temperature
    humidity:
      name: Humidity
      id: aht20humidity
    update_interval: 60s

  - platform: ens160_i2c
    address: 0x53
    eco2:
      name: "eCO2"
    tvoc:
      name: "ENS160 Total Volatile Organic Compounds"
    aqi:
      name: "ENS160 Air Quality Index"
      id: ens160_air_quality_index
    update_interval: 60s
```

Device starts, I2C scan finds devices:

```
[12:11:56.160][C][i2c.arduino:073]: I2C Bus:
[12:11:56.160][C][i2c.arduino:074]:   SDA Pin: GPIO5
[12:11:56.160][C][i2c.arduino:074]:   SCL Pin: GPIO4
[12:11:56.160][C][i2c.arduino:074]:   Frequency: 50000 Hz
[12:11:56.172][C][i2c.arduino:090]:   Recovery: bus successfully recovered
[12:11:56.172][I][i2c.arduino:100]: Results from bus scan:
[12:11:56.184][I][i2c.arduino:106]: Found device at address 0x38
[12:11:56.184][I][i2c.arduino:106]: Found device at address 0x53
[12:11:56.198][I][i2c.arduino:106]: Found device at address 0x77
```

But then is unable to connect to any of them:

```
[12:11:56.358][C][bmp280_i2c.sensor:009]:   Address: 0x77
[12:11:56.359][C][bmp280.sensor:131]: BMP280:
[12:11:56.359][E][bmp280.sensor:134]: Communication failed
[12:11:56.364][C][bmp280.sensor:143]:   IIR Filter: OFF
[12:11:56.364][C][bmp280.sensor:362]:   Update Interval: 60.0s
[12:11:56.378][C][bmp280.sensor:015]:   Temperature 'TemperatureBMP'
[12:11:56.378][C][bmp280.sensor:015]:     State Class: 'measurement'
[12:11:56.378][C][bmp280.sensor:015]:     Unit of Measurement: '°C'
[12:11:56.378][C][bmp280.sensor:015]:     Accuracy Decimals: 1
[12:11:56.392][C][bmp280.sensor:025]:     Device Class: 'temperature'
[12:11:56.392][C][bmp280.sensor:147]:     Oversampling: 16x
[12:11:56.401][C][bmp280.sensor:015]:   Pressure 'Pressure'
[12:11:56.401][C][bmp280.sensor:015]:     State Class: 'measurement'
[12:11:56.401][C][bmp280.sensor:015]:     Unit of Measurement: 'hPa'
[12:11:56.401][C][bmp280.sensor:015]:     Accuracy Decimals: 1
[12:11:56.413][C][bmp280.sensor:025]:     Device Class: 'pressure'
[12:11:56.413][C][bmp280.sensor:149]:     Oversampling: 16x
[12:11:56.429][E][component:154]:   bmp280_base is marked FAILED: Communication failed
[12:11:56.429][C][aht10:156]: AHT10:
[12:11:56.429][C][aht10:157]:   Address: 0x38
[12:11:56.429][E][aht10:159]: Communication failed
[12:11:56.440][C][aht10:015]:   Temperature 'TemperatureAHT20'
[12:11:56.440][C][aht10:015]:     State Class: 'measurement'
[12:11:56.440][C][aht10:015]:     Unit of Measurement: '°C'
[12:11:56.440][C][aht10:015]:     Accuracy Decimals: 2
[12:11:56.460][C][aht10:025]:     Device Class: 'temperature'
[12:11:56.469][C][aht10:015]:   Humidity 'Humidity'
[12:11:56.469][C][aht10:015]:     State Class: 'measurement'
[12:11:56.469][C][aht10:015]:     Unit of Measurement: '%'
[12:11:56.469][C][aht10:015]:     Accuracy Decimals: 2
[12:11:56.469][C][aht10:025]:     Device Class: 'humidity'
[12:11:56.469][E][component:154]:   aht10.sensor is marked FAILED: unspecified
[12:11:56.481][C][ens160:276]: ENS160:
[12:11:56.481][E][ens160:280]: Communication failed
[12:11:56.495][I][ens160:301]: Firmware Version: 0.0.0
[12:11:56.495][C][ens160:362]:   Update Interval: 60.0s
[12:11:56.505][C][ens160:015]:   CO2 Sensor: 'eCO2'
[12:11:56.505][C][ens160:015]:     State Class: 'measurement'
[12:11:56.505][C][ens160:015]:     Unit of Measurement: 'ppm'
[12:11:56.505][C][ens160:015]:     Accuracy Decimals: 0
[12:11:56.506][C][ens160:025]:     Device Class: 'carbon_dioxide'
[12:11:56.516][C][ens160:029]:     Icon: 'mdi:molecule-co2'
[12:11:56.530][C][ens160:015]:   TVOC Sensor: 'ENS160 Total Volatile Organic Compounds'
[12:11:56.530][C][ens160:015]:     State Class: 'measurement'
[12:11:56.530][C][ens160:015]:     Unit of Measurement: 'ppb'
[12:11:56.530][C][ens160:015]:     Accuracy Decimals: 0
[12:11:56.539][C][ens160:025]:     Device Class: 'volatile_organic_compounds_parts'
[12:11:56.539][C][ens160:029]:     Icon: 'mdi:radiator'
[12:11:56.553][C][ens160:015]:   AQI Sensor: 'ENS160 Air Quality Index'
[12:11:56.553][C][ens160:015]:     State Class: 'measurement'
[12:11:56.553][C][ens160:015]:     Unit of Measurement: ''
[12:11:56.553][C][ens160:015]:     Accuracy Decimals: 0
[12:11:56.553][C][ens160:025]:     Device Class: 'aqi'
[12:11:56.564][C][ens160:029]:     Icon: 'mdi:chemical-weapon'
[12:11:56.564][C][ens160:313]:   Compensation: Not configured
[12:11:56.576][C][ens160_i2c.sensor:028]:   Address: 0x53
[12:11:56.576][E][component:154]:   ens160_base is marked FAILED: unspecified
```

Tried with `external_components` hack but it changed nothing.

linucks, 2026-01-05:
I've been experiencing similar issues. If I use the esp-idf framework, it sometimes works, but it's unreliable and I've had no success at all with the arduino framework. I've tried this with multiple different ens160 sensors and esp32 chips, so it's not a hardware issue.

If I use the following yaml file and esphome 2025.12.4:

```
esphome:
  name: ens160
  friendly_name: ENS160 Test

esp32:
  board: dfrobot_firebeetle2_esp32e
  framework:
    type: esp-idf
    # type: arduino # for captive portal support - esp160 not working currently

# Enable logging
logger:
  level: VERBOSE
  logs:
    ens160: VERBOSE
    ens160_i2c: VERBOSE

# --- I²C bus ---
i2c:
  id: bus_a
  sda: GPIO21
  scl: GPIO22
  scan: true
  frequency: 50kHz

sensor:
  - platform: ens160_i2c
    i2c_id: bus_a
    address: 0x52
    update_interval: 5s
    eco2:
      name: "ENS160 eCO2"
    tvoc:
      name: "ENS160 Total Volatile Organic Compounds"
    aqi:
      name: "ENS160 Air Quality Index"
```

with the esp-idf framework I can (sometimes) get it to work as the following shows:
```
INFO Successfully uploaded program.
INFO Starting log output from /dev/cu.wchusbserial1120 with baud rate 115200
[19:14:52.498]\xa1jmYum\xa5\xc9c.idf:181]: Sending 3 jobs
[19:14:52.501][V][i2c.idf:184]: TX to 15 failed: not acked
[19:14:52.506][V][i2c.idf:139]: Writing 0 bytes, reading 0 bytes
[19:14:52.512][V][i2c.idf:142]: 0x16 BUS PROBE
[19:14:52.515][V][i2c.idf:181]: Sending 3 jobs
[19:14:52.520][V][i2c.idf:184]: TX to 16 failed: not acked
[19:14:52.526][V][i2c.idf:139]: Writing 0 bytes, reading 0 bytes
[19:14:52.529][V][i2c.idf:142]: 0x17 BUS PROBE
<snip/>
[19:14:54.240][V][i2c.idf:142]: 0x77 BUS PROBE
[19:14:54.245][V][i2c.idf:181]: Sending 3 jobs
[19:14:54.251][V][i2c.idf:184]: TX to 77 failed: not acked
[19:14:54.254][C][component:208]: Setup i2c took 1990ms
[19:14:54.259][V][component:201]: Setup preferences
[19:14:54.265][V][scheduler:144]: Scheduler interval for  is 60000ms, offset 3595ms
[19:14:54.271][C][component:208]: Setup preferences took 0ms
[19:14:54.276][V][component:201]: Setup ens160_base
[19:14:54.282][V][scheduler:144]: Scheduler interval for update is 20000ms, offset 4650ms
[19:14:54.287][V][i2c.idf:139]: Writing 1 bytes, reading 2 bytes
[19:14:54.293][V][i2c.idf:150]: 0x52 TX 00
[19:14:54.295][V][i2c.idf:162]: 0x52 RX bytes 2
[19:14:54.298][V][i2c.idf:181]: Sending 8 jobs
[19:14:54.304][V][i2c.idf:139]: Writing 2 bytes, reading 0 bytes
[... 504 lines cut ...]
[19:27:26.474][V][ens160:158]: Status: ENS160 VALIDITY FLAG 0x00
[19:27:26.479][V][ens160:159]: Status: ENS160 NEWDAT bit    0x0
[19:27:26.485][V][ens160:161]: Status: ENS160 NEWGPR bit    0x0
[19:27:26.492][D][ens160:170]: ENS160 readings unavailable - Normal Operation but readings not ready
[19:27:46.446][V][i2c.idf:139]: Writing 1 bytes, reading 1 bytes
[19:27:46.451][V][i2c.idf:150]: 0x52 TX 20
[19:27:46.454][V][i2c.idf:162]: 0x52 RX bytes 1
[19:27:46.457][V][i2c.idf:181]: Sending 7 jobs
[19:27:46.462][V][ens160:154]: Status: ENS160 STATAS bit    0x0
[19:27:46.468][V][ens160:156]: Status: ENS160 STATER bit    0x0
[19:27:46.473][V][ens160:158]: Status: ENS160 VALIDITY FLAG 0x00
[19:27:46.479][V][ens160:159]: Status: ENS160 NEWDAT bit    0x0
[19:27:46.484][V][ens160:161]: Status: ENS160 NEWGPR bit    0x0
[19:27:46.492][D][ens160:170]: ENS160 readings unavailable - Normal Operation but readings not ready
[19:28:06.445][V][i2c.idf:139]: Writing 1 bytes, reading 1 bytes
[19:28:06.450][V][i2c.idf:150]: 0x52 TX 20
[19:28:06.453][V][i2c.idf:162]: 0x52 RX bytes 1
[19:28:06.456][V][i2c.idf:181]: Sending 7 jobs
[19:28:06.462][V][ens160:154]: Status: ENS160 STATAS bit    0x0
[19:28:06.467][V][ens160:156]: Status: ENS160 STATER bit    0x0
[19:28:06.473][V][ens160:158]: Status: ENS160 VALIDITY FLAG 0x00
[19:28:06.478][V][ens160:159]: Status: ENS160 NEWDAT bit    0x0
[19:28:06.484][V][ens160:161]: Status: ENS160 NEWGPR bit    0x0
[19:28:06.491][D][ens160:170]: ENS160 readings unavailable - Normal Operation but readings not ready
```

linucks, 2026-01-06:
Just to add to my comment above. I've tried changing the bus speed to 400khz as suggested and that seems to work with both arduino and esp-idf frameworks, so it looks like an issue with the i2c bus code.

B48D81EFCC, 2026-01-11:
Did you set` scan: true`  on the i2c hub?
Were two devices found?

MrZaibot, 2026-03-04:
Setting I2C speed to 800kHz fixed my problem.