esphome/esphome #10515: Latest ESP-IDF Version breaks UART Connectivity - and downgrading breaks I2C (from 8.2)

Xorfee, 2025-09-01:
### The problem

This was my post previously to get the UART Pins used for my mmWave Sensor:
https://github.com/esphome/esphome/issues/10405

The workaround was to upgrade it to a lower esp-idf version in order to make the dedicated UART Pins 
usable with my mmWave Sensor.

Now - changing the parameters (recompile) and going back to V5.4.1 and lower makes it impossible to
compile my I2C interface, since I get an error now saying:

  This feature requires at least framework version 5.4.2.

  - sda: 0
    scl: 1
    frequency: 15khz

This happens after ESPHome 8.2 Upgrade (from 8.1)

This is a serious issue for me, because I am using nearly every pin on my ESP32-C6 from SPI to I2C along with UART. This also breaks a lot of client devices since those are using the same pins.

### Which version of ESPHome has the issue?

latest 8.2

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

UART / i2C

### YAML Config

```yaml
esphome:
  name: wific6matter
  friendly_name: WiFiC6Matter
  platformio_options:
    board_build.f_cpu: 160000000L
    board_build.flash_mode: dio

esp32:
  board: esp32-c6-devkitm-1
  framework:
    type: esp-idf
    version: 5.4.1

    

wifi:
  ssid: IoT
  password: SmartHome2024
  ap:
    ssid: "AiOHotspot"
    password: "SmartHome2024"

mdns:
  disabled: false

i2c:
  sda: GPIO0
  scl: GPIO1
  frequency: 15khz
  scan: false
#  timeout: 5s

ld2412:
  id: ld2412_radar
  uart_id: presence 

uart:
  - id: zigbee
    rx_pin: 02
    tx_pin: 03
    baud_rate: 115200
  - id: presence
    tx_pin: 17
    rx_pin: 16
    baud_rate: 115200
    parity: NONE
    stop_bits: 1
```

### Anything in the logs that might be useful for us?

```txt
See my previous post. 

Work around was to downgrade to ESPHome 8.1 and use Version 5.4.1 as standard,
but for future updates, I am not able to upgrade anything anymore.
```

tabascoz, 2025-09-02:
looks similar to the issue I reported here https://github.com/esphome/esphome/issues/10370 as the component is dependent on serial connection. 

May I ask you would it be possible to test changing the serial pins to the ones I have at the board it works ok?? I am physically away from the hw  and I cannot test now - only in a couple of weeks. 

```
  tx_pin: GPIO18
  rx_pin: GPIO19 
```

i know it looks silly but i did a diff between the boards and despite some thermostat and wifi connection parameters, everything with serial and dalybms object are exactly the same...

Xorfee, 2025-09-02:
> looks similar to the issue I reported here [#10370](https://github.com/esphome/esphome/issues/10370) as the component is dependent on serial connection.
>
> May I ask you would it be possible to test [...]

Hi it won't work - cause this is a custom PCB which doesnt have any exposed pins, yet all free pins are already assigned.

axa88, 2025-12-09:
> I am wondering how setbacks like that can be possible .....

highly correlated to the `total users` : `users who can and will contribute` ratio

RussH007, 2026-02-18:
affects me too using waveshare esp32-s3-lcd-7 and a02yyuw.  Was working last year but whatever I try with idf versions or regressing i2c , ch422g and uart components the distance sensor is stuck at NA.

closed 2026-05-27 by github-actions[bot]