esphome/esphome #10516: LD2420 memory leak issue?

nanomonster, 2025-09-01:
### The problem

I am experimenting with an esp32 devkit v1 MCU w. LD2420 sensor using ESPHome in Home Assistant. Everything works as it should for a number of hours, then the MCU hangs, requiring a full hardware reset to restore operation.

This issue was reported here prior: [https://community.home-assistant.io/t/ld2420-based-presence-sensor-suddenly-unstable/913508/7](url)

My best guess is a memory leak that ends up shutting down the MCU.

### Which version of ESPHome has the issue?

2025.8.2

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

esp32 devkit v1

### YAML Config

```yaml
esphome: 
  min_version: 2025.5.0
  name_add_mac_suffix: false
  name: presence-sensor-a
  friendly_name: Presence Sensor A

esp32: 
  variant: ESP32
  board: esp32dev
  framework:
    type: esp-idf

# Enable logging 
logger: 

api:
  encryption:
    key: "mMOEeUoP8h7exyESy4ueWEL4DMH8t3n4co/IcIzgIPg="

ota: 
  platform: esphome
  password: "1b17aa44abc807daea66d1466906ffad" 

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password 
  fast_connect: true 

  ap:
    ssid: "Presence-Sensor-1"
    password: "8EvZiZx1q8tv"

captive_portal: 
web_server: 
  port: 80 
  auth: 
    username: admin
    password: "59295448" #the password to access, possibly to the web server ESPHome on the board 

uart: 
  id: ld2420_uart 
  tx_pin: GPIO22 
  rx_pin: GPIO23 
  baud_rate: 115200 
  parity: NONE 
  stop_bits: 1 

ld2420: 
text_sensor: 
  - platform: ld2420 
    fw_version: 
      name: Firmware version 
sensor: 
  - platform: ld2420 
    moving_distance: 
      name : Moving distance 
binary_sensor: 
  - platform: ld2420 
    has_target: 
      name: Presence 
switch: 
  - platform: restart 
    name: Restart ESPHome node 
select: 
  - platform: ld2420 
    operating_mode: 
      name: Operating mode 
number: 
  - platform: ld2420 
    presence_timeout: 
      name: Presence detection timeout 
    min_gate_distance: 
      name: Minimum gate distance 
    max_gate_distance: 
      name: Maximum gate distance 
    gate_select: 
      name: Gate selection 
    still_threshold: 
      name: Set stillness threshold 
    move_threshold: 
      name: Set movement threshold 
button: 
  - platform: ld2420 
    apply_config: 
      name: Apply configuration 
    factory_reset: 
      name: Reset to factory settings 
    restart_module: 
      name: Restart module 
    revert_config: 
      name: Undo changes
```

### Anything in the logs that might be useful for us?

```txt
Log downloaded from presence detection module
Everything seem normal until it shuts down

INFO ESPHome 2025.8.2
INFO Reading configuration /config/esphome/esphome-web-1a1b4c.yaml...
INFO Starting log output from 10.0.0.80 using esphome API
INFO Successfully resolved presence-sensor-a @ 10.0.0.80 in 0.000s
INFO Successfully connected to presence-sensor-a @ 10.0.0.80 in 0.070s
INFO Successful handshake with presence-sensor-a @ 10.0.0.80 in 0.084s
[09:47:48][I][app:200]: ESPHome version 2025.8.1 compiled on Aug 30 2025, 11:12:26
[09:47:48][C][wifi:659]: WiFi:
[09:47:48][C][wifi:442]:   Local MAC: EC:E3:34:1A:1B:4C
[09:47:48][C][wifi:447]:   SSID: 'Videtron wifi'[redacted]
[09:47:48][C][wifi:450]:   IP Address: 10.0.0.80
[09:47:48][C][wifi:454]:   BSSID: 1E:9D:72:35:EF:CB[redacted]
[09:47:48][C][wifi:454]:   Hostname: 'presence-sensor-a'
[09:47:48][C][wifi:454]:   Signal strength: -57 dB ▂▄▆█
[09:47:48][C][wifi:465]:   Channel: 11
[09:47:48][C][wifi:465]:   Subnet: 255.255.255.0
[09:47:48][C][wifi:465]:   Gateway: 10.0.0.1
[09:47:48][C][wifi:465]:   DNS1: 24.201.245.77
[09:47:48][C][wifi:465]:   DNS2: 24.200.243.189
[09:47:48][C][logger:252]: Logger:
[09:47:48][C][logger:252]:   Max Level: DEBUG
[09:47:48][C][logger:252]:   Initial Level: DEBUG
[09:47:48][C][logger:258]:   Log Baud Rate: 115200
[09:47:48][C][logger:258]:   Hardware UART: UART0
[09:47:48][C][logger:265]:   Task Log Buffer Size: 768
[09:47:48][C][uart.idf:153]: UART Bus 1:
[09:47:48][C][uart.idf:154]:   TX Pin: GPIO22
[09:47:48][C][uart.idf:155]:   RX Pin: GPIO23
[09:47:48][C][uart.idf:157]:   RX Buffer Size: 256
[09:47:48][C][uart.idf:159]:   Baud Rate: 115200 baud
[09:47:48][C][uart.idf:159]:   Data Bits: 8
[09:47:48][C][uart.idf:159]:   Parity: NONE
[09:47:48][C][uart.idf:159]:   Stop bits: 1
[09:47:48][C][ld2420:187]: LD2420:
[09:47:48][C][ld2420:187]:   Firmware version:  v1.6.1
[09:47:48][C][ld2420:192]: Number:
[09:47:48][C][ld2420:193]:   Gate Timeout: 'Presence detection timeout'
[09:47:48][C][ld2420:193]:     Icon: 'mdi:timelapse'
[09:47:48][C][ld2420:193]:     Unit of Measurement: 's'
[09:47:48][C][ld2420:194]:   Gate Max Distance: 'Maximum gate distance'
[09:47:48][C][ld2420:194]:     Icon: 'mdi:motion-sensor'
[09:47:48][C][ld2420:194]:     Device Class: 'distance'
[09:47:48][C][ld2420:195]:   Gate Min Distance: 'Minimum gate distance'
[09:47:48][C][ld2420:195]:     Icon: 'mdi:motion-sensor'
[09:47:48][C][ld2420:195]:     Device Class: 'distance'
[09:47:48][C][ld2420:196]:   Gate Select: 'Gate selection'
[09:47:48][C][ld2420:196]:     Icon: 'mdi:motion-sensor'
[09:47:48][C][ld2420:196]:     Device Class: 'distance'
[09:47:48][C][ld2420:198]:   Gate Move Threshold: 'Set movement threshold'
[09:47:48][C][ld2420:198]:     Icon: 'mdi:motion-sensor'
[09:47:48][C][ld2420:199]:   Gate Still Threshold:: 'Set stillness threshold'
[09:47:48][C][ld2420:199]:     Icon: 'mdi:motion-sensor'
[09:47:48][C][ld2420:203]:   Apply Config: 'Apply configuration'
[09:47:48][C][ld2420:203]:     Icon: 'mdi:restart-alert'
[09:47:48][C][ld2420:204]:   Revert Edits: 'Undo changes'
[09:47:48][C][ld2420:204]:     Icon: 'mdi:restart'
[09:47:48][C][ld2420:205]:   Factory Reset: 'Reset to factory settings'
[09:47:48][C][ld2420:205]:     Icon: 'mdi:database'
[09:47:48][C][ld2420:206]:   Restart Module: 'Restart module'
[09:47:48][C][ld2420:206]:     Icon: 'mdi:database'
[09:47:48][C][ld2420:208]: Select:
[09:47:48][C][ld2420:209]:   Operating Mode 'Operating mode'
[09:47:48][C][ld2420.text_sensor:011]: Text Sensor:
[09:47:48][C][ld2420.text_sensor:012]:   Firmware 'Firmware version'
[09:47:48][C][ld2420.text_sensor:012]:     Icon: 'mdi:chip'
[09:47:48][C][ld2420.sensor:011]: Sensor:
[09:47:48][C][ld2420.sensor:012]:   Distance 'Moving distance'
[09:47:48][C][ld2420.sensor:012]:     State Class: ''
[09:47:48][C][ld2420.sensor:012]:     Unit of Measurement: 'cm'
[09:47:48][C][ld2420.sensor:012]:     Accuracy Decimals: 0
[09:47:48][C][ld2420.sensor:012]:     Device Class: 'distance'
[09:47:48][C][ld2420.binary_sensor:011]: Binary Sensor:
[09:47:48][C][ld2420.binary_sensor:012]:   Presence 'Presence'
[09:47:48][C][ld2420.binary_sensor:012]:     Device Class: 'occupancy'
[09:47:48][C][restart:087]: Restart Switch 'Restart ESPHome node'
[09:47:48][C][restart:087]:   Restore Mode: always OFF
[09:47:48][C][restart:095]:   Icon: 'mdi:restart'
[09:47:48][C][captive_portal:099]: Captive Portal:
[09:47:48][C][web_server:309]: Web Server:
[09:47:48][C][web_server:309]:   Address: presence-sensor-a.local:80
[09:47:48][C][esphome.ota:075]: Over-The-Air updates:
[09:47:48][C][esphome.ota:075]:   Address: presence-sensor-a.local:3232
[09:47:48][C][esphome.ota:075]:   Version: 2
[09:47:48][C][esphome.ota:082]:   Password configured
[09:47:48][C][safe_mode:018]: Safe Mode:
[09:47:48][C][safe_mode:019]:   Boot considered successful after 60 seconds
[09:47:48][C][safe_mode:019]:   Invoke after 10 boot attempts
[09:47:48][C][safe_mode:019]:   Remain for 300 seconds
[09:47:48][C][web_server.ota:224]: Web Server OTA
[09:47:48][C][api:205]: Server:
[09:47:48][C][api:205]:   Address: presence-sensor-a.local:6053
[09:47:48][C][api:210]:   Noise encryption: YES
[09:47:48][C][mdns:124]: mDNS:
[09:47:48][C][mdns:124]:   Hostname: presence-sensor-a
[09:47:50][D][sensor:103]: 'Moving distance': Sending state 35.00000 cm with 0 decimals of accuracy
[09:47:56][D][sensor:103]: 'Moving distance': Sending state 74.00000 cm with 0 decimals of accuracy
[09:47:57][D][sensor:103]: 'Moving distance': Sending state 66.00000 cm with 0 decimals of accuracy
...
```

### Additional information

none

KrX3D, 2025-09-03:
i get the same problem with ld2450 and esp32 c3 super mini AND esp32 s2 mini

nanomonster, 2025-09-13:
Last update: it's been working non-stop for several days now so I'm closing this issue.

closed 2025-09-13 by nanomonster