esphome/esphome #10521: Wifi wont connect when using ssd1306_i2c screen on esp8266 (wemos d1 mini)

ZombiesLoveMe, 2025-09-01:
### The problem

when using the display component on esphome it now breaks the wifi with the error below, if i remove the display component everything works as it used to

`[17:37:58][I][wifi:329]: Connecting to 'SmartHome'
[17:38:00][W][wifi_esp8266:513]: Disconnected ssid='SmartHome' bssid=[redacted] reason='Auth Expired'
[17:38:01][W][wifi:716]: Connecting to network failed
[17:38:01][D][wifi:771]: Retrying with hidden networks`

### Which version of ESPHome has the issue?

2025.8.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Component causing the issue

Display specifically ssd1306_i2c

### YAML Config

```yaml
esphome:
  name: mantella
  friendly_name: Mantella
  on_boot:
    priority: 800
    # ...
    then:
      - switch.turn_off: Wemos_light

substitutions:
  devicename: mantella
  friendly: "Mantella"
  temp_threshold: "27.0"
  humid_threshold: "70.0"

wifi:
  ssid: "REDACTED"
  password: "REDACTED"
  
  power_save_mode: none
  reboot_timeout: 15min
  ap:
    ssid: "REDACTED"
    password: "REDACTED"

api:
  encryption:
    key: "REDACTED"

ota:
  - platform: esphome
    password: "REDACTED"

packages:
  base: !include common/ESP_Probe_Base_Conf.yaml

##### CONTINUED (display component currently commented out) #####

substitutions:
  devicename: esp_probe
  friendly: "ESP Probe"
  temp_threshold: "27.0"
  humid_threshold: "70.0"

esphome:
  name: ${devicename}
  friendly_name: ${friendly}
  on_boot:
    - priority: 800
      then:
        - switch.turn_off: Wemos_light
    - priority: -10
      then:
        - script.execute: watchdog

esp8266:
  board: d1_mini
  framework:
    version: latest

logger:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: none
  reboot_timeout: 15min
  ap:
    ssid: "${friendly} Fallback"
    password: !secret wifi_fallback

captive_portal:

script:
  - id: watchdog
    mode: restart
    then:
      - delay: 10min
      - lambda: ESP.restart();

external_components:
  - source: github://pr#7217
    components: [ sht3xd ]

globals:
  - id: screen_page
    type: int
    restore_value: no
    initial_value: '0'

  - id: max_temp
    type: float
    restore_value: yes
    initial_value: ${temp_threshold}

  - id: min_humidity
    type: float
    restore_value: yes
    initial_value: ${humid_threshold}

interval:
  - interval: 30s
    then:
      - lambda: |-
          id(screen_page)++;
          if (id(screen_page) > 2) id(screen_page) = 0;

# NeoPixel
light:
  - platform: neopixelbus
    id: ${devicename}_statuslight
    type: GRB
    variant: ws2812
    pin: D6
    num_leds: 8
    name: "${friendly} Fan Status"

# Fonts
#font:
#  - file: 'BebasNeue-Regular.ttf'
#    id: font2
#    size: 28

# I2C & OLED
#i2c:
#  sda: D2
#  scl: D1
#  scan: True
#  timeout: 10ms

#image:
#  - file: mdi:thermometer
#    type: binary
#    id: Thermometericon
#    resize: 25x25
#  - file: mdi:water-percent
#    type: binary
#    id: humidityicon
#    resize: 25x25

#display:
#  - platform: ssd1306_i2c
#    model: "sh1106 128x64"
#    reset_pin: D0
#    address: 0x3C
#    contrast: 100%
#    lambda: |-
#      if (id(screen_page) == 0) {
#        it.image(0, 2, id(Thermometericon));
#        it.printf(25, 0, id(font2), "%.1fc°", id(${devicename}_temp).state);
#        it.image(0, 32, id(humidityicon));
#        it.printf(25, 30, id(font2), "%.1f%%", id(${devicename}_hum).state);
#      }
#      else if (id(screen_page) == 1) {
#        it.printf(0, 0, id(font2), "WiFi: %.0f dB", id(${devicename}_wifi_signal).state);
#        it.printf(0, 25, id(font2), "HA: %s", id(api_connected).state ? "OK" : "Lost");
#      }
#      else if (id(screen_page) == 2) {
#        bool warn = false;
#        if (id(${devicename}_temp).state > id(max_temp)) {
#          it.printf(0, 0, id(font2), "!! TEMP HIGH !!");
#          warn = true;
#        }
#        if (id(${devicename}_hum).state < id(min_humidity)) {
#          it.printf(0, 25, id(font2), "!! HUMID LOW !!");
#          warn = true;
#        }
#        if (!warn) {
#          it.printf(0, 15, id(font2), "All conditions OK");
#        }
#      }

sensor:
  - platform: sht3xd
    id: probe_sensor
    temperature:
      name: "${friendly} Temperature"
      id: ${devicename}_temp
    humidity:
      name: "${friendly} Humidity"
      id: ${devicename}_hum
    address: 0x44
    update_interval: 60s

  - platform: wifi_signal
    name: "${friendly} WiFi Signal"
    id: ${devicename}_wifi_signal
    update_interval: 60s

  - platform: template
    name: "${friendly} Dew Point"
    unit_of_measurement: "°C"
    lambda: |-
      const float a = 17.27;
      const float b = 237.7;
      float temp = id(${devicename}_temp).state;
      float hum = id(${devicename}_hum).state;
      float alpha = ((a * temp) / (b + temp)) + log(hum/100.0);
      return (b * alpha) / (a - alpha);

binary_sensor:
  - platform: template
    name: "${friendly} SHT3xd Heater State"
    lambda: return id(probe_sensor).is_heater_enabled();

  - platform: status
    id: api_connected

switch:
  - platform: gpio
    id: Wemos_light
    inverted: True
    name: "${friendly} Wemos Light"
    pin: 
      number: D4

button:
  - platform: template
    name: "${friendly} Enable SHT3xd Heater"
    on_press:
      then:
        - lambda: id(probe_sensor).set_heater_enabled(true);

  - platform: template
    name: "${friendly} Disable SHT3xd Heater"
    on_press:
      then:
        - lambda: id(probe_sensor).set_heater_enabled(false);
```

### Anything in the logs that might be useful for us?

```txt
[17:43:02][I][wifi:329]: Connecting to 'SmartHome'
[17:43:04][W][wifi_esp8266:513]: Disconnected ssid='SmartHome' bssid=[redacted] reason='Auth Expired'
[17:43:05][W][wifi:716]: Connecting to network failed
[17:43:05][D][wifi:771]: Retrying with hidden networks
[17:43:05][I][wifi:329]: Connecting to 'SmartHome'
[17:43:07][W][wifi_esp8266:513]: Disconnected ssid='SmartHome' bssid=[redacted] reason='Auth Expired'
[17:43:07][W][wifi:716]: Connecting to network failed
[17:43:07][D][wifi:771]: Retrying with hidden networks
[17:43:12][I][wifi:329]: Connecting to 'SmartHome'
[17:43:15][W][wifi_esp8266:513]: Disconnected ssid='SmartHome' bssid=[redacted] reason='Auth Expired'
[17:43:15][W][wifi:716]: Connecting to network failed
[17:43:15][D][wifi:771]: Retrying with hidden networks
[17:43:15][I][wifi:329]: Connecting to 'SmartHome'
[17:43:17][W][wifi_esp8266:513]: Disconnected ssid='SmartHome' bssid=[redacted] reason='Auth Expired'
[17:43:17][W][wifi:716]: Connecting to network failed
[17:43:18][D][wifi:771]: Retrying with hidden networks
[17:43:22][I][wifi:329]: Connecting to 'SmartHome'
[17:43:25][W][wifi_esp8266:513]: Disconnected ssid='SmartHome' bssid=[redacted] reason='Auth Expired'
[17:43:25][W][wifi:716]: Connecting to network failed
[17:43:25][D][wifi:771]: Retrying with hidden networks
[17:43:25][I][wifi:329]: Connecting to 'SmartHome'
[17:43:27][W][component:423]: interval took a long time for an operation (327 ms)
[17:43:27][W][component:424]: Components should block for at most 30 ms
[17:43:27][W][wifi_esp8266:513]: Disconnected ssid='SmartHome' bssid=[redacted] reason='Auth Expired'
[17:43:28][W][wifi:716]: Connecting to network failed
[17:43:28][D][wifi:771]: Retrying with hidden networks
[17:43:33][I][wifi:329]: Connecting to 'SmartHome'
[17:43:35][W][wifi_esp8266:513]: Disconnected ssid='SmartHome' bssid=[redacted] reason='Auth Expired'
[17:43:35][W][wifi:716]: Connecting to network failed
[17:43:35][W][wifi:128]: Restarting adapter
[17:43:35][I][wifi:329]: Connecting to 'SmartHome'
[17:43:38][W][wifi_esp8266:513]: Disconnected ssid='SmartHome' bssid=[redacted] reason='Auth Expired'
[17:43:38][W][wifi:716]: Connecting to network failed
[17:43:38][D][wifi:771]: Retrying with hidden networks
[17:43:43][I][wifi:329]: Connecting to 'SmartHome'
[17:43:46][W][wifi_esp8266:513]: Disconnected ssid='SmartHome' bssid=[redacted] reason='Auth Expired'
[17:43:46][W][wifi:716]: Connecting to network failed
[17:43:46][D][wifi:771]: Retrying with hidden networks
[17:43:46][I][wifi:329]: Connecting to 'SmartHome'
[17:43:48][W][wifi_esp8266:513]: Disconnected ssid='SmartHome' bssid=[redacted] reason='Auth Expired'
[17:43:48][W][wifi:716]: Connecting to network failed
```

JochenFriedrich, 2025-09-05:
Same here.

Setting:
```
i2c:
  scan: false
  speed: 100kHz
```

makes it work again.

rtdmp2y9sw-ops, 2025-11-08:
Thanks Jochen,

After changing 'speed: 100kHz' in 'frequency: 100kHz' it works also in my application.