esphome/esphome #10522: 2025.8.x on esp32s2: adc delivers bizarre values

HenningGithub, 2025-09-01:
### The problem

starting with 2025.8.0, adc delivers bizarre values.
no changes up to and including 2025.8.2

I could claim esphome is trying to drown my plants ;)

### Which version of ESPHome has the issue?

2025.8.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

adc

### YAML Config

```yaml
esp32:
  board: lolin_s2_mini
  framework:
    type: esp-idf

sensor:
  - platform: adc
    pin: GPIO3
    name: "${prefix} Moisture1 raw"
    id: moisture1_raw   
    update_interval: 15s
    attenuation: auto
    filters:
      - sliding_window_moving_average:
          window_size: 20
          send_every: 4
```

### Anything in the logs that might be useful for us?

```txt
[19:38:50][C][adc.esp32:121]: ADC Sensor 'Wohnzimmer Plants Moisture1 raw'
[19:38:50][C][adc.esp32:121]:   State Class: 'measurement'
[19:38:50][C][adc.esp32:121]:   Unit of Measurement: 'V'
[19:38:50][C][adc.esp32:121]:   Accuracy Decimals: 2
[19:38:50][C][adc.esp32:121]:   Device Class: 'voltage'
[19:38:50][C][adc.esp32:122]:   Pin: GPIO3
[19:38:50][C][adc.esp32:123]:   Channel:       2
[19:38:50][C][adc.esp32:123]:   Unit:          ADC1
[19:38:50][C][adc.esp32:123]:   Attenuation:   Auto
[19:38:50][C][adc.esp32:123]:   Samples:       1
[19:38:50][C][adc.esp32:123]:   Sampling mode: average
[19:38:50][C][adc.esp32:133]:   Setup Status:
[19:38:50][C][adc.esp32:133]:     Handle Init:  OK
[19:38:50][C][adc.esp32:133]:     Config:       OK
[19:38:50][C][adc.esp32:133]:     Calibration:  OK
[19:38:50][C][adc.esp32:133]:     Overall Init: OK
[19:38:50][C][adc.esp32:143]:   Update Interval: 15.0s

[19:39:45][D][sensor:103]: 'Wohnzimmer Plants Moisture1 raw': Sending state 30038410.00000 V with 2 decimals of accuracy
[19:40:45][D][sensor:103]: 'Wohnzimmer Plants Moisture1 raw': Sending state 35233488.00000 V with 2 decimals of accuracy
```

edwardtfn, 2025-09-02:
Could you please run one more test by adding `raw: true` and let me know what you see as results?

edwardtfn, 2025-09-02:
Ok, so I think I have something to work with. I will try to work on this soon and if I find something I come back asking for your support on testing it.
Meanwhile, try this configuration longer and let me know is the values still plausible in the long term. 🙂

edwardtfn, 2025-09-05:
Was this test with `attenuation: auto`? You got reasonable values this time and I haven't changed anything in the calcs. 🤔

edwardtfn, 2025-09-06:
So, do you mind testing it a last time before I remove the extra logs?

HenningGithub, 2025-09-08:
Lookin'  good!

[11:36:05][C][adc.esp32:121]: ADC Sensor 'Wohnzimmer Plants Moisture1 raw'
[11:36:05][C][adc.esp32:121]:   State Class: 'measurement'
[11:36:05][C][adc.esp32:121]:   Unit of Measurement: 'V'
[11:36:05][C][adc.esp32:121]:   Accuracy Decimals: 2
[11:36:05][C][adc.esp32:121]:   Device Class: 'voltage'
[11:36:05][C][adc.esp32:122]:   Pin: GPIO3
[11:36:05][C][adc.esp32:123]:   Channel:       2
[11:36:05][C][adc.esp32:123]:   Unit:          ADC1
[11:36:05][C][adc.esp32:123]:   Attenuation:   Auto
[11:36:05][C][adc.esp32:123]:   Samples:       1
[11:36:05][C][adc.esp32:123]:   Sampling mode: average
[11:36:05][C][adc.esp32:133]:   Setup Status:
[11:36:05][C][adc.esp32:133]:     Handle Init:  OK
[11:36:05][C][adc.esp32:133]:     Config:       OK
[11:36:05][C][adc.esp32:133]:     Calibration:  OK
[11:36:05][C][adc.esp32:133]:     Overall Init: OK
[11:36:05][C][adc.esp32:143]:   Update Interval: 15.0s

[11:36:15][D][adc.esp32:256]: DEBUG Autorange atten=3: Calibration handle creation SUCCESS (err=0)
[11:36:15][D][adc.esp32:262]: DEBUG Autorange atten=3: Raw ADC read SUCCESS, value=3448 (err=0)
[11:36:15][D][adc.esp32:284]: DEBUG Autorange atten=3: CALIBRATED - raw=3448 -> 1104mV -> 1.104000V
[11:36:15][D][adc.esp32:256]: DEBUG Autorange atten=2: Calibration handle creation SUCCESS (err=0)
[11:36:15][D][adc.esp32:262]: DEBUG Autorange atten=2: Raw ADC read SUCCESS, value=6425 (err=0)
[11:36:15][D][adc.esp32:284]: DEBUG Autorange atten=2: CALIBRATED - raw=6425 -> 1102mV -> 1.102000V
[11:36:15][D][adc.esp32:349]: DEBUG Autorange summary:
[11:36:15][D][adc.esp32:350]:   Raw readings: 12db=3448, 6db=6425, 2.5db=4095, 0db=4095
[11:36:15][D][adc.esp32:351]:   Voltages: 12db=1.104000, 6db=1.102000, 2.5db=0.000000, 0db=0.000000
[11:36:15][D][adc.esp32:352]:   Coefficients: c12=2048, c6=0, c2=1, c0=0, sum=2049
[11:36:15][D][adc.esp32:360]: Autorange final: (1.104000*2048 + 1.102000*0 + 0.000000*1 + 0.000000*0)/2049 = 1.103461V

and the sensors which are fed by this adc show plausible values.
Thanks a lot for fixing this issue!

closed 2025-09-09 by jesserockz