esphome/esphome #10542: web_server results in Watchdog triggers using ESP-IDF

KuhlaJusa, 2025-09-03:
### The problem

The 'web_server' component, combined with an unreliable client connection, results in watchdog triggers when using esp-idf.

I can reproduce this behaviour using a smartphone by constantly refreshing the webpage while moving away from my access point. Using the Arduino framework, this does not result in resets.

### Which version of ESPHome has the issue?

2025.8.2, but i found the issue a couple of weeks ago using 2025.7.something

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Component causing the issue

web_server using esp-idf

### YAML Config

```yaml
esphome:
  name: minimalwebserver
  friendly_name: Minimal_Web_Server
  name_add_mac_suffix: True

logger:
  level: debug

esp32:
  board: seeed_xiao_esp32s3
  variant: esp32s3
  cpu_frequency: 240MHz
  framework: 
    type: esp-idf
    # type: arduino

sensor:
  - platform: wifi_signal
    name: "WiFi Signal Strength"
    update_interval: 10s

text_sensor:
  - platform: debug
    reset_reason:
      name: "Reset Reason"

debug:

wifi:
# wifi stuff here...

web_server:
```

### Anything in the logs that might be useful for us?

```txt
[19:56:07][I][wifi:329]: Connecting to '-------'
[19:56:08][I][wifi:675]: Connected
[19:56:08][W][wifi:678]: Network '--------' should be marked as hidden
[19:56:08][C][wifi:444]:   Local ----------
[19:56:08][C][wifi:449]:   SSID: '-------'
[19:56:08][C][wifi:452]:   IP Address: --------
[19:56:08][C][wifi:456]:   BSSID: -----------
[19:56:08][C][wifi:456]:   Hostname: 'minimalwebserver'
[... 92 lines cut ...]
[19:57:50]E (106706) task_wdt: Print CPU 0 (current core) backtrace

[19:57:50]Backtrace: 0x4037a45f:0x3fca5d10 0x4201b14d:0x3fca5d30 0x4037e13f:0x3fca5d50
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x4037a45f: xt_utils_wait_for_intr at C:\Users\...\.platformio\packages\framework-espidf\components\xtensa\include/xt_utils.h:82
 (inlined by) esp_cpu_wait_for_intr at C:\Users\...\.platformio\packages\framework-espidf\components\esp_hw_support/cpu.c:55
WARNING Decoded 0x4201b14d: esp_vApplicationIdleHook at C:\Users\...\.platformio\packages\framework-espidf\components\esp_system/freertos_hooks.c:58
WARNING Decoded 0x4037e13f: prvIdleTask at C:\Users\...\.platformio\packages\framework-espidf\components\freertos\FreeRTOS-Kernel/tasks.c:4341 (discriminator 1)
```

### Additional information

Arduino Framework:

Nothing special except
```
[20:02:45][W][web_server:133]: Closing stuck EventSource connection after 2500 failed sends
[20:02:54][D][sensor:103]: 'WiFi Signal Strength': Sending state -47.00000 dBm with 0 decimals of accuracy
[20:02:54][W][web_server:133]: Closing stuck EventSource connection after 2501 failed sends
[20:03:04][D][sensor:103]: 'WiFi Signal Strength': Sending state -47.00000 dBm with 0 decimals of accuracy
[20:03:04][W][web_server:133]: Closing stuck EventSource connection after 2502 failed sends
[20:03:14][D][sensor:103]: 'WiFi Signal Strength': Sending state -47.00000 dBm with 0 decimals of accuracy
```

(I remember the first time I found it and compared it against the Arduino framework. With the Arduino, I received lots of "Event message queue overflow: discard message" errors, but I don't have any logs for that. My guess was that the TCP connection is buffering too many packets, and at some point, the ESP gets "flooded.")

ptryk, 2025-10-04:
this seems to also impact arduino frame work for me -  I guess the reason & solution are the same? or should i open another issue?

rguca, 2025-10-04:
```
20:54:11 [W] [web_server:133] Closing stuck EventSource connection after 2936 failed sends
```
I also get this on arduino framework

KuhlaJusa, 2025-10-04:
> ```
> 20:54:11 [W] [web_server:133] Closing stuck EventSource connection after 2936 failed sends
> ```
> 
> I also get this on arduino framework

I haven't done much further testing, but since creating the issue I've been running the Arduino framework without problems. Yes, while testing, I also received your error message, but the watchdog never triggered. 
So, I think it would be interesting to know whether your watchdog triggers or not.

bdraco, 2025-10-04:
IDF is blocking so instead of getting the failed sends message the watchdog triggers

bdraco, 2025-10-04:
wait for the fix https://github.com/esphome/esphome/pull/11002

closed 2025-10-05 by jesserockz