esphome/esphome #10574: kmeteriso.sensor component fails with 2025.8.3, i2c issue ???

Hiekkaharju, 2025-09-04:
### The problem

kmeteriso.sensor component fails at startup with 2025.8.3

last known working with 2025.7.2.
ESP-IDF
k-meter device is connected to a m5stack atom lite

As can be seen from the logs, kmeteriso fails without indication of reason even on Very Verbose.
Just "[E][component:145]:   kmeteriso.sensor is marked FAILED: unspecified"
i2c initialization seems ok, and k-meter address is found, but then nothing but that FAILED

### Which version of ESPHome has the issue?

2025.8.3

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

kmeteriso

### YAML Config

```yaml
substitutions:
  name: kmeter2
  friendly_name: Kmeter2
  
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}

esp32:
  board: esp32dev
  framework:
    type: esp-idf

# Enable logging
logger:
  level: VERY_VERBOSE
  logs:
    component: VERY_VERBOSE
    sensor: VERY_VERBOSE
    api.service: INFO
    api: INFO
    web_server: DEBUG

# Enable Home Assistant API
api:
  encryption:
    key: redacted

ota:
  - platform: esphome
    password: redacted

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
        static_ip: 192.168.0.226
        gateway: 192.168.0.253
        subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Kmeter2 Fallback Hotspot"
    password: redacted

i2c:
  sda: 26
  scl: 32

sensor:
- platform: kmeteriso
  update_interval: 15s
  temperature:
    name: Temperature

  internal_temperature:
    name: k-meter Internal temperature

binary_sensor:
  - platform: gpio
    name: "kmeter2 painike"
    pin:
      number: GPIO39
      inverted: true
      mode:
        input: true
        pullup: False

# Enable restart from HA button
button:
  - platform: restart
    name: "kmeter2 Restart"

web_server:
  local: true
  include_internal: true
```

### Anything in the logs that might be useful for us?

```txt
[17:53:20][I][app:200]: ESPHome version 2025.8.3 compiled on Sep  4 2025, 17:47:32
[17:53:20][C][wifi:661]: WiFi:
[17:53:20][C][wifi:444]:   Local MAC: ...
[17:53:20][C][wifi:449]:   SSID: '...'[redacted]
[17:53:20][C][wifi:452]:   IP Address: 192.168.0.226
[17:53:20][C][wifi:456]:   BSSID: ...[redacted]
[17:53:20][C][wifi:456]:   Hostname: 'kmeter2'
[17:53:20][C][wifi:456]:   Signal strength: -46 dB ▂▄▆█
[17:53:20][V][wifi:464]:   Priority: 0.0
[17:53:20][C][wifi:467]:   Channel: 3
[17:53:20][C][wifi:467]:   Subnet: 255.255.255.0
[17:53:20][C][wifi:467]:   Gateway: 192.168.0.253
[17:53:20][C][wifi:467]:   DNS1: 0.0.0.0
[17:53:20][C][wifi:467]:   DNS2: 0.0.0.0
[17:53:20][C][logger:252]: Logger:
[17:53:20][C][logger:252]:   Max Level: VERY_VERBOSE
[17:53:20][C][logger:252]:   Initial Level: VERY_VERBOSE
[17:53:20][C][logger:258]:   Log Baud Rate: 115200
[17:53:20][C][logger:258]:   Hardware UART: UART0
[17:53:20][C][logger:265]:   Task Log Buffer Size: 768
[17:53:20][C][logger:270]:   Level for 'api': INFO
[17:53:20][C][logger:270]:   Level for 'api.service': INFO
[17:53:20][C][logger:270]:   Level for 'component': VERY_VERBOSE
[17:53:20][C][logger:270]:   Level for 'sensor': VERY_VERBOSE
[17:53:20][C][logger:270]:   Level for 'web_server': DEBUG
[17:53:20][C][i2c.idf:081]: I2C Bus:
[17:53:20][C][i2c.idf:082]:   SDA Pin: GPIO26
[17:53:20][C][i2c.idf:082]:   SCL Pin: GPIO32
[17:53:20][C][i2c.idf:082]:   Frequency: 50000 Hz
[17:53:20][C][i2c.idf:092]:   Recovery: bus successfully recovered
[17:53:20][C][i2c.idf:102]: Results from bus scan:
[17:53:20][C][i2c.idf:108]: Found device at address 0x66
[17:53:20][C][gpio.binary_sensor:052]: GPIO Binary Sensor 'kmeter2 painike'
[17:53:20][C][gpio.binary_sensor:053]:   Pin: GPIO39
[17:53:20][C][gpio.binary_sensor:055]:   Mode: interrupt
[17:53:20][C][gpio.binary_sensor:072]:   Interrupt Type: ANY_EDGE
[17:53:20][E][component:145]:   kmeteriso.sensor is marked FAILED: unspecified
```

Pel1can111, 2025-09-05:
I have been running ESP-IDF where I first noticed the problem. Switching to Arduino seems to result in the same behaviour.

Hiekkaharju, 2025-09-05:
Now I had possibility to take usb log, it shows (before wifi gets ready)

[16:29:04][V][i2c.idf:130]: Writing 0 bytes, reading 0 bytes
[16:29:04][V][i2c.idf:133]: 0x66 BUS PROBE
[16:29:04][V][i2c.idf:172]: Sending 3 jobs
[16:29:04][C][kmeteriso.sensor:027]: Could write to the address 102.
[16:29:04][V][i2c.idf:130]: Writing 1 bytes, reading 1 bytes
[16:29:04][V][i2c.idf:141]: 0x66 TX 20
[16:29:04][V][i2c.idf:153]: 0x66 RX bytes 1
[16:29:04][V][i2c.idf:172]: Sending 7 jobs
[16:29:04][C][kmeteriso.sensor:037]: Could not read from the device.
[16:29:04][E][component:204]: kmeteriso.sensor was marked as failed
[16:29:04][E][component:296]: kmeteriso.sensor set Error flag: unspecified
[16:29:04][C][component:164]: Setup kmeteriso.sensor took 61ms

log file attached

[20250905 cleaned  kmeter2_usb_logs.txt](https://github.com/user-attachments/files/22173170/20250905.cleaned.kmeter2_usb_logs.txt)

clydebarrow, 2025-09-05:
Ah, I see the problem - the previous call to `read_bytes` returned true or false, the `read_register` call does the same operation but returns ERR_OK for success, which happens to be equal to 0, i.e. false. The kmeteriso code needs to check the value against ERR_OK.

closed 2025-09-06 by clydebarrow