esphome/esphome #10592: any update to fix  updating still cant upgrade from 7.5 to 8.3 now

comet424, 2025-09-04:
### The problem

still unable to to upgrade from 7.5 to 8.3 now.  same path error  same idf tools error 

any word of a fix coming?

### Which version of ESPHome has the issue?

2025.8.3

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

all componets

### YAML Config

```yaml
i unable to update any of my 54 esp devices running on these setups.. they will not update.. ive uninstalled and reinstalled esphome in home assistant.. keep gettin timeouts etc

esp32:
  board: esp32dev
  framework:
    type: esp-idf 

esp32:
  board: esp32dev
  framework:
    type: arduino

esp8266:
  board: nodemcuv2
  framework:
    version: recommended

esp8266:
  board: d1_mini
  framework:
    version: recommended

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: esp-idf
  flash_size: 16MB
```

### Anything in the logs that might be useful for us?

```txt

```

comet424, 2025-09-07:
it took me a few hours to download the zip file i dunno why things so slow.  its so buggered up  but i finally got things updated but couldnt get it to install right  i had to uninstall reinstall .. and still took several hours to download

comet424, 2025-09-07:
my internet is slow i only get  3 megabit downloads so like 400kb/s  download... but even then updates used to be faster... its like the new idf tools file  must be 20 times the size of the other firmwares.. its crazy

comet424, 2025-09-07:
or whatever files it downloaded... seems to be like 20 times normal..  and then sometimes it just goes  and says its installing packages... yet  pfsense still shows home assistant still downloading  so esphome is downloading even though it was done downloading... ever since 8.0  its been alot of issues

swoboda1337, 2025-09-07:
The compiler is about 200 megs which should only take you about 10 minutes at 3mbits. Someone using a PI and a slow SD card reported it took ~1.5 hours. However it was his hardware not internet that was the issue.

swoboda1337, 2025-09-07:
once it installs the compiler it shouldnt havent to do it again, further devices and compiles will be fast

swoboda1337, 2025-09-07:
I can download everything and build in just over a 1 minute on 2.5gb internet