esphome/esphome #10601: Switching to ESP-IDF and flashing a lolin_d32 bricks the esp

sztanpet, 2025-09-05:
### The problem

I upgraded to 2025.8.3 and update my lolin_d32 just fine, but I also noticed while it was compiling that the arduino framework is deprecated now, so I thought I would give it a go and switched the framework to esp-idf as instructed.
The compilation went smoothly, as well as the flashing of the new firmware, but the esp32 never came back online.
I had to manually reflash the esp32 by USB to make it work again with the arduino framework.

### Which version of ESPHome has the issue?

2025.8.3

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

N/A

### YAML Config

```yaml
substitutions:
  node_name: konyha

esp32:
  board: lolin_d32
esphome:
  name: $node_name
  on_boot:
    priority: -100
    then:
      - if:
          condition:
            lambda: 'return id(global_switch_nappali) || id(global_switch_furdo) || id(global_switch_halo);'
          then:
            - switch.turn_on: switch_heater
            - logger.log: "halo/nappali/furdo on: enabling heater"
          
<<: !include .common.yaml

ota:
  password: "supersecret"

output:
  - platform: ledc
    pin: GPIO19
    id: konyha_led
    frequency: "9765Hz" # 4882Hz -> 16384step  | 9765Hz -> 8192step
    max_power: 0.90 # do not have the leds at 100% duty cycle for power-saving reasons
    min_power: 0.03 # doesnt even turn on before this
    zero_means_zero: True

  - platform: gpio
    id: relay_csillar
    pin:
      number: GPIO22
      inverted: true
      mode: OUTPUT_OPEN_DRAIN

sensor:
  - platform: homeassistant
    id: nappali_temp
    entity_id: sensor.temp2_temperature
  - platform: homeassistant
    id: halo_temp
    entity_id: sensor.temp4_temperature
[... 288 lines cut ...]
              transition_length: 0s

  - id: heater_emergency_off
    mode: restart
    then:
      - delay: 6min # homeassistants climate module has a keepalive of 5min
                    # so trying to switch the heater off every 6min should only
                    # ever fire if homeassistant is down, thus ensuring no
                    # constant on condition thus burning insane amounts of gas
      - if:
          condition:
            switch.is_on: switch_heater
          then:
          - switch.turn_off: switch_heater
          - logger.log:
              format: "Emergency shutoff of heater, no keepalive received in 6min"
              level: "ERROR"
              tag: "EMERGENCY"
```

### Anything in the logs that might be useful for us?

```txt
could not connect to the esp32 via wifi when it was having the issue, so sadly no logs
```

swoboda1337, 2025-09-07:
Well arduino is not deprecated it will just not be the default. Need a serial log to debug.

bdraco, 2025-09-07:
Please see https://esphome.io/guides/troubleshooting/