esphome/esphome #10608: On esphome 2025.8 stopped working LD2410 & LD2450

jcd420, 2025-09-05:
### The problem

On 2025.7.5 everything works fine. but when I update to 2025.8 I get erro " Invalid status". If I set the esp-idf
version: 5.4.1 the i2c stop working but the ld2410 works

### Which version of ESPHome has the issue?

2025.8

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

ld2410 & ld2450 & i2c

### YAML Config

```yaml
substitutions:
  esphome_name: espsuit2bath
  esproom: Suit2Bath

esphome:
  name: ${esphome_name}
  friendly_name: ${esphome_name}

esp32:
  board: mhetesp32minikit 
  framework:
    type: esp-idf
#    version: 5.5.0
#    release: 3.3.0
    #type: arduino

packages:
  common: !include custom/commonESP32.yaml
  ld2410: !include custom/ld2410.yaml
logger:
  baud_rate: 0
  level: ERROR #INFO # ERROR INFO DEBUG VERBOSE

i2c:
  - id: bus_a
    sda: 21
    scl: 22
    #scan: true
    timeout: 5s

  - id: bus_b
    sda: 16
    scl: 17
    scan: true
    

uart:
  - id: ld2410_uart
    tx_pin: RX
    rx_pin: TX
    baud_rate: 256000
    parity: NONE
    stop_bits: 1

binary_sensor:
  - platform: gpio
    pin:
      number: 36
      mode:
        input: true
      #  pullup: true
    name: ${esproom}PIR
    filters:
      - delayed_on: 100ms
      - delayed_off: 5s
      
  

sensor:
  - platform: bh1750
    name: "Illuminance"
    i2c_id: bus_b
    accuracy_decimals: 3
    update_interval: 60s  

  - platform: htu21d
    i2c_id: bus_a
    temperature:
      name: ${esproom}Temp
      accuracy_decimals: 2
    humidity:
      name: ${esproom}Humidity
      accuracy_decimals: 1
    update_interval: 30s 
    model: htu21d

----------------------------- ld2410.yaml  ------------------------------
esphome:
  on_boot:
    priority: -100.0
    then:
      - switch.turn_off:
          id: bluetoothld2410

ld2410:
  - uart_id: ld2410_uart 

binary_sensor:
  - platform: ld2410
    has_target:
      name: Presence
    has_moving_target:
      name: Moving Target
    has_still_target:
      name: Still Target
    out_pin_presence_status:
      name: Out pin presence status

sensor:
  - platform: ld2410
    light:
      name: Light
    moving_distance:
      name: Moving Distance
    still_distance:
      name: Still Distance
    moving_energy:
      name: Move Energy
    still_energy:
      name: Still Energy
    detection_distance:
      name: Detection Distance
    g0:
      move_energy:
        name: G0 move energy
      still_energy:
        name: G0 still energy
    g1:
      move_energy:
        name: G1 move energy
      still_energy:
        name: G1 still energy
    g2:
      move_energy:
        name: G2 move energy
      still_energy:
        name: G2 still energy
    g3:
      move_energy:
        name: G3 move energy
      still_energy:
        name: G3 still energy
    g4:
      move_energy:
        name: G4 move energy
      still_energy:
        name: G4 still energy
    g5:
      move_energy:
        name: G5 move energy
      still_energy:
        name: G5 still energy
    g6:
      move_energy:
        name: G6 move energy
      still_energy:
        name: G6 still energy
    g7:
      move_energy:
        name: G7 move energy
      still_energy:
        name: G7 still energy
    g8:
      move_energy:
        name: G8 move energy
      still_energy:
        name: G8 still energy

switch:
  - platform: ld2410
    engineering_mode:
      name: Engineering mode
    bluetooth:
      name: Control bluetooth
      id: bluetoothld2410        

number:
  - platform: ld2410
    timeout:
      name: Timeout
    light_threshold:
      name: Light threshold
    max_move_distance_gate:
      name: Max move distance gate
    max_still_distance_gate:
      name: Max still distance gate
    g0:
      move_threshold:
        name: G0 move threshold
      still_threshold:
        name: G0 still threshold
    g1:
      move_threshold:
        name: G1 move threshold
      still_threshold:
        name: G1 still threshold
    g2:
      move_threshold:
        name: G2 move threshold
      still_threshold:
        name: G2 still threshold
    g3:
      move_threshold:
        name: G3 move threshold
      still_threshold:
        name: G3 still threshold
    g4:
      move_threshold:
        name: G4 move threshold
      still_threshold:
        name: G4 still threshold
    g5:
      move_threshold:
        name: G5 move threshold
      still_threshold:
        name: G5 still threshold
    g6:
      move_threshold:
        name: G6 move threshold
      still_threshold:
        name: G6 still threshold
    g7:
      move_threshold:
        name: G7 move threshold
      still_threshold:
        name: G7 still threshold
    g8:
      move_threshold:
        name: G8 move threshold
      still_threshold:
        name: G8 still threshold

button:
  - platform: ld2410
    factory_reset:
      name: Factory reset
    restart:
      name: Restart
    query_params:
      name: Query params

select:
  - platform: ld2410
    distance_resolution:
      name: Distance resolution
    baud_rate:
      name: Baud rate
    light_function:
      name: Light function
    out_pin_level:
      name: Out pin level

----------------------------- commonESP32.yaml  --------------------------
esphome:
  name_add_mac_suffix: false
  
logger:
  baud_rate: 0
  level: ERROR #INFO # ERROR INFO DEBUG VERBOSE

api:
ota:
  - platform: esphome
    id: my_ota
#    password: "66a22214a585aa0f345a2ecc77ff9284"

captive_portal:
safe_mode:
  disabled: true

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true
  reboot_timeout: 60s  
  #power_save_mode: none
  on_connect:
    - delay: 5s  
    - ble.disable:

esp32_ble:
  enable_on_boot: false

button:
  - platform: restart
    name: esp${esproom}Restart

text_sensor:
  
- platform: wifi_info
  ip_address:
    name: ESP IP Address    

sensor:
  - platform: wifi_signal # Reports the WiFi signal strength/RSSI in dB
    name: "WiFi Signal dB"
    id: wifi_signal_db
    update_interval: 30min
    entity_category: "diagnostic"

  - platform: copy # Reports the WiFi signal strength in %
    source_id: wifi_signal_db
    name: "WiFi Signal Percent"
    filters:
      - lambda: return min(max(2 * (x + 100.0), 0.0), 100.0);
    unit_of_measurement: "Signal %"
    entity_category: "diagnostic"
    device_class: ""

output:
- platform: ledc
  id: led
  pin: 2  

light: 
- platform: monochromatic
  name: "led"
  output: led
  id: ledlight
  restore_mode: RESTORE_DEFAULT_OFF
```

### Anything in the logs that might be useful for us?

```txt

```

ssieb, 2025-09-15:
@wesselvans what do you mean "no other i2c pin"?  You can do i2c on almost any pin.

axa88, 2025-09-16:
> [@wesselvans](https://github.com/wesselvans) what do you mean "no other i2c pin"? You can do i2c on almost any pin.

i think some of us find a rando online pin-out diagram that defines a particular set of pins as i²c and then assume a software bus cant be used.
I think it be helpful to link any ESPhome documentation that describes whats correct.

danielholm, 2025-09-17:
> for now i think the options are downgrading esphome to 2025.7.x version or, if you don't use i2c, downgrade esp-idf version to 5.4.1 . If you do need i2c also, 8.* versions do require newer esp-idf versions.

I think it meant that i2c doesnt work on esp-idf 5.4.1+, but that the ld24XX does. If you need i2c you should use version above 5.4.1 - in which case the ld24XX does not work.

jcd420, 2025-09-17:
I have 15 ESP devices installed in hard to reach places. It would be to much work changing there PINs.

closed 2025-11-12 by jcd420